carla-simulator/carla #7837: Add VR to UE

Ikaros520520, 2024-06-22:
Hello everyone.

What I want to do: the experimenter wears a motion capture device to control a pedestrian blueprint model walking.
I place the camera at the pedestrian blueprint head position. When the experimenter starts walking, the pedestrian model starts walking as well. But the VR camera remains in place. It does not follow the pedestrian model.
What can I do to make the VR camera bound to the pedestrian blueprint?

Device:
PNS()
VR headset: htc vive pro2 
UE: 4.26

Blyron, 2024-06-23:
Hello you will need to create a special sensor to implement this. 
Moreover we are not aimming to support VR applicattions, there is VR implementation from the community;
look at https://github.com/carla-simulator/carla/discussions/5667

closed 2024-06-23 by Blyron

donghwan-shin, 2024-09-15:
Hi, I also looked for the same and found this: https://github.com/HARPLab/DReyeVR
I haven't tried it yet, but it looks like a solution.

ajay1606, 2025-03-06:
@donghwan-shin Unfortunately, VIVE doesnt provides any support SDK for UE 4.26.

Do you have any alternative suggestions ?

donghwan-shin, 2025-03-06:
Sorry, I don't have any more good suggestions.