carla-simulator/carla #7838: [IMPROVEMENT] Add UAV (flying drone) to the simulation

chris-redfield, 2024-06-22:
Hi, I`ve been doing some research using Carla. Thanks for this incredible tool! 
it would be nice to have a feature like this in the future, it doesn't take a lot of effort to have a bare minimum if we improvise, ideas:
- add a simple drone model to the list of vehicle blueprints
- add the drone to z + X so it would be flying (remove physics so we don`t have to worry about the process of it flying)
- add it to the traffic manager as if it was a car

With these we could have UAVS with sensors at least running about in the traffic manager (of course there would be bugs).

Blyron, 2024-07-31:
We are not planning to bring Drones to CARLA atm. Feel free to implement those yourself

closed 2024-07-31 by Blyron

yrik, 2024-11-19:
Would you please recommend a resource or a suggestion on how to do it? On a high level. It would be such a cool feature, especially now when AirSim is basically not maintained.

chris-redfield, 2024-11-21:
I was able to get a static drone inside the simulation, but could not make it work with the traffic manager, as for that I would have to program the drone to move somehow, and my project didn't have time for that.
So as useless as it may sound, for my usecases it was ok for it to be stationary.

In order to reproduce:
0. Use carla with the UE Editor, not the "built version"
1. Import a drone model, got this one for me: https://www.cgtrader.com/free-3d-models/vehicle/industrial-vehicle/drone-d79e19a3-4ae8-44c1-999d-d81de12734d1
2.used  this tutorial to import the model to carla: https://www.youtube.com/watch?v=0F3ugwkISGk&ab_channel=HaoweiZhang
 - import the model
 - skip the wheel configuration in blender
 - export and import at carla
 - "fool" carla into thinking this is a 4 wheeled vehicle, drone rotors should be specified as wheels
 - also 2 of the wheels need to have their angle changed, do as the video says
 - skip the animation part for the asset at carla

 3. After the model is imported at carla, when adding it to your simulation, turn of physics, like:
```console
drone_blueprint = world.get_blueprint_library().filter('vehicle.drone.drone1')[0]

    drone_location = carla.Location(x=-43.010139, y=22.077925, z=12.224535)
    drone_rotation = carla.Rotation(pitch=-16.181599, yaw=89.799080, roll=0.000033)
    drone_spawn_point = carla.Transform(drone_location, drone_rotation)
    drone_1 = world.try_spawn_actor(drone_blueprint, drone_spawn_point)
    drone_1.set_simulate_physics(False)
```

There is a complete page about how to import vehicles at carla: https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/
But I just followed the youtube video, looked faster at the time.

donghaiwang, 2025-10-22:
Our team have added AirSim plugin into Carla on [hutb](https://github.com/OpenHUTB/hutb) .

[image]