carla-simulator/carla #7841: Empty CARLA UE5 Editor

GoodarzMehr, 2024-06-25:
CARLA version: ue5-dev
Platform/OS: Ubuntu 22.04
Problem you have experienced: I followed the instructions [here](https://carla.readthedocs.io/en/latest/build_linux_ue5/) and the build process seemed to go without a hitch. However, when I launch the editor I get a whole bunch of these _Asset Not Found_ errors, and an empty UE5 editor, as shown in the image below. A lot of folders under `CarlaUE5/Unreal/CarlaUnreal/Content/Carla` seem to be empty (for instance the `Maps` folder).

```Bash
1001[2024.06.25-03.26.01:859][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/GenericMaterials/RoadStencil/RoadStencil': Can't find file.
[2024.06.25-03.26.01:859][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/GenericMaterials/RoadStencil/RoadStencil'
1001[2024.06.25-03.26.01:860][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/Cracks/DI_DecalRoad_2': Can't find file.
[2024.06.25-03.26.01:860][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/Decals/Road/Cracks/DI_DecalRoad_2'
1001[2024.06.25-03.26.01:861][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack09': Can't find file.
[2024.06.25-03.26.01:861][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack09'
1001[2024.06.25-03.26.01:862][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack11': Can't find file.
[2024.06.25-03.26.01:862][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack11'
1001[2024.06.25-03.26.01:864][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack15': Can't find file.
[2024.06.25-03.26.01:864][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/Decals/Road/Cracks/DI_RoadCrack15'
1001[2024.06.25-03.26.01:865][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/RoadDirt/DI_RoadDirt_02': Can't find file.
[2024.06.25-03.26.01:865][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/Decals/Road/RoadDirt/DI_RoadDirt_02'
1001[2024.06.25-03.26.01:866][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset: Failed to load '/Game/Carla/Static/Decals/Road/RoadDirt/DI_RoadDirt_03': Can't find file.
[2024.06.25-03.26.01:866][  0]LogLinker: Warning: [AssetLog] 
[2024.06.25-03.26.01:925][  0]LogUObjectGlobals: Warning: Failed to find object 'MaterialInstanceConstant /Game/Carla/Static/GenericMaterials/LargeMap_materials/largeM_sidewalk/tile01/MI_largeM_tile02.MI_largeM_tile02'
[2024.06.25-03.26.01:925][  0]LogUObjectGlobals: Error: CDO Constructor (PrepareAssetsForCookingCommandlet): Failed to find MaterialInstanceConstant'/Game/Carla/Static/GenericMaterials/LargeMap_materials/largeM_sidewalk/tile01/MI_largeM_tile02.MI_largeM_tile02'

1001[2024.06.25-03.26.01:934][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/BP_TLOpenDrive.uasset: Failed to load '/Game/Carla/Static/TrafficLight/Materials/M_StreetlightsMaster_T10': Can't find file.
[2024.06.25-03.26.01:934][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/BP_TLOpenDrive.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/TrafficLight/Materials/M_StreetlightsMaster_T10'
1001[2024.06.25-03.26.01:935][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/BP_TLOpenDrive.uasset: Failed to load '/Game/Carla/Static/TrafficLight/Materials/M_StreetlightsMasked_T10': Can't find file.
[2024.06.25-03.26.01:935][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/BP_TLOpenDrive.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/TrafficLight/Materials/M_StreetlightsMasked_T10'
1001[2024.06.25-03.26.01:936][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: Failed to load '/Game/Carla/Static/TrafficLight/SM_TrafficLight__Light01': Can't find file.
[2024.06.25-03.26.01:936][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/TrafficLight/SM_TrafficLight__Light01'
1001[2024.06.25-03.26.01:937][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: Failed to load '/Game/Carla/Static/TrafficLight/SM_TrafficLight__Light02': Can't find file.
[2024.06.25-03.26.01:937][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/TrafficLight/SM_TrafficLight__Light02'
1001[2024.06.25-03.26.01:938][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: Failed to load '/Game/Carla/Static/TrafficLight/SM_TrafficLight__Light03': Can't find file.
[2024.06.25-03.26.01:938][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: VerifyImport: Failed to load package for import object 'Package /Game/Carla/Static/TrafficLight/SM_TrafficLight__Light03'
1001[2024.06.25-03.26.01:938][  0]LogLinker: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Blueprints/TrafficLight/ST_TrafficLightHead.uasset: Failed to load '/Game/Carla/Static/TrafficLight/SM_TrafficLight_holster': Can't find file.
[2024.06.25-03.26.01:938][  0]LogLinker: Warning: [AssetLog]

1001[2024.06.25-03.26.01:981][  0]LogLinker: Warning: Failed to load '/Game/Carla/Static/TrafficSign/BP_SpeedLimit60': Can't find file.
CDO Constructor (TrafficLightManager): Failed to find /Game/Carla/Static/TrafficSign/BP_SpeedLimit60.BP_SpeedLimit60_C
1001[2024.06.25-03.26.01:981][  0]LogLinker: Warning: Failed to load '/Game/Carla/Static/TrafficSign/BP_SpeedLimit60': Can't find file.
[... cut ...]
[2024.06.25-03.26.04:018][  0]LogBlueprint: Warning: [AssetLog] /home/goodi/Programs/CARLA/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Content/OnroadMapGenerator/BP_OpenDriveToMap.uasset: [Compiler] Save Current Level : Usage of 'Save Current Level' has been deprecated. The Editor Scripting Utilities Plugin is deprecated - Use the function in Level Editor Subsystem
```

[image]

What could be causing this? Does it have anything to do with my Nvidia driver version being 470.256.02 when it recommends version 515 and above? I prefer not to update the driver because it will mess up a lot of other parts of my project.

Blyron, 2024-06-25:
Hello @GoodarzMehr 

It seems the git download of the assets repo has failed.  
I recommend to go to CarlaUE5/Unreal/CarlaUnreal/Content/Carla using terminal and use a git pull. For UE5-DEV version
CarlaUE5/Unreal/CarlaUnreal/Content/Carla is a git repo pointing this https://bitbucket.org/carla-simulator/carla-content/src/ue5-dev/

About the driver recommended, we recommend what Epic Games recommend, so you can excpect unexpected behaviours regarding illumination and meshes as we are using latests Nanite and Lumen features

GoodarzMehr, 2024-06-25:
That resolved the issue, thanks!

closed 2024-06-25 by GoodarzMehr