carla-simulator/carla #7842: make launch failed at 95% on ubuntu20.04 with carla0.9.15

HONGcalmJIN, 2024-06-25:
I try to make carla0.9.15 on ubuntu20.04. 
I totally followed the guide.
However, after 'run launch', I got these errors and Unreal Engine startup progress is 95%. Then it exited. Just the last 5%!
Hope someone can help.

`EpicAccountId:
Caught signal 11 Segmentation fault
libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:704]
libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:731]
libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheckPreCalculated(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, FVector4 const&, TTraversalHistory<unsigned int>, int*) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:825]
libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:750]
libUE4Editor-MeshUtilities.so!FMeshDistanceFieldAsyncTask::DoWork() [/home/marmot/UnrealEngine_4.26/Engine/Source/Developer/MeshUtilities/Private/MeshDistanceFieldUtilities.cpp:263]
libUE4Editor-MeshUtilities.so!FAsyncTask<FMeshDistanceFieldAsyncTask>::DoWork() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:271]
libUE4Editor-MeshUtilities.so!FAsyncTask<FMeshDistanceFieldAsyncTask>::DoThreadedWork() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:295]
libUE4Editor-Core.so!FQueuedThread::Run() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/ThreadingBase.cpp:990]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185]
libpthread.so.0!UnknownFunction(0x8608)
libc.so.6!clone(+0x42)`

Sometimes I got this:
`2024.06.24-08.50.29:648][  0]LogMeshUtilities: Finished distance field build in 0.2s - 38x36x128 distance field, 32 triangles, Range [-0.0, 1.0], SM_RoadSigns01
[2024.06.24-08.50.29:648][  0]LogMeshUtilities: Finished distance field build in 0.0s - 8x8x8 distance field, 420 triangles, Range [-0.2, 0.8], SM_TrafficLight__Light03
[2024.06.24-08.50.29:648][  0]LogMeshUtilities: Finished distance field build in 0.2s - 41x28x82 distance field, 232 triangles, Range [-0.0, 1.0], SM_TrafficLight_holster
[2024.06.24-08.50.29:648][  0]LogMeshUtilities: Finished distance field build in 0.2s - 40x40x128 distance field, 432 triangles, Range [-0.0, 1.0], SM_TrafficLight__VPole_Main
[2024.06.24-08.50.29:648][  0]BlueprintLog: New page: 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11`

Blyron, 2024-06-25:
I suggest to try again until it works, it will work, your case happens in some machine, do not why, this question should be for Epic. it seems your GPU running out of memory which happens. Because at 95% Unreal execute shader compilation and other render resources like building of mesh distance fields

HONGcalmJIN, 2024-06-25:
Many thanks for your reply.
I have tried many times, lasting about one week... After trying many solutions, I finally decided to come here for help.

My GPU is 4090, so I think my GPU memory should be sufficient, and the situation monitored by nvidia-smi also shows this.
I have tried updating the GPU driver but it didn't work.

Sometimes it crashes at 83% or 71% and 95% most. 
I also got this sometimes:
`[2024.06.24-08.45.26:683][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign110
[2024.06.24-08.45.28:861][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign100
[2024.06.24-08.45.29:878][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign90
[2024.06.24-08.45.31:042][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign80
[2024.06.24-08.45.32:383][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign70
[2024.06.24-08.45.33:351][  0]LogAudioCaptureCore: Display: No Audio Capture implementations found. Audio input will be silent.
[2024.06.24-08.45.33:351][  0]LogAudioCaptureCore: Display: No Audio Capture implementations found. Audio input will be silent.
[2024.06.24-08.45.33:610][  0]LogMeshUtilities: Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign50
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
/home/marmot/RUANJIAN/program/carla-0.9.15/Util/BuildTools/BuildCarlaUE4.sh：line 225: 903093 Segmentation fault                ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} ${EDITOR_FLAGS}
make: *** [Util/BuildTools/Linux.mk:8：launch] error 139`

Hope to have more suggestions, thank you very much!

Blyron, 2024-06-25:
Driver version?

Blyron, 2024-06-25:
How much RAM, not vram?

HONGcalmJIN, 2024-06-25:
hello!
driver version: NVIDIA UNIX x86_64 Kernel Module  535.161.08
RAM: 64 GB, swap ram: 128GB

Hope this information is useful, thanks a lot

Blyron, 2024-06-25:
Sorry but don't know what can be going on

HONGcalmJIN, 2024-06-26:
OK, many thanks!
I'll keep the issue open, hopefully someone else can help, thanks again

Blyron, 2024-06-26:
You can get more info about what is going on using logs which are saved at:
CARLA_ROOT/Unreal/CarlaUE4/Saved/Logs

PatrickPromitzer, 2024-06-26:
I had a similar problem, but you'll have to look in the Log to find the problem.

The problem I had was a file in the
Unreal/CarlaUE4/Content
folder.
One Object got corrupted (or just had a problem), and UnrealEngine didn't start until I replaced the Object file with a backup.

HONGcalmJIN, 2024-06-26:
Many thanks for all the replies!
The content in CarlaUE4.log is shown below :
`
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign100
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign90
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign80
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign70
No Audio Capture implementations found. Audio input will be silent.
No Audio Capture implementations found. Audio input will be silent.
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign60
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign50
Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign30
Loaded 0 collections in 0.000953 seconds
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Saved/Collections/' took 0.00s
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Developers/marmot/Collections/' took 0.00s
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Collections/' took 0.00s
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign40
Finished distance field build in 0.2s - 92x13x81 distance field, 152 triangles, Range [-0.0, 1.0], SM_yield_
Finished distance field build in 0.2s - 38x36x128 distance field, 32 triangles, Range [-0.0, 1.0], SM_RoadSigns01
Finished distance field build in 0.1s - 60x19x31 distance field, 2332 triangles, Range [-0.0, 1.0], SM_TrafficLight__HPole01_Short
Starting the Houdini Engine module...
Failed locating or loading libHAPIL.so
Initializing Engine...
UGameplayTagsManager::InitializeManager -  0.000 s
Missing cached shader map for material LevelColorationUnlitMaterial, compiling. Is special engine material.
Missing cached shader map for material MAT_LevelColorationLitLightmapUV, compiling. Is special engine material.
Worker (1/29): shaders left to compile 474
Worker (2/29): shaders left to compile 464
Worker (3/29): shaders left to compile 454
Worker (4/29): shaders left to compile 444
Worker (5/29): shaders left to compile 434
Worker (6/29): shaders left to compile 424
Worker (7/29): shaders left to compile 414
Worker (8/29): shaders left to compile 404
Worker (9/29): shaders left to compile 394
Worker (10/29): shaders left to compile 384
Worker (11/29): shaders left to compile 374
Worker (12/29): shaders left to compile 364
Worker (13/29): shaders left to compile 354
Worker (14/29): shaders left to compile 344
Worker (15/29): shaders left to compile 334
Worker (16/29): shaders left to compile 324
Worker (17/29): shaders left to compile 314
Worker (18/29): shaders left to compile 304
Worker (19/29): shaders left to compile 294
Worker (20/29): shaders left to compile 284
Worker (21/29): shaders left to compile 274
Worker (22/29): shaders left to compile 264
Worker (23/29): shaders left to compile 254
Worker (24/29): shaders left to compile 244
Worker (25/29): shaders left to compile 234
Worker (26/29): shaders left to compile 224
Worker (27/29): shaders left to compile 214
Worker (28/29): shaders left to compile 204
Worker (29/29): shaders left to compile 194
Missing cached shader map for material ShadedLevelColorationLitMaterial, compiling. Is special engine material.
Missing cached shader map for material ShadedLevelColorationUnlitMateri, compiling. Is special engine material.
Missing cached shader map for material VertexColorMaterial, compiling. 
Missing cached shader map for material VertexColorViewMode_ColorOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_AlphaAsColor, compiling. 
Missing cached shader map for material VertexColorViewMode_RedOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_GreenOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_BlueOnly, compiling. 
Missing cached shader map for material RemoveSurfaceMaterial, compiling. 
Missing cached shader map for material DebugMeshMaterial, compiling. 
Missing cached shader map for material EmissiveMeshMaterial, compiling. 
Missing cached shader map for material M_InvalidLightmapSettings, compiling. 
Missing cached shader map for material GizmoMaterial, compiling. 
Missing cached shader map for material PhAT_JointLimitMaterial, compiling. 
Missing cached shader map for material GeomMaterial, compiling. 
[... 55 lines cut ...]
OS version Linux 5.15.0-107-generic (network name: Cisco-2)
Running 24 x86_64 processors (32 logical cores)
Exception was "SIGSEGV: invalid attempt to write memory at address 0x0000000000000003"

<SOURCE START>
<SOURCE END>

<CALLSTACK START>
libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
[... cut ...]
libUE4Editor-Core.so!FQueuedThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/ThreadingBase.cpp:990]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185]
libpthread.so.0!UnknownFunction(0x8608)
libc.so.6!clone(+0x42)

<CALLSTACK END>
`

HONGcalmJIN, 2024-06-26:
Many thanks for all the replies!
The content in CarlaUE4.log is shown below :
```
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign100
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign90
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign80
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign70
No Audio Capture implementations found. Audio input will be silent.
No Audio Capture implementations found. Audio input will be silent.
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign60
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign50
Finished distance field build in 1.4s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign30
Loaded 0 collections in 0.000953 seconds
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Saved/Collections/' took 0.00s
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Developers/marmot/Collections/' took 0.00s
Scanning file cache for directory '/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Collections/' took 0.00s
Finished distance field build in 1.3s - 128x56x128 distance field, 140 triangles, Range [-0.0, 1.0], SM_SpeedSign40
Finished distance field build in 0.2s - 92x13x81 distance field, 152 triangles, Range [-0.0, 1.0], SM_yield_
Finished distance field build in 0.2s - 38x36x128 distance field, 32 triangles, Range [-0.0, 1.0], SM_RoadSigns01
Finished distance field build in 0.1s - 60x19x31 distance field, 2332 triangles, Range [-0.0, 1.0], SM_TrafficLight__HPole01_Short
Starting the Houdini Engine module...
Failed locating or loading libHAPIL.so
Initializing Engine...
UGameplayTagsManager::InitializeManager -  0.000 s
Missing cached shader map for material LevelColorationUnlitMaterial, compiling. Is special engine material.
Missing cached shader map for material MAT_LevelColorationLitLightmapUV, compiling. Is special engine material.
Worker (1/29): shaders left to compile 474
Worker (2/29): shaders left to compile 464
Worker (3/29): shaders left to compile 454
Worker (4/29): shaders left to compile 444
Worker (5/29): shaders left to compile 434
Worker (6/29): shaders left to compile 424
Worker (7/29): shaders left to compile 414
Worker (8/29): shaders left to compile 404
Worker (9/29): shaders left to compile 394
Worker (10/29): shaders left to compile 384
Worker (11/29): shaders left to compile 374
Worker (12/29): shaders left to compile 364
Worker (13/29): shaders left to compile 354
Worker (14/29): shaders left to compile 344
Worker (15/29): shaders left to compile 334
Worker (16/29): shaders left to compile 324
Worker (17/29): shaders left to compile 314
Worker (18/29): shaders left to compile 304
Worker (19/29): shaders left to compile 294
Worker (20/29): shaders left to compile 284
Worker (21/29): shaders left to compile 274
Worker (22/29): shaders left to compile 264
Worker (23/29): shaders left to compile 254
Worker (24/29): shaders left to compile 244
Worker (25/29): shaders left to compile 234
Worker (26/29): shaders left to compile 224
Worker (27/29): shaders left to compile 214
Worker (28/29): shaders left to compile 204
Worker (29/29): shaders left to compile 194
Missing cached shader map for material ShadedLevelColorationLitMaterial, compiling. Is special engine material.
Missing cached shader map for material ShadedLevelColorationUnlitMateri, compiling. Is special engine material.
Missing cached shader map for material VertexColorMaterial, compiling. 
Missing cached shader map for material VertexColorViewMode_ColorOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_AlphaAsColor, compiling. 
Missing cached shader map for material VertexColorViewMode_RedOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_GreenOnly, compiling. 
Missing cached shader map for material VertexColorViewMode_BlueOnly, compiling. 
Missing cached shader map for material RemoveSurfaceMaterial, compiling. 
Missing cached shader map for material DebugMeshMaterial, compiling. 
Missing cached shader map for material EmissiveMeshMaterial, compiling. 
Missing cached shader map for material M_InvalidLightmapSettings, compiling. 
Missing cached shader map for material GizmoMaterial, compiling. 
Missing cached shader map for material PhAT_JointLimitMaterial, compiling. 
Missing cached shader map for material GeomMaterial, compiling. 
[... 56 lines cut ...]
OS version Linux 5.15.0-107-generic (network name: Cisco-2)
Running 24 x86_64 processors (32 logical cores)
Exception was "SIGSEGV: invalid attempt to write memory at address 0x0000000000000003"

<SOURCE START>
<SOURCE END>

<CALLSTACK START>
libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
[... cut ...]
libUE4Editor-Core.so!FQueuedThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/ThreadingBase.cpp:990]
libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/marmot/RUANJIAN/program/UnrealEngine_4.26/Engine/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185]
libpthread.so.0!UnknownFunction(0x8608)
libc.so.6!clone(+0x42)

<CALLSTACK END>
```

PatrickPromitzer, 2024-06-26:
There is one thing I just noticed.
You can try to change the default and startup map in
Unreal/CarlalUE4/Config/DefaultEngine.ini
for example
```
EditorStartupMap=/Game/Carla/Maps/OpenDriveMap.OpenDriveMap
```

I can't say if it helps, but it would be worth a try.

HONGcalmJIN, 2024-06-27:
Thank you very much, I tried again with your advice. But it crashed at 84% now after a number of attempts.
The content of the log is shown below. We can see "Found 2 device(s)
Device 0: NVIDIA GeForce RTX 4090". But there is no output of "Device 1". Is there any problem here?

`Found 2 device(s)
Device 0: NVIDIA GeForce RTX 4090
- API 1.3.242(0x4030f2) Driver 0x85edc040 VendorId 0x10de
- DeviceID 0x2684 Type Discrete GPU
- Max Descriptor Sets Bound 32 Timestamps 1
- Device Properties driverName: NVIDIA
- Device Properties driverInfo: 535.183.01
-    Found device layer VK_LAYER_MESA_device_select
-    Found device extension VK_AMD_buffer_marker
-    Found device extension VK_EXT_4444_formats
-    Found device extension VK_EXT_attachment_feedback_loop_layout
-    Found device extension VK_EXT_blend_operation_advanced
-    Found device extension VK_EXT_border_color_swizzle
-    Found device extension VK_EXT_buffer_device_address
-    Found device extension VK_EXT_calibrated_timestamps
-    Found device extension VK_EXT_color_write_enable
-    Found device extension VK_EXT_conditional_rendering
-    Found device extension VK_EXT_conservative_rasterization
-    Found device extension VK_EXT_custom_border_color
-    Found device extension VK_EXT_depth_clamp_zero_one
-    Found device extension VK_EXT_depth_clip_control
-    Found device extension VK_EXT_depth_clip_enable
-    Found device extension VK_EXT_depth_range_unrestricted
-    Found device extension VK_EXT_descriptor_buffer
-    Found device extension VK_EXT_descriptor_indexing
-    Found device extension VK_EXT_discard_rectangles
-    Found device extension VK_EXT_display_control
-    Found device extension VK_EXT_extended_dynamic_state
-    Found device extension VK_EXT_extended_dynamic_state2
-    Found device extension VK_EXT_extended_dynamic_state3
-    Found device extension VK_EXT_external_memory_host
-    Found device extension VK_EXT_fragment_shader_interlock
-    Found device extension VK_EXT_global_priority
-    Found device extension VK_EXT_global_priority_query
-    Found device extension VK_EXT_graphics_pipeline_library
-    Found device extension VK_EXT_host_query_reset
-    Found device extension VK_EXT_image_2d_view_of_3d
-    Found device extension VK_EXT_image_drm_format_modifier
-    Found device extension VK_EXT_image_robustness
-    Found device extension VK_EXT_image_sliced_view_of_3d
-    Found device extension VK_EXT_image_view_min_lod
-    Found device extension VK_EXT_index_type_uint8
-    Found device extension VK_EXT_inline_uniform_block
-    Found device extension VK_EXT_line_rasterization
-    Found device extension VK_EXT_load_store_op_none
-    Found device extension VK_EXT_memory_budget
-    Found device extension VK_EXT_memory_priority
-    Found device extension VK_EXT_mesh_shader
-    Found device extension VK_EXT_multi_draw
-    Found device extension VK_EXT_mutable_descriptor_type
-    Found device extension VK_EXT_non_seamless_cube_map
-    Found device extension VK_EXT_opacity_micromap
-    Found device extension VK_EXT_pageable_device_local_memory
-    Found device extension VK_EXT_pci_bus_info
-    Found device extension VK_EXT_physical_device_drm
-    Found device extension VK_EXT_pipeline_creation_cache_control
-    Found device extension VK_EXT_pipeline_creation_feedback
-    Found device extension VK_EXT_pipeline_library_group_handles
-    Found device extension VK_EXT_pipeline_robustness
-    Found device extension VK_EXT_post_depth_coverage
-    Found device extension VK_EXT_primitive_topology_list_restart
-    Found device extension VK_EXT_primitives_generated_query
-    Found device extension VK_EXT_private_data
-    Found device extension VK_EXT_provoking_vertex
-    Found device extension VK_EXT_queue_family_foreign
-    Found device extension VK_EXT_robustness2
[... 137 lines cut ...]
0x00007fbec80fa082 Dladdr: 0.008849ms Open: 0.001437ms Search: 0.000355ms
0x0000000000233238 Dladdr: 0.001408ms Open: 0.001493ms Search: 0.007523ms
FPlatformStackWalk::StackWalkAndDump -  0.000 s
=== Handled ensure: ===

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]

Stack: 
[Callstack] 0x00007fbc7d615ea3 libUE4Editor-VulkanRHI.so!FVulkanDevice::FVulkanDevice(FVulkanDynamicRHI*, VkPhysicalDevice_T*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:182]
[Callstack] 0x00007fbc7d6af8c6 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::SelectAndInitDevice() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:551]
[Callstack] 0x00007fbc7d6abbf3 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::InitInstance() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:694]
[Callstack] 0x00007fbc7d6ab808 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::Init() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:241]
[Callstack] 0x00007fbecb7c918d libUE4Editor-RHI.so!RHIInit(bool) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp:228]
[Callstack] 0x00000000002530a3 UE4Editor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2555]
[... cut ...]
[Callstack] 0x00007fbec82d6286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
[Callstack] 0x00007fbec80fa083 libc.so.6!__libc_start_main(+0xf2)
[Callstack] 0x0000000000233239 UE4Editor!_start()

               SubmitErrorReport -  0.000 s`

I finally installed successfully in my Windows PC. So maybe the failure in this ubuntu20 PC is acceptable. 
Anyway i really want to make it in my ubuntu PC which has 4090.
Many Thanks to everyone for your suggestions. 
Sincerely

Blyron, 2024-06-27:
The only thing it comes to my mind is that during download of content something got corrupted. 
You can try by Deleting Carla/Unreal/CarlaUE4/Content/Carla folder content and then in that folder clone the repo itself:
https://bitbucket.org/carla-simulator/carla-content/src/master/

PatrickPromitzer, 2024-06-27:
If it is about the Content.tar.gz file, you can try to create a sha256sum hash, to see if there is a problem.

Download the file directly:
https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz
and create a sha256 hash
```command line
sha256sum 20231108_c5101a5.tar.gz
```
Hash:
(file from 14.12.2023)
158d15ac7ced5881b645244055af6117fcc521bca1af7b58a119cb28137a8906  20231108_c5101a5.tar.gz
(file from 27.06.2024; current)
**9287d628547d4e925d25de72b97b61dc0f9173725286bf018449edb3eeecba8b**  20231108_c5101a5.tar.gz

It the Hash is different, there is a problem with the download.

If the Hash is the same, you can do 2 things.
1.) Edit the "Update.sh" file to not download the Content.tar.gz file and use your file.
2.) copy the files from the Content.tar.gz in the Unreal/CarlaUE4/Content folder (delete and create a new Content folder)

I hope the Hash is the same.
If it isn't try to sha256sum command again. If it is different each time, you have a deeper problem.
(I had a broken RAM which made me create different sha256sum each time I created one.)

HONGcalmJIN, 2024-06-27:
I tried this before, but it did not work. This will raise new issue:
`The file '../../../../RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset' contains unrecognizable data, check that it is of the expected type.
The file '../../../../RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Content/Carla/Blueprints/LevelDesign/RoadPainterPreset.uasset' contains unrecognizable data, check that it is of the expected type.
looking to “Blueprint /Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset” failed
CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'

=== Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000038

0x00007fc9c082cdd4 libUE4Editor-Carla.so!ULoadAssetMaterialsCommandlet::ULoadAssetMaterialsCommandlet() [/home/marmot/RUANJIAN/program/carla-0.9.15/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/LoadAssetMaterialsCommandlet.cpp:38]
0x00007fcc8bab5ff4 libUE4Editor-CoreUObject.so!UClass::CreateDefaultObject() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:3707]
0x00007fcc8bd5b608 libUE4Editor-CoreUObject.so!ProcessNewlyLoadedUObjects(FName, bool) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectBase.cpp:996]
0x00007fcc8bc8ef2e libUE4Editor-CoreUObject.so!TBaseStaticDelegateInstance<void (FName, bool), FDefaultDelegateUserPolicy>::ExecuteIfSafe(FName, bool) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:729]
0x00007fcc8c76b388 libUE4Editor-Core.so!TMulticastDelegate<void (FName, bool), FDefaultDelegateUserPolicy>::Broadcast(FName, bool) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:955]
0x00007fcc8c742a27 libUE4Editor-Core.so!FModuleManager::LoadModuleWithFailureReason(FName, EModuleLoadResult&) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Modules/ModuleManager.cpp:516]
0x00007fcc86ccadde libUE4Editor-Projects.so!FModuleDescriptor::LoadModulesForPhase(ELoadingPhase::Type, TArray<FModuleDescriptor, TSizedDefaultAllocator<32> > const&, TMap<FName, EModuleLoadResult, FDefaultSetAllocator, TDefaultMapHashableKeyFuncs<FName, EModuleLoadResult, false> >&) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/ModuleDescriptor.cpp:560]
0x00007fcc86cde48c libUE4Editor-Projects.so!FPluginManager::TryLoadModulesForPlugin(FPlugin const&, ELoadingPhase::Type) const [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1293]
0x00007fcc86ce02ee libUE4Editor-Projects.so!FPluginManager::LoadModulesForEnabledPlugins(ELoadingPhase::Type) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1369]
0x0000000000260f9e UE4Editor!FEngineLoop::LoadStartupModules() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3808]
0x000000000025c2c8 UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3198]
0x0000000000249feb UE4Editor!GuardedMain(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
0x00007fcc833e0286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
0x00007fcc83204083 libc.so.6!__libc_start_main(+0xf2)
0x0000000000233239 UE4Editor!_start()

Executing StaticShutdownAfterError
`

If i run ./UE4Editor, i got this. But this will disappear when i `make` again in UE4Editor. If I `make launch` again afer remake of the UE4Editor, it will fail stil.
[image]

Since UE4 use a new repo [EpicGames-Mirror-A](https://github.com/EpicGames-Mirror-A/UnrealEngine) which do not contain the Carla branch, i use the download version [CarlaUnreal](https://github.com/CarlaUnreal/UnrealEngine) as described in the docs from my friends. It seems that the UE4Editor has some problems.
I would highly appreciate it if someone could share me the code under [git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26] or [ git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5].

HONGcalmJIN, 2024-06-27:
> If it is about the Content.tar.gz file, you can try to create a sha256sum hash, to see if there is a problem.
> 
> Download the file directly: https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz and create a sha256 hash
> 
> ```shell
> sha256sum 20231108_c5101a5.tar.gz
> ```
> 
> Hash: (file from 14.12.2023) 158d15ac7ced5881b645244055af6117fcc521bca1af7b58a119cb28137a8906 20231108_c5101a5.tar.gz (file from 27.06.2024; current) **9287d628547d4e925d25de72b97b61dc0f9173725286bf018449edb3eeecba8b** 20231108_c5101a5.tar.gz
> 
> It the Hash is different, there is a problem with the download.
> 
> If the Hash is the same, you can do 2 things. 1.) Edit the "Update.sh" file to not download the Content.tar.gz file and use your file. 2.) copy the files from the Content.tar.gz in the Unreal/CarlaUE4/Content folder (delete and create a new Content folder)
> 
> I hope the Hash is the same. If it isn't try to sha256sum command again. If it is different each time, you have a deeper problem. (I had a broken RAM which made me create different sha256sum each time I created one.)
Many thanks!
I got this:
`9287d628547d4e925d25de72b97b61dc0f9173725286bf018449edb3eeecba8b  20231108_c5101a5.tar.gz`.
Therefore, it looks like the problem isn't here.

Blyron, 2024-06-27:
you have

> I tried this before, but it did not work. This will raise new issue:
>
> ```
> The file [...]

This first issue is becasue the folders are not properly placed it have to look like
CARLA_ROOT/Unreal/CarlaUE4/Content/Carla
                                                                                       Blueprint
                                                                                       Config
                                                                                       HDMaps
                                                                                       Maps
                                                                                       road_xodr
                                                                                       Static

HONGcalmJIN, 2024-06-27:
Yes, it's interesting that if the CARLA_ROOT is "carla-0.9.15", this error will happen.
After handling this, the `make launch` crashed at 71% with the problem before. 
Therefore, maybe the problem lies in the install of UE4Editor as I mentioned before.
```
0x00007fa0550c1ea2 Dladdr: 0.008174ms Open: 0.017208ms Search: 0.012832ms
0x00007fa05515b8c5 Dladdr: 0.005635ms Open: 0.002040ms Search: 0.010827ms
0x00007fa055157bf2 Dladdr: 0.003039ms Open: 0.001174ms Search: 0.009574ms
0x00007fa055157807 Dladdr: 0.002604ms Open: 0.001110ms Search: 0.007074ms
0x00007fa2a31a318c Dladdr: 0.004888ms Open: 0.003845ms Search: 0.009639ms
0x00000000002530a2 Dladdr: 0.002492ms Open: 0.003681ms Search: 0.008858ms
0x0000000000249fcb Dladdr: 0.000867ms Open: 0.001646ms Search: 0.007130ms
0x00007fa29fcb0285 Dladdr: 0.000300ms Open: 0.003487ms Search: 0.004890ms
0x00007fa29fad4082 Dladdr: 0.009001ms Open: 0.005437ms Search: 0.000463ms
0x0000000000233238 Dladdr: 0.001497ms Open: 0.001458ms Search: 0.007496ms
FPlatformStackWalk::StackWalkAndDump -  0.000 s
=== Handled ensure: ===

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]

Stack: 
[Callstack] 0x00007fa0550c1ea3 libUE4Editor-VulkanRHI.so!FVulkanDevice::FVulkanDevice(FVulkanDynamicRHI*, VkPhysicalDevice_T*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:182]
[Callstack] 0x00007fa05515b8c6 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::SelectAndInitDevice() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:551]
[Callstack] 0x00007fa055157bf3 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::InitInstance() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:694]
[Callstack] 0x00007fa055157808 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::Init() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:241]
[Callstack] 0x00007fa2a31a318d libUE4Editor-RHI.so!RHIInit(bool) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp:228]
[Callstack] 0x00000000002530a3 UE4Editor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2555]
[Callstack] 0x0000000000249fcc UE4Editor!GuardedMain(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
[Callstack] 0x00007fa29fcb0286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
[Callstack] 0x00007fa29fad4083 libc.so.6!__libc_start_main(+0xf2)
[Callstack] 0x0000000000233239 UE4Editor!_start()

               SubmitErrorReport -  0.000 s
```

Blyron, 2024-06-27:
> Yes, it's interesting that if the CARLA_ROOT is "carla-0.9.15", this error will happen. After handling this, the `make launch` crashed at 71% with the problem before. Therefore, maybe the problem [...]

Then try to execute again in UE4 source code folder
Setup.sh 
and generateprojectfiles.sh

PatrickPromitzer, 2024-06-27:
I forgot about the sub-folder "Carla" in which the content of the Content.tar.gz file should go into.
I update my command to make it clear for people in the future.

You can try make a clean install of CarlaUnreal and Carla, but it will take some time.
Maybe you 2 are right, and some files in the CarlaUnreal folder got a problem.

HONGcalmJIN, 2024-06-28:
Thank you very much.
I tried again from the very beginning (many times in the past week in fact), but it is very confusing that i met this problem again, though i am sure that the folders are properly placed. It loos like :
```
CARLA_ROOT/Unreal/CarlaUE4/Content/Carla
Blueprint
Config
HDMaps
Maps
road_xodr
Static
```
The error:
```
0x00007fa0550c1ea2 Dladdr: 0.008174ms Open: 0.017208ms Search: 0.012832ms
0x00007fa05515b8c5 Dladdr: 0.005635ms Open: 0.002040ms Search: 0.010827ms
0x00007fa055157bf2 Dladdr: 0.003039ms Open: 0.001174ms Search: 0.009574ms
0x00007fa055157807 Dladdr: 0.002604ms Open: 0.001110ms Search: 0.007074ms
0x00007fa2a31a318c Dladdr: 0.004888ms Open: 0.003845ms Search: 0.009639ms
0x00000000002530a2 Dladdr: 0.002492ms Open: 0.003681ms Search: 0.008858ms
0x0000000000249fcb Dladdr: 0.000867ms Open: 0.001646ms Search: 0.007130ms
0x00007fa29fcb0285 Dladdr: 0.000300ms Open: 0.003487ms Search: 0.004890ms
0x00007fa29fad4082 Dladdr: 0.009001ms Open: 0.005437ms Search: 0.000463ms
0x0000000000233238 Dladdr: 0.001497ms Open: 0.001458ms Search: 0.007496ms
FPlatformStackWalk::StackWalkAndDump -  0.000 s
=== Handled ensure: ===

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]

Stack: 
[Callstack] 0x00007fa0550c1ea3 libUE4Editor-VulkanRHI.so!FVulkanDevice::FVulkanDevice(FVulkanDynamicRHI*, VkPhysicalDevice_T*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:182]
[Callstack] 0x00007fa05515b8c6 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::SelectAndInitDevice() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:551]
[Callstack] 0x00007fa055157bf3 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::InitInstance() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:694]
[Callstack] 0x00007fa055157808 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::Init() [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:241]
[Callstack] 0x00007fa2a31a318d libUE4Editor-RHI.so!RHIInit(bool) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp:228]
[Callstack] 0x00000000002530a3 UE4Editor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2555]
[Callstack] 0x0000000000249fcc UE4Editor!GuardedMain(char16_t const*) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
[Callstack] 0x00007fa29fcb0286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/marmot/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
[Callstack] 0x00007fa29fad4083 libc.so.6!__libc_start_main(+0xf2)
[Callstack] 0x0000000000233239 UE4Editor!_start()

               SubmitErrorReport -  0.000 s
```

HONGcalmJIN, 2024-06-28:
I also tried the latest ''dev'' branch, whos "./Update.sh" can run normally. But it still failed at 71% with the same error. :)

```
Worker (28/29): shaders left to compile 668
Compiling Textures：T_LicensePlate_d（AutoDXT、512X512）
=== Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x00000000ffffffff

0x00007f59c8172d11 libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:759]
0x00007f59c817311a libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheckPreCalculated(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, FVector4 const&, TTraversalHistory<unsigned int>, int*) const [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:818]
0x00007f59c8172d67 libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:750]
0x00007f59c8172da0 libUE4Editor-MeshUtilities.so!TkDOPNode<FMeshBuildDataProvider const, unsigned int>::LineCheck(TkDOPLineCollisionCheck<FMeshBuildDataProvider const, unsigned int>&, TTraversalHistory<unsigned int>) const [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Engine/Public/kDOP.h:737]
0x00007f59c811f91b libUE4Editor-MeshUtilities.so!FMeshDistanceFieldAsyncTask::DoWork() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Developer/MeshUtilities/Private/MeshDistanceFieldUtilities.cpp:263]
0x00007f59c817579e libUE4Editor-MeshUtilities.so!FAsyncTask<FMeshDistanceFieldAsyncTask>::DoWork() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:271]
0x00007f59c8175409 libUE4Editor-MeshUtilities.so!FAsyncTask<FMeshDistanceFieldAsyncTask>::DoThreadedWork() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Public/Async/AsyncWork.h:295]
0x00007f5c4898eae8 libUE4Editor-Core.so!FQueuedThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/HAL/ThreadingBase.cpp:990]
0x00007f5c489899b7 libUE4Editor-Core.so!FRunnableThreadPThread::Run() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.cpp:25]
0x00007f5c4894ddc3 libUE4Editor-Core.so!FRunnableThreadPThread::_ThreadProc(void*) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/HAL/PThreadRunnableThread.h:185]
0x00007f5c492da609 libpthread.so.0!UnknownFunction(0x8608)
0x00007f5c3f765353 libc.so.6!clone(+0x42)

Executing StaticShutdownAfterError
```
It looks like the process failed when compiling textures is about to complete.

HONGcalmJIN, 2024-06-28:
It's also very interesting that if I change the "EditorStartupMap" etc. from " Town10HD_Opt" to " Town05_Opt".
I got this and the computer will reboot automatically. 
Therefore, maybe the essential problem lies in the graphics card related drivers with UE4Editor? 
Since 4090 is relatively a new card.
I also met this problem before hahaha. 
It crashed at about 73%.
```
0x00007f8fb5b0d082 Dladdr: 0.008302ms Open: 0.001730ms Search: 0.000429ms
0x0000000000233258 Dladdr: 0.001369ms Open: 0.001305ms Search: 0.007302ms
FPlatformStackWalk::StackWalkAndDump -  0.000 s
=== Handled ensure: ===

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]

Stack: 
[Callstack] 0x00007f8d6b235043 libUE4Editor-VulkanRHI.so!FVulkanDevice::FVulkanDevice(FVulkanDynamicRHI*, VkPhysicalDevice_T*) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:182]
[Callstack] 0x00007f8d6b2cea66 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::SelectAndInitDevice() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:551]
[Callstack] 0x00007f8d6b2cad93 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::InitInstance() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:694]
[Callstack] 0x00007f8d6b2ca9a8 libUE4Editor-VulkanRHI.so!FVulkanDynamicRHI::Init() [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp:241]
[Callstack] 0x00007f8fb91dc24d libUE4Editor-RHI.so!RHIInit(bool) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/RHI/Private/DynamicRHI.cpp:228]
[Callstack] 0x00000000002530c3 UE4Editor!FEngineLoop::PreInitPreStartupScreen(char16_t const*) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2555]
[Callstack] 0x0000000000249fec UE4Editor!GuardedMain(char16_t const*) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
[Callstack] 0x00007f8fb5ce9286 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
[Callstack] 0x00007f8fb5b0d083 libc.so.6!__libc_start_main(+0xf2)
[Callstack] 0x0000000000233259 UE4Editor!_start()

               SubmitErrorReport -  0.000 s
```

PatrickPromitzer, 2024-06-28:
I found a post about the "Ensure condition failed: VendorId != EGpuVendorId::Unknown" error.
https://github.com/carla-simulator/carla/discussions/4644

They fined it with installing a newer driver version (550?), but I look if I can find more about the problem.

The 
"Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000038"
error is still strange.

PatrickPromitzer, 2024-06-28:
You can try installing some packages, too
```
sudo apt-get install -y libnvidia-gl
sudo apt-get install -y mesa-vulkan-drivers vulkan-utils
```

I found a post about a setting which could make a problem.
https://stackoverflow.com/questions/73790229/unreal-engine-4-x-on-linux-crashes-with-error-sigsegv-invalid-attempt-to-write

\Config\DefaultGame.ini 
```
[/Script/UnrealEd.ProjectPackagingSettings]
bShareMaterialShaderCode=False
```

Someone wrote about randomly getting "read memory" error, and suggested testing the RAM. (I don't think this is the problem, but you can test it to be safe)
```
sudo apt-get install memtester
```
sudo memtester ${amount of memory to test} ${number of test iterations}
example:
```
sudo memtester 1G 1
```

HONGcalmJIN, 2024-06-28:
> I found a post about the "Ensure condition failed: VendorId != EGpuVendorId::Unknown" error. #4644
>
> They fined it with installing a newer driver version (550?), but I look if I can find more about [...]

My version is 535, which match the output of " ubuntu-drivers --gpgpu devices".
So maybe the driver version is new enough.
Yes, i have tested the RAM before and check the system log, but it seems that my RAM is ok.

HONGcalmJIN, 2024-06-28:
> You can try installing some packages, too
>
> ```
> sudo apt-get install -y libnvidia-gl
> sudo apt-get install -y mesa-vulkan-drivers vulkan-utils
> ```
>
> I found a post about a setting which could make a [...]

Thank you very much !
After install these, the error still :)
Yes, i have tested the RAM with memtester before and check the system log, but it seems that my RAM is ok.

PatrickPromitzer, 2024-06-28:
> My version is 535, which match the output of " ubuntu-drivers --gpgpu devices". So maybe the driver version is new enough.

Someone wrote who had a problem with the 535 driver version.

At this point I am guessing how you could fix it.

I have a list a packages you could try to install.
I used it to create a docker image of Carla 0.9.14
```
sudo apt-get  install python3-distutils aria2 xorg-dev libsdl2-2.0-0 xserver-xorg libvulkan1 libomp5 vulkan-utils 
```

HONGcalmJIN, 2024-06-29:
Thank you very much. 
I installed these and reboot but the problem still:

```
PlatformStackWalk::StackWalkAndDump -  0.000 s
=== Handled ensure: ===

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182]

```

Good news is that we finally locate the problem.

Blyron, 2024-06-29:
> Thank you very much. I installed these and reboot but the problem still:
>
> ```
> PlatformStackWalk::StackWalkAndDump -  0.000 s
> === Handled ensure: ===
>
> Ensure condition failed: VendorId != [...]

The Handled ensure is there from long time ago. Is not a really issue, everyone works with CARLA properly despite of having that "handled ensure" The issue must come later, surely the crashes you are experimenting during the load of the project.

HONGcalmJIN, 2024-06-29:
Yes, you're right.
I solved this " VendorId != EGpuVendorId::Unknown" problem by directly set ``` GpuCount=1; ``` in Engine/Source/Runtime/VulkanRHI/Private/VulkanRHI.cpp, line 547.

And then, if i change the default map from "Town10HD_Opt”  to “Town05_Opt” or "Town06”. Finally, the carlaUE4 can be launched. There will be this error, but I can still use the CarlaUE4.  
[image]
It seems like an error if we donot put the assets in right directory.

If I use the default “Town10HD_Opt”, I will encounter this Annoying mistakes, which I haven't solved yet.
```
[2024.06.29-12.08.40:993][  0]LogShaderCompilers: Display: Worker (14/29): shaders left to compile 772
[2024.06.29-12.08.40:993][  0]LogShaderCompilers: Display: Worker (27/29): shaders left to compile 762
[2024.06.29-12.08.40:993][  0]LogCore: Error: appError called: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
Array index out of bounds: 1043292983 from an array of size 1045388557

[2024.06.29-12.08.41:007][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x0000000000000003

[2024.06.29-12.08.41:007][  0]LogCore: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
Array index out of bounds: 1043292983 from an array of size 1045388557

0x00007f787658725e libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
0x00007f78767d256b libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
0x00007f7876752b66 libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:104]
```
Many thanks to you all! After continuous attempts, the CarlaUE4 can finally be launched anyway. ：）

PatrickPromitzer, 2024-07-01:
You are Welcome.

Good to hear that you can launch Carla.
You could try switching to the "Town10HD_Op" after the editor opened, but I believe a file has a problem not not be able to start the Map.
If you don't need the "Town10HD_Op" Map, I would say you are good for now.

HONGcalmJIN, 2024-07-01:
Yes, it will output  the following typical error if I switch to ```Town10HD_Opt```:
```
[2024.07.01-08.55.37:562][353]LogShaderCompilers: Display: Worker (13/29): shaders left to compile 738
[2024.07.01-08.55.37:562][353]LogStaticMesh: Building static mesh SM_Skyscraper_P03_Merged...
[2024.07.01-08.55.38:947][353]LogCore: Error: appError called: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
Array index out of bounds: 1059070274 from an array of size 1055160870

[2024.07.01-08.55.38:963][353]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x0000000000000003

[2024.07.01-08.55.38:963][353]LogCore: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
Array index out of bounds: 1059070274 from an array of size 1055160870

0x00007ff92660625e libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
0x00007ff92685156b libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
```

The error I met when launching ```Town06``` is also very strange. Hope it can be solved someday, and the above errors.
```
[2024.07.01-08.19.26:817][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Rendering_and_Lighting_components) is flushing async loading
[2024.07.01-08.19.26:887][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Buildings) is flushing async loading
[2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for BoolProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:BlockSpawn_12_4FEC57D947F4B49EA11F31A19F18EB48 Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
[2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for ObjectProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:Item_16_6D77EDC54B56B5883D9DC7809D6C704A Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
[2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for StructProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:Transform_5_626174554EFC313F198DA2AED94C7F1B Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
[2024.07.01-08.19.27:236][  0]Warning:  applied to a non-external Texture2D sampler. This may work by chance on some platforms but is not portable. Please change sampler type to 'External'. Parameter 'EmissiveMap' (slot 1) in material 'M_BuildingMaterial'
[2024.07.01-08.19.27:236][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Foliage) is flushing async loading
```

ba1Sta, 2024-07-05:
Thank you all for your efforts on this! I have been through the same issue. It seems the unreal engine can be successfully launched after I reboot my computer. I am working with 0.9.15 and ubuntu 20.04.

closed 2024-09-04 by Blyron

CaMi1le, 2024-10-21:
> Yes, it will output the following typical error if I switch to `Town10HD_Opt`:
> 
> ```
> [2024.07.01-08.55.37:562][353]LogShaderCompilers: Display: Worker (13/29): shaders left to compile 738
> [2024.07.01-08.55.37:562][353]LogStaticMesh: Building static mesh SM_Skyscraper_P03_Merged...
> [2024.07.01-08.55.38:947][353]LogCore: Error: appError called: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
> Array index out of bounds: 1059070274 from an array of size 1055160870
> 
> [2024.07.01-08.55.38:963][353]LogCore: === Critical error: ===
> Unhandled Exception: SIGSEGV: invalid attempt to write memory at address 0x0000000000000003
> 
> [2024.07.01-08.55.38:963][353]LogCore: Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] 
> Array index out of bounds: 1059070274 from an array of size 1055160870
> 
> 
> 0x00007ff92660625e libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
> 0x00007ff92685156b libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/marmot/RUANJIAN/program/UnrealEngine4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
> ```
> 
> The error I met when launching `Town06` is also very strange. Hope it can be solved someday, and the above errors.
> 
> ```
> [2024.07.01-08.19.26:817][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Rendering_and_Lighting_components) is flushing async loading
> [2024.07.01-08.19.26:887][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Buildings) is flushing async loading
> [2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for BoolProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:BlockSpawn_12_4FEC57D947F4B49EA11F31A19F18EB48 Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
> [2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for ObjectProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:Item_16_6D77EDC54B56B5883D9DC7809D6C704A Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
> [2024.07.01-08.19.27:187][  0]LoadErrors: Error: /Game/Carla/Blueprints/LevelDesign/BuildingMaster : Failed import for StructProperty /Game/Carla/Blueprints/LevelDesign/BuildingSpawnableItems.BuildingSpawnableItems:Transform_5_626174554EFC313F198DA2AED94C7F1B Referenced by export Function /Game/Carla/Blueprints/LevelDesign/BuildingMaster.BuildingMaster_C:PlaceItems
> [2024.07.01-08.19.27:236][  0]Warning:  applied to a non-external Texture2D sampler. This may work by chance on some platforms but is not portable. Please change sampler type to 'External'. Parameter 'EmissiveMap' (slot 1) in material 'M_BuildingMaterial'
> [2024.07.01-08.19.27:236][  0]LogStreaming: Display: ULevelStreaming::RequestLevel(/Game/Carla/Maps/Sublevels/Town05_Opt/T05_Foliage) is flushing async loading
> ```
> 
> Just cannot understand why the asset files has problems. No matter I import them through the modified asset link in the latest `ContentVersions.txt` in the `dev` branch and put them to the right place, or I run `Update.sh` after modify the wrong link in it, or I import them through ` git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla`

so how you fix this problem, I change to 05, 06 , both have this problem

HONGcalmJIN, 2024-10-23:
Not yet, maybe the precompiled one is enough

CaMi1le, 2024-11-17:
> Not yet, maybe the precompiled one is enough

thanks, let me know if you fix it