carla-simulator/carla #7862: [dev]  `GLIBCXX_3.4.30'  not found when using Python 3.10 egg and pygame

Daraan, 2024-06-27:
I am using the current `dev` version of the simulator and the PythonAPI.

Since a few weeks I sometimes run into the error:

```py
ImportError: /home/miniconda3/envs/leaderboard/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found
(required by /home/.cache/Python-Eggs/carla-0.9.15-py3.10-linux-x86_64.egg-tmp/carla/libcarla.cpython-310-x86_64-linux-gnu.so)
```

### Code to reproduce(?) with python 3.10

Important: Add the carla.egg to the `PYTHONPATH` 

```sh
>>> PYTHONPATH=$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.15-py3.10-linux-x86_64.egg:$PYTHONPATH python
```
```python
import pygame
import carla
```

**I could not reproduce this with the Python 3.7 egg file.**

### Temporary Solution:

Carla needs to be imported *before* pygame.

```sh
>>> PYTHONPATH=$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.15-py3.10-linux-x86_64.egg:$PYTHONPATH python
```
```python
import carla
import pygame
```

PatrickPromitzer, 2024-06-27:
Hi,
it would be interesting to know if the same problem appears if you install the Carla package with pip.
```console
pip install $CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.15-cp310-cp310-linux-x86_64.whl
```
or
```console
path/to/python -m pip install $CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.15-cp310-cp310-linux-x86_64.whl
```

Daraan, 2024-06-27:
Thank you for the reply. I realized some things:

* yes installing the wheel file and verified that it access not the egg file throws the same error
* I made a new conda python3.10 envionment, installed the wheel and pygame: same error

* I used a different environment, **which uses the same egg file**, no error
  * pygame 2.6.0 was installed in the crashing environments, and the working one 2.5.2, however downgrading it did not fix it.

I'll insert the `pip freeze` outputs below in a few minutes

PatrickPromitzer, 2024-06-28:
I tested it with the pip package and it worked.
After testing it with a .whl file, I got the same error message, but the order of carla/pygame didn't matter.

Try this commands
```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install --only-upgrade libstdc++6
```
For me it worked after upgrading this package