carla-simulator/carla #7877: "Import Carla" leading to module not found for Python3.12.24

AlexMiamiACN, 2024-06-27:
CARLA version: 0.9.13
Platform/OS: Windows
Problem you have experienced: Trying to import .xodr file into carla using Vscode script, but using 

**import carla

def main():
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0)**

and returning error
 **"Traceback (most recent call last):
  File "c:\Carla\carla-dev\Build\osm2odr-source\src\TestFile.py", line 13, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'"** 

I've tried to see if Carla is not compatible with the Python I am running, which is 3.12.24, and have downgraded to 3.8.10. However, the import carla still does not work and it says the module not recognized. Any help would be appreciated!

Daraan, 2024-06-29:
For carla 0.9.13 you probably need to go back to Python 3.7  and use the .egg / wheel provided within that installation.

On PyPy you have these options: https://pypi.org/project/carla/0.9.13/#files

---

Currently 0.9.15: Python 3.7-3.10 are the versions that are mostly supported. Personally I haven't tried going higher.

https://pypi.org/project/carla/#files

GoodarzMehr, 2024-06-29:
You probably have to add the CARLA .egg/.whl file path to the $PYTHONPATH environment variable for Python to know where to find it.