carla-simulator/carla #7884: How to give path to carla server if build from source

SExpert12, 2024-06-29:
CARLA version:0.9.14
Platform/OS:Ubuntu
Problem you have experienced: Launch Carla server

Hi,
I am reproducing results for RL algorithms from this repo -

https://github.com/yanlai00/RL-Carla/issues/new

But the problem is I am using "make launch" to start the carla server, so I don't understand how to write this line in my code.        

server = subprocess.Popen(f'DISPLAY= ' + str(os.path.join(os.environ.get("home/Downloads/CARLA14"), "CarlaUE4.sh")) + f' -opengl '+ f' -carla-rpc-port={port}' + f" -quality-level=Epic ", stdout=None, stderr=subprocess.STDOUT, preexec_fn=os.setsid, env=env, shell=True)

I tried to give the path of CarlaGear server but it is not working the one which is suggested by @felipecode to collect the data.

/home/.local/lib/python3.8/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:572: DeprecationWarning: `np.object` is a deprecated alias for the builtin `object`. To silence this warning, use `object` by itself. Doing this will not modify any behavior and is safe. 
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  (np.object, string),
/home/.local/lib/python3.8/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:573: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  (np.bool, bool),
/home/.local/lib/python3.8/site-packages/tensorboard/util/tensor_util.py:111: DeprecationWarning: `np.object` is a deprecated alias for the builtin `object`. To silence this warning, use `object` by itself. Doing this will not modify any behavior and is safe. 
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  np.object: SlowAppendObjectArrayToTensorProto,
/home/.local/lib/python3.8/site-packages/tensorboard/util/tensor_util.py:112: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  np.bool: SlowAppendBoolArrayToTensorProto,
2024-06-29 07:37:26.688240: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory
2024-06-29 07:37:26.688263: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
/home/.local/lib/python3.8/site-packages/h5py/__init__.py:46: DeprecationWarning: `np.typeDict` is a deprecated alias for `np.sctypeDict`.
  from ._conv import register_converters as _register_converters
/home/.local/lib/python3.8/site-packages/scipy/fft/__init__.py:97: DeprecationWarning: The module numpy.dual is deprecated.  Instead of using dual, use the functions directly from numpy or scipy.
  from numpy.dual import register_func
/home/.local/lib/python3.8/site-packages/scipy/sparse/sputils.py:17: DeprecationWarning: `np.typeDict` is a deprecated alias for `np.sctypeDict`.
  supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
/home/.local/lib/python3.8/site-packages/flatbuffers/compat.py:19: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/home/.local/lib/python3.8/site-packages/matplotlib/__init__.py:152: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(module.__version__) < minver:
/home/RL-Carla-main/Carla-RL/yanlai/lib/python3.8/site-packages/setuptools/_distutils/version.py:337: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  other = LooseVersion(other)
/home/.local/lib/python3.8/site-packages/pygame/pkgdata.py:25: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import resource_stream, resource_exists
/home/RL-Carla-main/Carla-RL/yanlai/lib/python3.8/site-packages/pkg_resources/__init__.py:2976: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
/home/RL-Carla-main/Carla-RL/yanlai/lib/python3.8/site-packages/pkg_resources/__init__.py:2976: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('mpl_toolkits')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
/home/.local/lib/python3.8/site-packages/scipy/special/orthogonal.py:81: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  from numpy import (exp, inf, pi, sqrt, floor, sin, cos, around, int,
Traceback (most recent call last):
  File "train_sac.py", line 79, in <module>
    main(model_name, load_model, town, fps, im_width, im_height, repeat_action, start_transform_type, sensors, 
  File "train_sac.py", line 15, in main
    env = CarlaEnv(town, fps, im_width, im_height, repeat_action, start_transform_type, sensors,
  File "/home/RL-Carla-main/Carla-RL/carla_env.py", line 29, in __init__
    self.client, self.world, self.frame, self.server = setup(town=town, fps=fps, client_timeout=timeout)
  File "/home/RL-Carla-main/Carla-RL/setup.py", line 66, in setup
    server = subprocess.Popen(f'DISPLAY= ' + str(os.path.join(os.environ.get("home/Downloads/CARLA14"), "CarlaUE4.sh")) + f' -opengl '+ f' -carla-rpc-port={port}' + f" -quality-level=Epic ", stdout=None, stderr=subprocess.STDOUT, preexec_fn=os.setsid, env=env, shell=True)
  File "/usr/lib/python3.8/posixpath.py", line 76, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType

How to solve this?

GoodarzMehr, 2024-06-29:
If I understand correctly that line launches the server, if you are launching the server using `make launch` (and presumably hitting play in UE4) can't you just remove that line?

Alternatively, you could build a packaged version of CARLA using `make package` (it'll be built in the `Dist` folder) and then use that.

SExpert12, 2024-06-30:
It is a variable taken as a argument in many file like this one file.

import carla
import numpy as np
import transforms3d.euler
from absl import logging

logging.set_verbosity(logging.DEBUG)

def setup(
    town: str,
    fps: int = 20,
    server_timestop: float = 30.0,
    client_timeout: float = 20.0,
    num_max_restarts: int = 10,
):
    """Returns the `CARLA` `server`, `client` and `world`.

    Args:
        town: The `CARLA` town identifier.
        fps: The frequency (in Hz) of the simulation.
        server_timestop: The time interval between spawing the server
        and resuming program.
        client_timeout: The time interval before stopping
        the search for the carla server.
        num_max_restarts: Number of attempts to connect to the server.

    Returns:
        client: The `CARLA` client.
        world: The `CARLA` world.
        frame: The synchronous simulation time step ID.
        server: The `CARLA` server.
    """
    assert town in ("Town01", "Town02", "Town03", "Town04", "Town05")

    # The attempts counter.
    attempts = 0

    while attempts < num_max_restarts:
        logging.debug("{} out of {} attempts to setup the CARLA simulator".format(
            attempts + 1, num_max_restarts))

        # Random assignment of port.
        port = np.random.randint(2000, 3000)

        # Start CARLA server.
        env = os.environ.copy()
        env["SDL_VIDEODRIVER"] = "offscreen"
        env["SDL_HINT_CUDA_DEVICE"] = "0"
        logging.debug("Inits a CARLA server at port={}".format(port))
        server = subprocess.Popen(f'DISPLAY= ' + str(os.path.join(os.environ.get("CARLA_ROOT"), "CarlaUE4.sh")) + f' -opengl '+ f' -carla-rpc-port={port}' + f" -quality-level=Epic ", stdout=None, stderr=subprocess.STDOUT, preexec_fn=os.setsid, env=env, shell=True)
        atexit.register(os.killpg, server.pid, signal.SIGKILL)
        time.sleep(server_timestop)

        # Connect client.
        logging.debug("Connects a CARLA client at port={}".format(port))
        try:
            client = carla.Client("localhost", port)  # pylint: disable=no-member
            client.set_timeout(client_timeout)
            client.load_world(map_name=town)
            world = client.get_world()
            world.set_weather(carla.WeatherParameters.ClearNoon)  # pylint: disable=no-member
            frame = world.apply_settings(
                carla.WorldSettings(  # pylint: disable=no-member
                    synchronous_mode=True,
                    fixed_delta_seconds=1.0 / fps,
                ))
            logging.debug("Server version: {}".format(client.get_server_version()))
            logging.debug("Client version: {}".format(client.get_client_version()))
            return client, world, frame, server
        except RuntimeError as msg:
            logging.debug(msg)
            attempts += 1
            logging.debug("Stopping CARLA server at port={}".format(port))
            os.killpg(server.pid, signal.SIGKILL)
            atexit.unregister(lambda: os.killpg(server.pid, signal.SIGKILL))

    logging.debug(
        "Failed to connect to CARLA after {} attempts".format(num_max_restarts))
    sys.exit()

So as per my understanding I cannot remove that line. 

Second option may work out. But would it launch the Carla server like CarlaUE4.sh command? Do you know that?

Thanks for your help.

GoodarzMehr, 2024-06-30:
The `server` object returned by `Popen` is a system process. If you are using `make launch` I think you can change your code to get it from the system process list using something like
``` Python
processes = [p for p in psutil.process_iter() if 'carla' in p.name().lower()]
```

The second option generates a packaged version of CARLA, similar to the ones available for download. It includes a `CarlaUE4.sh` file that launches the server.

SExpert12, 2024-07-01:
Okay. Thanks. Let me try and I get back to you.

SExpert12, 2024-07-05:
Hi,
I use the packaged version of carla and from .sh file I am able to open a server. So I change line like this:

server = subprocess.Popen(f'DISPLAY= ' + str(os.path.join("/home/local/carla0914/","CarlaUE4.sh")) + f' -opengl '+ f' -carla-rpc-port={port}' + f" -quality-level=Epic ", stdout=None, stderr=subprocess.STDOUT, preexec_fn=os.setsid, env=env, shell=True)

It started like this:
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.

But then I got this error.

Error in atexit._run_exitfuncs:
ProcessLookupError: [Errno 3] No such process

Where I am making mistake? How to resolve this now?

GoodarzMehr, 2024-07-05:
Hi,

I'm not really sure. At least the beginning (`4.26.2-0+++UE4+Release-4.26 522 0 Disabling core dumps.`) is what you'd expect. [This](https://stackoverflow.com/questions/32218427/why-does-killing-this-subprocess-raise-a-processlookuperror) is helpful in explaining what the error means, and you may need to do some more debugging around the `atexit` line(s) to figure out the root cause of the problem. I suspect for whatever reason the `atexit` after the `server` line cannot find `server.pid`, but I may be wrong.

SExpert12, 2024-07-06:
Okay.

Thanks

closed 2025-11-05 by Blyron