carla-simulator/carla #7889: Rework of Semantic Segmentation sensor

Blyron, 2024-07-01:
Our way to label actors is too "hardcoded" by having to follow a folder structure. 
Look at  crp::CityObjectLabel GetLabelByPath (by the way this shouldn't be a template, it should be an UObject parameter

We need to remove this. The 1st approach should be using Material IDs, for this we will need to create master material for each type of object. Art and Core will need to work closely.

closed 2024-10-18 by Blyron

utkuoguzman, 2025-02-01:
This isn't done, why closed as done?

Blyron, 2025-02-02:
Is done; The work has ben done for ue5

utkuoguzman, 2025-02-02:
But it still uses the meshes inside the folder structure for the semantic. UE5 version don't return anything with material ID. Am I wrong? Is there any explanation somewhere?

Blyron, 2025-02-02:
Yeah and for the moment it will use. The rework was to use a render pass instead of mesh proxys to make it cheaper. We are not gonna use material ids, for now