carla-simulator/carla #7895: Crash when use 4 cameras simultaneously but okay when using 2

LeonardMendicantBias, 2024-07-02:
Dear Carla developers, contributors, and users.

I am getting familar with Carla (version 0.9.15) to generate data. In my scenario, I attach (4) cameras on static places (i.e., on a traffic light). Since I need to obtain the exact locations for the cameras, I built Carla on Windows and played around with the UE 4 editor. I am able to place the cameras at the locations I wanted. Following the instruction from Carla, I spawn cameras and save the image via the callback function. The problem is that the simulation provide data if there are at most 2 cameras. If there are 3 cameras, the simulation starts, generate a single image for each spawned cameras, and crash with the error: "time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000". This confused me because the error is simply not true. The simulation did generate 1 image.

I attached my code for error reproduction. It is adapted from generate_traffic.py.

import glob
import os
import sys
import time
import copy
from functools import partial

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import argparse
import logging
from numpy import random
import numpy as np

import carla
from carla import VehicleLightState as vls
from pascal_voc_writer import Writer

def get_actor_blueprints(world, filter, generation):
    bps = world.get_blueprint_library().filter(filter)
    if generation.lower() == "all":
        return bps
    # If the filter returns only one bp, we assume that this one needed
    # and therefore, we ignore the generation
    if len(bps) == 1:
        return bps
    try:
        int_generation = int(generation)
        # Check if generation is in available generations
        if int_generation in [1, 2, 3]:
            bps = [x for x in bps if int(x.get_attribute('generation')) == int_generation]
            return bps
        else:
            print("   Warning! Actor Generation is not valid. No actor will be spawned.")
            return []
    except:
        print("   Warning! Actor Generation is not valid. No actor will be spawned.")
        return []

def spawn_vehicles(world, client, args, synchronous_master=False):
    traffic_manager = client.get_trafficmanager(args.tm_port)
    vehicles_list = []
    blueprints = get_actor_blueprints(world, args.filterv, args.generationv)
    if not blueprints:
        raise ValueError("Couldn't find any vehicles with the specified filters")
    if args.safe:
        blueprints = [x for x in blueprints if x.get_attribute('base_type') == 'car']
    blueprints = sorted(blueprints, key=lambda bp: bp.id)
    spawn_points = world.get_map().get_spawn_points()
    number_of_spawn_points = len(spawn_points)
    if args.number_of_vehicles < number_of_spawn_points:
        random.shuffle(spawn_points)
    elif args.number_of_vehicles > number_of_spawn_points:
        msg = 'requested %d vehicles, but could only find %d spawn points'
        logging.warning(msg, args.number_of_vehicles, number_of_spawn_points)
        args.number_of_vehicles = number_of_spawn_points
    # @todo cannot import these directly.
    SpawnActor = carla.command.SpawnActor
[... 221 lines cut ...]
    except Exception as error:
        print("error")
        print(error)
    finally:
        if not args.asynch and synchronous_master:
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.no_rendering_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)
        time.sleep(2)
        print("\ndestroying %d vehicles" % len(vehicles_list))
        client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])
        print("\ndestroying %d camera" % len(camera_list))
        for camera in camera_list:
            camera.destroy()
        time.sleep(2)  # finishing saving the last images

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print("\ndone.")

STARKEY13, 2024-07-02:
The default timeout of 10 seconds might not be sufficient when using multiple cameras. Increase the timeout to give the simulator more time to process the requests

LeonardMendicantBias, 2024-07-03:
Thank you for the quick response. Unfortunately, the error persists. I modified the following.
`client.set_timeout(50.0)`
The simulations simply wait 50 seconds before showing the same error.

One more "anomaly", I switch the simulation to async mode and everything works. However, this would not do since I need the synchronous mode.

PatrickPromitzer, 2024-07-04:
Hi,
the "time-out" error happens if Carla didn't answered.
It could be the connection, Carla crashed, Carla took too long, or everything else that could prevent a answer from the server in the timeout limit.

You can try to add some debug messages to see where the script got an error.
Did the Server still run after the time-out error?

btw. the code is hard to read because of the formatting. You can use 
```
   ```python
    code
    ` ` `
```
to keep the format. (without space)

LeonardMendicantBias, 2024-07-04:
@PatrickPromitzer Thank you for the coding quote tips. I did not know about triple ```. The github's editing tool only shows single `. I edited the code for future reading.

The UE 4 editor is running while the script is waiting for a response! After the time-out message, I can still move around in the UE 4 editor (using WASD). However, If I start the script again, the script returns the time-out message without the initial images. I think that using 4 cameras crashes the Carla server and prevents later connections. But since the UE 4 editor is still "functional", I really don't know where to debug. Do you have any suggestion where should I watch for a possible bug?

p/s: I start the UE 4 editor in the folder "./Unreal/CarlaUE4/CarlaUE4.uproject" after I finished build Carla from source.

PatrickPromitzer, 2024-07-04:
Normally, if the server crashes the unreal editor crashes, too. (maybe only on Ubuntu)
You could search if an old instance is running and close everything Carla related.
It is not a solution, but maybe something in the background is creating a problem.

You could look if your firewall is blocking something.

You could try making a package and test it with that package.

One thing that helps for me is adding "world.tick()" in between adding sensors or listeners.

You could try using one blueprint instance for each camera (just to be sure, even if should not make a difference)

You can watch your CPU/GPU usage, too.

For 
```
camera_1.listen(lambda image: image.save_to_disk(f'{args.directory}/camera-1/{image.frame:06d}.png'))
```
you could call an custom function and log the function call.
```python
with open("log_file.txt", "a") as f:    
    f.write("Function call: {0}".format(camera_name))
```

Maybe I come up with more ideas at a later time.

LeonardMendicantBias, 2024-07-04:
@PatrickPromitzer Thank you so much for your suggestions. My firewall has been turned off. I do have a defined function for the callback. 

I don't know what I have changed, but somehow the script works.  I mark this as solved but I really don't know what was the problem.

LeonardMendicantBias, 2024-07-05:
I found the problem. It is the sensor_tick. When I set the value too low (i.e., 0.02 for 50 fps), the simulator time-out while it works well with sensor_tick=1. I guess I need a better computer.

PatrickPromitzer, 2024-07-05:
The normal tickrate is 0.1, and at some point, the documentation said you should not go below 0.05

Try which rate works for you before buying new hardware.

LeonardMendicantBias, 2024-07-05:
Yes, the document states that the simulation should not be less than 10 FPS (0.01) or the physic would get weird. But sensor_tick is the update rate of the camera, not the simulator itself. 

Also, I am able to utilize `sensor_tick=0.02` by adding the image into a global queue in the callback function. Then, after the `world.tick()`, I save the queued images. This makes the simulations not real-time anymore, but that does not matter since I only need the synthetic data.