carla-simulator/carla #7897: Make package including custom map

Ahnhojin1223, 2024-07-03:
CARLA version: 0.9.15
Platform/OS: Ubuntu.20.04

If I create a custom map using Roadrunner, put it in the Imoprt folder, type "make import ARGS="--package=map_name" into command, that custom map is created in Carla. After that, type "make package ARGS="--package=map_name" into command, only the package in that map is created.
By the way, if I use "make package" to create the entire package including the custom map, the generated package does not contain the custom map. Is there a way to fix this???

PatrickPromitzer, 2024-07-04:
Hi,
You need to add the custom Map in the packaging settings.
[image]

With using "make package", Carla is creating a package with the Maps listed in the packaging settings.

Madecu, 2024-07-05:
Hi @Ahnhojin1223,

You can also do a package of your map as an additional map. You can edit /Content/Carla/Config/AdditionalMaps.Package.json and add your map and then run _make package ARGS="--package=AdditionalMaps --clean-intermediate"_

chuanye-Wang, 2024-07-29:
@Madecu The same issue, is there any way to export my custom map individually as an umap, so I can import it to another computer? Thanks

Madecu, 2024-07-30:
Hi @chuanye-Wang,

You can locate the .umap file in the content folder on your disk. However, ensure that all the assets referenced in the map are in the same locations; otherwise, the references will be broken.

chuanye-Wang, 2024-07-30:
Thank you, I solved this issue yesterday, but still have some question out of curious. I followed the guide to create my cusMap.Package.json like this, and it worked.
'''JSON
{
  "maps": [
    {
        "path": "/Game/map_package/Maps/cusMap",
        "name": "cusMap",
        "use_carla_materials": true
      }
  ],
  "props": []
}
'''
But I took a look at Town06_Opt.Package.json, and it was quite complicated:
[image]

My cusMap worked fine, so I just ignored the difference between them yesterday.

closed 2024-09-19 by Ahnhojin1223