carla-simulator/carla #7898: CARLA simulator container using Docker

pennyfea, 2024-07-03:
**CARLA version:** 0.9.15
**Platform/OS:** Windows 11 Home
**Problem You Have Experienced:** When running the CARLA simulator container using Docker, the container exits with the following error:
```
[+] Running 1/0
✔ Container carla-vss-carla-1  Created 0.0s 
Attaching to carla-1
carla-1  | 4.26.2-0+++UE4+Release-4.26 522 0
carla-1  | Disabling core dumps.
carla-1  | sh: 1: xdg-user-dir: not found
carla-1 exited with code 1
```
**What You Expected to Happen:** The container should start and run the CARLA simulator without errors.

**Steps to Reproduce:**

1.  Dockerfile
```
# syntax=docker/dockerfile:1
FROM ubuntu:22.04

# Update and install necessary packages
RUN apt-get -y update && apt-get install -y \
    python3 \
    curl \
    gnupg \
    ca-certificates \
    software-properties-common \
    xdg-user-dirs \
    xdg-utils \
    mesa-utils \
    xserver-xorg \
    libgl1-mesa-glx

RUN xdg-user-dir

# Configure the NVIDIA Container Toolkit repository
RUN curl -fsSL https://nvidia.github.io/libnvidia-container/gpgkey | gpg --dearmor -o /usr/share/keyrings/nvidia-container-toolkit-keyring.gpg \
  && curl -s -L https://nvidia.github.io/libnvidia-container/stable/ubuntu22.04/amd64/libnvidia-container.list | \
    sed 's#deb https://#deb [signed-by=/usr/share/keyrings/nvidia-container-toolkit-keyring.gpg] https://#g' | \
    tee /etc/apt/sources.list.d/nvidia-container-toolkit.list

# Update the packages list and install the NVIDIA Container Toolkit
RUN apt-get update && apt-get install -y nvidia-container-toolkit

# Set up the NVIDIA runtime for Docker
RUN nvidia-ctk runtime configure --runtime=docker
```

2. Compose.yaml
```
services:
  carla:
    image: carlasim/carla:latest
    command: ["bash", "./CarlaUE4.sh", "-opengl"]
    runtime: nvidia
    ports:
     - "2000-2002:2000-2002"
    environment:
      - DISPLAY=${DISPLAY}
      - NVIDIA_VISIBLE_DEVICES=all
      - NVIDIA_DRIVER_CAPABILITIES=all
      - QT_X11_NO_MITSHM=1
      - SDL_VIDEODRIVER=x11
    user: ${UID}
    privileged: true
    network_mode: "host"
    stdin_open: true
    tty: true

```

Apoorvgarg-creator, 2024-07-15:
@pennyfea @joel-mb Is there a solution already here to run Carla Simulator using Docker ?

pennyfea, 2024-07-17:
@joel-mb No, sorry I couldn't figure it out and abandoned this approach.

PatrickPromitzer, 2024-07-26:
Hi,
there are some parts that doesn't look right.

"/tmp/.X11unit" 
u"/usr/share/vulkan/icd.d"
looks like linux paths and not a windows path.

Do you use a x11 server on your windows machine?
If you try to use X11 on widows, you need a software to show a window (or you use "-RenderOffScreen" for no window)

On an ubuntu system, it looks like this
```
sudo apt-get install -y nvidia-docker2
sudo systemctl restart docker
xhost +
sudo docker run --privileged --gpus all --net=host -e DISPLAY=$DISPLAY -e SDL_VIDEODRIVER=x11 -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.13 /bin/bash ./CarlaUE4.sh -vulkan
```

Spyridonidis98, 2025-02-11:
i have the same problem and have found no solution !! i am using wsl on windows

rui-rocha-42, 2025-02-12:
I'm having the exact same issue, it just exits with exit code 1 but without any error

PatrickPromitzer, 2025-02-18:
Please write down everything you tried.

Did you open a command line in the docker container and check if nvidia-smi works?
Did you use "-RenderOffScreen"?
Did you follow the documentation?