carla-simulator/carla #7905: Remove traffic sign

Ikaros520520, 2024-07-04:
CARLA version:0.9.12
Platform/OS:Windows

I removed some traffic signs like traffic lights and speed limit signs in the default map "Town2". 
[image]
[image]

But when I clicked PLAY in UE to start running, I found that the traffic signs appeared again. Just changed the sign.
[image]

I want to actually remove these traffic signs from the map. Can anyone tell me how to do this? Thank you so much!

Blyron, 2024-07-04:
You need to remove the signs and traffic lights from the OpenDrive file, otherwise, you can look and modify this function
ATrafficLightManager::RegisterLightComponentFromOpenDRIVE in Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Traffic\TrafficLightManager.cpp

Ikaros520520, 2024-07-05:
> You need to remove the signs and traffic lights from the OpenDrive file, otherwise, you can look and modify this function ATrafficLightManager::RegisterLightComponentFromOpenDRIVE in Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Traffic\TrafficLightManager.cpp

For Carla and python I'm still learning. Not really sure how to modify .cpp files. I found an easy way. Just move the traffic sign underground.

closed 2024-07-05 by Blyron

CoderXuans, 2024-10-19:
@Ikaros520520 
Hello.Even though I moved them underground and ran the evaluation code for the leaderboard, the traffic lights still appear. I moved the nearby low houses underground, and they disappeared, but when I tried to back up the map data and delete the traffic lights, clicking play caused Unreal Engine to crash.