carla-simulator/carla #7907: Would you help me install u5? (splite dependency error)

ray9583, 2024-07-04:
This is a persistent issue even after a reinstallation. This occurs during Setup.bat attempts.

```
Buiding CARLA...
[0/2] Re-checking globbed directories...
[32/414] Building C object CMakeFiles\sqlite3.dir\_deps\sqlite3-src\shell.c.obj
FAILED: CMakeFiles/sqlite3.dir/_deps/sqlite3-src/shell.c.obj
C:\PROGRA~1\MIB055~1\2022\COMMUN~1\VC\Tools\MSVC\1440~1.338\bin\Hostx64\x64\cl.exe  /nologo -D_CRT_SECURE_NO_WARNINGS -IC:\Users\USER\CarlaUE5\Build\_deps\zlib-src -IC:\Users\USER\CarlaUE5\Build\_deps\zlib-build -IC:\Users\USER\CarlaUE5\Build\_deps\libpng-src -IC:\Users\USER\CarlaUE5\Build\_deps\libpng-build /DWIN32 /D_WINDOWS /O2 /Ob2 /DNDEBUG -std:c11 -MD /wd4005 /showIncludes /FoCMakeFiles\sqlite3.dir\_deps\sqlite3-src\shell.c.obj /FdCMakeFiles\sqlite3.dir\ /FS -c C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(1): warning C4819: The file contains a character that cannot be represented in the current code page (949). Save the file in Unicode format to prevent data loss
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26303): warning C4819: The file contains a character that cannot be represented in the current code page (949). Save the file in Unicode format to prevent data loss
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26338): error C2001: newline in constant
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26339): error C2143: syntax error: missing ';' before 'const'
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26353): error C2065: 'zBom': undeclared identifier
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26353): warning C4047: '=': 'int' differs in levels of indirection from 'const char *'
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26420): error C2065: 'zBom': undeclared identifier
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26420): error C2065: 'zBom': undeclared identifier
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26420): warning C4047: 'function': 'const char *' differs in levels of indirection from 'int'
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26420): warning C4024: 'oPutsUtf8': different types for formal and actual parameter 1
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26433): error C2065: 'zBom': undeclared identifier
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26433): error C2065: 'zBom': undeclared identifier
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26433): warning C4047: 'function': 'const char *' differs in levels of indirection from 'int'
C:\Users\USER\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26433): warning C4024: 'oPutsUtf8': different types for formal and actual parameter 1
[49/414] Building C object CMakeFiles\libsqlite3.dir\_deps\sqlite3-src\sqlite3.c.obj
ninja: build stopped: subcommand failed.

```

Blyron, 2024-07-04:
https://github.com/carla-simulator/carla/issues/7904

ray9583, 2024-07-04:
> #7904

Unfortunately, it hasn't been solved. (I wrote it)

rldiaz14, 2024-07-06:
I did pass the setup.bat in window, check if the unreal editor work, if it work just download another ue5 carla and follow the guide to finish the build. I having issue with assest and that is another problem. That I fixing yet have not resolved.

ray9583, 2024-07-07:
> I did pass the setup.bat in window, check if the unreal editor work, if it work just download another ue5 carla and follow the guide to finish the build. I having issue with assest and that is another problem. That I fixing yet have not resolved.

First of all, thank you for your reply. Are you saying that I should ignore and pass even if an error occurs during the Setup.bat process? If I ignore the error and proceed, what process should I proceed next?

rldiaz14, 2024-07-08:
> First of all, thank you for your reply. Are you saying that I should ignore and pass even if an error occurs during the Setup.bat process? If I ignore the error and proceed, what process should I [...]

no at all, I deleted the version that is download with build and grab another one from this GitHub and download that one, and follow the same step with build and it works fine.

ray9583, 2024-07-11:
I realize, it is encoding problem

closed 2024-07-11 by ray9583

wkd72, 2024-07-19:
how to sovle this encoding problem?

Jinviz, 2024-07-19:
> how to sovle this encoding problem?

Open that file as notepad, save it as "UTF-8(BOM)" and rebuild it! (not just "UTF-8")

Was it helpful?

wkd72, 2024-07-19:
oh, thanks a lot

Jinviz, 2024-07-19:
> oh, thanks a lot

Good luck :)

Gragonfly, 2024-07-31:
hi, I met the same problem. However, when I had changed the way of coding to "UTF-8(BOM",  another problem happened.

[0/2] Re-checking globbed directories...
[1/168] Linking CXX executable _deps\recastnavigation-build\RecastBuilder\RecastBuilder.exe
FAILED: _deps/recastnavigation-build/RecastBuilder/RecastBuilder.exe
C:\Windows\system32\cmd.exe /C "cd . && "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=_deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100226~1.0\x64\mt.exe --manifests  -- C:\PROGRA~1\MIB055~1\2022\COMMUN~1\VC\Tools\MSVC\1440~1.338\bin\Hostx64\x64\link.exe /nologo _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Contrib\fastlz\fastlz.c.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\ChunkyTriMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\InputGeom.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\MeshLoaderObj.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\PerfTimer.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\SampleInterfaces.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample_SoloMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample_TileMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\main.cpp.obj  /out:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.exe /implib:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.lib /pdb:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:windows /SUBSYSTEM:CONSOLE  _deps\recastnavigation-build\DebugUtils\DebugUtils.lib  _deps\recastnavigation-build\Detour\Detour.lib  _deps\recastnavigation-build\DetourCrowd\DetourCrowd.lib  _deps\recastnavigation-build\DetourTileCache\DetourTileCache.lib  _deps\recastnavigation-build\Recast\Recast.lib  _deps\recastnavigation-build\Detour\Detour.lib  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
LINK: command "C:\PROGRA~1\MIB055~1\2022\COMMUN~1\VC\Tools\MSVC\1440~1.338\bin\Hostx64\x64\link.exe /nologo _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Contrib\fastlz\fastlz.c.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\ChunkyTriMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\InputGeom.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\MeshLoaderObj.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\PerfTimer.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\SampleInterfaces.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample_SoloMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\Sample_TileMesh.cpp.obj _deps\recastnavigation-build\RecastBuilder\CMakeFiles\RecastBuilder.dir\Source\main.cpp.obj /out:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.exe /implib:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.lib /pdb:_deps\recastnavigation-build\RecastBuilder\RecastBuilder.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:windows /SUBSYSTEM:CONSOLE _deps\recastnavigation-build\DebugUtils\DebugUtils.lib _deps\recastnavigation-build\Detour\Detour.lib _deps\recastnavigation-build\DetourCrowd\DetourCrowd.lib _deps\recastnavigation-build\DetourTileCache\DetourTileCache.lib _deps\recastnavigation-build\Recast\Recast.lib _deps\recastnavigation-build\Detour\Detour.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST:EMBED,ID=1" failed (exit code 1120) with the following output:
Recast.lib(RecastMeshDetail.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DetourCrowd.lib(DetourLocalBoundary.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DetourCrowd.lib(DetourPathCorridor.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Recast.lib(Recast.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Recast.lib(RecastRasterization.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Detour.lib(DetourNavMesh.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Detour.lib(DetourNavMeshQuery.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DetourCrowd.lib(DetourCrowd.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DetourCrowd.lib(DetourObstacleAvoidance.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Sample_SoloMesh.cpp.obj : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
Sample_TileMesh.cpp.obj : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DebugUtils.lib(DebugDraw.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
DebugUtils.lib(RecastDump.cpp.obj) : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
fastlz.c.obj : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
InputGeom.cpp.obj : error LNK2001: 无法解析的外部符号 __GSHandlerCheck
[... cut ...]
C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.39.33519\lib\x86\msvcprt.lib : warning LNK4272:库计算机类型“x86”与目标计算机类型“x64”冲突
_deps\recastnavigation-build\RecastBuilder\RecastBuilder.exe : fatal error LNK1120: 59 个无法解析的外部命令
[9/168] Building C object CMakeFiles\sqlite3.dir\_deps\sqlite3-src\shell.c.obj
D:\fgm\carlaUE5\CarlaUE5\Build\_deps\sqlite3-src\shell.c(26338): warning C4566: 由通用字符名称“\uFEFF”表示的字符不能在当前代码页(936)中表示出来
[17/168] Building CXX object LibCarla\CMakeFiles\carla-server.dir\source\carla\streaming\detail\tcp\Server.cpp.obj^C
D:\fgm\carlaUE5\CarlaUE5>

does anyone have seen this ?

Jinviz, 2024-07-31:
Delete the build folder and try to build again. If it's Windows 11, uninstall Windows 10 SDK (10.0.18362) and proceed

donghaiwang, 2024-12-25:
I have same build error in recent [ue5-dev](https://github.com/carla-simulator/carla/commit/ada75f920642e18cace0a9f85ecf9d4077ddb531) .

Platform: Windows 10.