carla-simulator/carla #7910: Problems with shared libraries when running carla

Gallifreey, 2024-07-04:
I want to use carla on a cloud server. But I have a problem on running carla. When I tried to run 
` ./CarlaUE4.sh` then I got:
 `./CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: error while loading shared libraries: libChronoEngine.so: cannot open shared object file: No such file or directory`.
 It seems that a file is missing ,but I could not install this file(use apt-get install). 
Is that this file is provided by system or I should install in other way?
CARLA version: 0.9.15
Platform/OS: Ubuntu18.04 and Ubuntu 20.04

PatrickPromitzer, 2024-07-05:
Hi,
Do you connect to the cloud server with ssh?
A window will open if you start Carla, and if there is no Monitor output (or you are not directly in front of the computer), Carla will not start with "./Carla.sh"

You can still start Carla without the Window
```terminal
./CarlaUE4.sh  -RenderOffScreen
```

Gallifreey, 2024-07-05:
> Hi, Do you connect to the cloud server with ssh? A window will open if you start Carla, and if there is no Monitor output (or you are not directly in front of the computer), Carla will not start with "./Carla.sh"
> 
> You can still start Carla without the Window
> 
> ```
> ./CarlaUE4.sh  -RenderOffScreen
> ```

> Hi, Do you connect to the cloud server with ssh? A window will open if you start Carla, and if there is no Monitor output (or you are not directly in front of the computer), Carla will not start with "./Carla.sh"
> 
> You can still start Carla without the Window
> 
> ```
> ./CarlaUE4.sh  -RenderOffScreen
> ```
Thank you for your reply!!
I have tried `RenderOffScreen` option and it still have the same error. It seems that it cannot run with `./CarlaUE4.sh` even I give wrong options. The server is running in a docker. Is it possible that the relevant libraries are missing from docker?

PatrickPromitzer, 2024-07-24:
Hi,
did you install the nvidia docker package?
```
NVIDIA Container Toolkit: The NVIDIA Container Toolkit is a library and toolset that exposes NVIDIA graphics
devices to Linux containers. It is designed specifically for Linux containers running on Linux host systems or
within Linux distributions under version 2 of the Windows Subsystem for Linux. Install the nvidia-docker2
 package by following the instructions
https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#installation-guide
```
https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#installation-guide

If you don't use the official docker image from Carla, you can try the official one image.
https://hub.docker.com/r/carlasim/carla/tags