carla-simulator/carla #7914: carla installation: make PythonAPI error

xieran1994, 2024-07-05:
I meet an error when 'make PythonAPI' in Ubuntu 22.04.

Setup.sh: Building rpclib with libc++.
CMake Error at CMakeLists.txt:2 (project):
  Running

   '/usr/local/bin/ninja' '--version'

  failed with:

   Traceback (most recent call last):
    File "/usr/local/bin/ninja", line 15, in <module>
      from pkg_resources import load_entry_point

  ImportError: No module named pkg_resources

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1

GoodarzMehr, 2024-07-06:
`pip install setuptools` may help.

xieran1994, 2024-07-08:
> `pip install setuptools` may help. I recommend following the guide and installing all the prerequisites before running `make PythonAPI`.

yes, I followed the guide and installed setuptool with recommanded version, but it doesn't work. Don't know if it is because i'm using a sub account in linux, python pip doesn't in the right place.

MarcelPiNacy-CVC, 2024-07-26:
Hi,

Which branch of CARLA are you using?

closed 2025-05-07 by MarcelPiNacy-CVC