carla-simulator/carla #7917: Permission denied [system:13 at ../boost-1.80.0-c8-install/include/boost/asio/detail/reactive_socket_service.hpp:161] Malloc error

SExpert12, 2024-07-06:
CARLA version:0.9.14
Platform/OS:Ubuntu
Problem you have experienced:Exception thrown: bind: Permission denied [system:13 at ../boost-1.80.0-c8-install/include/boost/asio/detail/reactive_socket_service.hpp:161]
What you expected to happen:Resolve this issue
Steps to reproduce: Run this file:
https://github.com/praveen-palanisamy/macad-agents/blob/master/src/macad_agents/baseline_agent.py
Hi,
I am working on Multi-agent reinforcement learning in CARLA and I am trying to run this file.

https://github.com/praveen-palanisamy/macad-agents/blob/master/src/macad_agents/baseline_agent.py

But I got this error:
Initializing new Carla server...
Running simulation in single-GPU mode
LowLevelFatalError [File:Unknown] [Line: 136] 
Exception thrown: bind: Permission denied [system:13 at ../boost-1.80.0-c8-install/include/boost/asio/detail/reactive_socket_service.hpp:161]
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Segmentation fault (core dumped)

Server got started and crashed suddenly.
How to resolve this.

Apart from this file server is working fine.

PatrickPromitzer, 2024-07-31:
Hi,
try to run
```
sudo chmod 755 /path/to/carla
```
to make sure the permissions are right.

If that didn't work, you can look if the default Carla ports are used.
```
sudo lsof -i:2000
sudo lsof -i:2001
sudo lsof -i:2002
```

kicoi, 2025-01-03:
Hi, I'm facing this issue too, the Carla version that I've installed is 0.9.15 on Ubuntu 22.04, I've tried to fix the permission issue with carla, and checked that there's no current process on carla ports suggested by @PatrickPromitzer , did you solve this issue @SExpert12? Or is there any other suggestions?

The error message:
`Initializing new Carla server...
Running simulation in single-GPU mode
MESA: warning: Driver does not support the 0x7d67 PCI ID.
MESA: warning: Driver does not support the 0x7d67 PCI ID.
LowLevelFatalError [File:Unknown] [Line: 136] 
Exception thrown: bind: Permission denied [system:13 at /home/jenkins/workspace/carla_0.9.15/Build/boost-1.80.0-c10-install/include/boost/asio/detail/reactive_socket_service.hpp:161:33 in function 'bind']
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Segmentation fault (core dumped)
`

SExpert12, 2025-01-23:
> Hi, try to run
> 
> ```
> sudo chmod 755 /path/to/carla
> ```
> 
> to make sure the permissions are right.
> 
> If that didn't work, you can look if the default Carla ports are used.
> 
> ```
> sudo lsof -i:2000
> sudo lsof -i:2001
> sudo lsof -i:2002
> ```

Thanks for your answer. I am extremely sorry for this late reply

SExpert12, 2025-01-23:
@kicoi 
Hi, You need to try installation of packages again and again. This is because of package mismatch. Eventually you will get rid of this.

Best wishes