carla-simulator/carla #7921: Excuse me, my om2odr source package keeps reporting errors, please help me solve it!

1019945975, 2024-07-08:
IMPORTANT

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in G:\Carla\carla-dev\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA Latest:
   You can download the assets from here:

       https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz

   Unzip it in the "G:\Carla\carla-dev\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in G:\Carla\carla-dev\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server --client --generator ""
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22631.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done (0.0s)
-- Generating done (0.3s)
-- Build files have been written to: G:/Carla/carla-dev/Build/libcarla-visualstudio
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 16.11.2+f32259642
版权所有(C) Microsoft Corporation。保留所有权利。

  Checking Build System
  carla_server.vcxproj -> G:\Carla\carla-dev\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> G:\Carla\carla-dev\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22631.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done (0.0s)
-- Generating done (0.2s)
-- Build files have been written to: G:/Carla/carla-dev/Build/libcarla-visualstudio
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 16.11.2+f32259642
版权所有(C) Microsoft Corporation。保留所有权利。

  Checking Build System
  carla_client.vcxproj -> G:\Carla\carla-dev\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client.lib
  carla_client_debug.vcxproj -> G:\Carla\carla-dev\Build\libcarla-visualstudio\LibCarla\cmake\client\Release\carla_client_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "G:\Carla\carla-dev\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "G:\Carla\carla-dev\PythonAPI\carla\dependencies\"!
-[BuildOSM2ODR]: [Batch params]: --generator "" --build --all
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  196M    0  196M    0     0  4893k      0 --:--:--  0:00:41 --:--:-- 8778k
tar: This does not look like a tar archive
tar: Skipping to next header
tar: Exiting with failure status due to previous errors
系统找不到指定的文件。
CMake Warning:
  Ignoring extra path from command line:

   "G:\Carla\carla-dev\Build\osm2odr-source\"

CMake Error: The source directory "G:/Carla/carla-dev/Build/osm2odr-visualstudio/x64" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
系统找不到指定的文件。
系统找不到指定的文件。
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "G:\Carla\carla-dev\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
E:\python\Lib\site-packages\setuptools\_distutils\cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running install_lib
running build_py
copying source\carla\command.py -> build\lib.win-amd64-cpython-312\carla
copying source\carla\__init__.py -> build\lib.win-amd64-cpython-312\carla
running build_ext
building 'carla.libcarla' extension
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -IE:\python\include -IE:\python\Include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-cpython-312\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的 字符。请将该文件保存为 Unicode 格式以防止数据丢失
G:\Carla\carla-dev\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: 无法打开包括文件: “OSM2ODR.h”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "G:\Carla\carla-dev\PythonAPI\carla\dist"!

TingzheYu, 2024-08-02:
you can get the solution in this link [https://github.com/carla-simulator/carla/issues/8000](url)

angian00, 2024-08-21:
it is using a version of tar.exe which does not support zip format. i had to edit BuildOSM2ODR.bat replacing the line 

`tar -xf OSM2ODR.zip`

with

`C:\Windows\System32\tar.exe -xf OSM2ODR.zip`