carla-simulator/carla #7928: The

lingyixia, 2024-07-09:


lingyixia, 2024-07-09:
Carla: 0.9.15
platform:Ubuntu 20.04.6 LTS
In town15, with script https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/automatic_control.py, manally setting start at carla.Transform(carla.Location(x=356.35000000, y=130.00000000, z=120.00000000), carla.Rotation(yaw=-90, pitch=0, roll=0))  to destination:carla.Location(x=359.26000000, y=128.72000000, z=120.00000000), the vehicle collided with a roundabout, it seemed that the vehicles did not align with the waypoints built by global planner. It there a method to ensure an entire travel ?
[image]

joel-mb, 2024-07-12:
Hi @lingyixia,
Seems that the vehicle controller is not doing a good work in this situation. The `automatic_control` script uses the `agents` library to control the vehicle (https://github.com/carla-simulator/carla/tree/dev/PythonAPI/carla/agents). It implements a quite simple controller and may fail for some vehicles in some turns (such as small roundabouts). You may want to fine-tune some of the internal parameters of the agent's controller for your use-case.

EmmettJZK, 2024-07-29:
HI @lingyixia ，
I haven't used version 0.9.15 of CARLA. However, I can help you analyze the possible reasons from a dynamics perspective: CARLA uses `PhysX` for dynamics simulation, and the official tuning of dynamics in CARLA isn't very good, especially in terms of parameters like turning or **_lateral stiffness_**. If your vehicle has "physics simulation" enabled, meaning it uses `PhysX` dynamics, it is possible that during turns on the planned path, the vehicle experiences **_lateral slipping_**. Therefore, different vehicles may have different execution results for the same routing planning.

我没有使用过0.9.15版本的CARLA。但是我可以从动力学的角度帮你分析一下可能的原因：CARLA使用的是 PhysX动力学，CARLA官方对动力学的调优不是很好，尤其是转弯或者说横向刚度这个参数的设置上。如果你的车辆开了“物理模拟”，即使用了PhysX动力学，则有可能是在规划的路径上，转弯时车辆发生了横向滑移。所以可能不同车辆，同样的路径规划效果会有不同的执行结果。

我加入了一些CARLA交流群，可以拉你加入；也接受有偿咨询。感兴趣的可以加我微信:1824729026