carla-simulator/carla #7929: the speed limits in Town04

ccconquer, 2024-07-09:
CARLA version:  0.9.15
Platform/OS: ubuntu
Problem you have experienced: Regarding the speed limits in Town04, why are the speed limits so high? There are no speed limits for roads after Road_ID52. Is there an issue?
Code: 
xodr_content = self.map.to_opendrive()
root = ET.fromstring(xodr_content)
speed_limit = None
for road in root.iter('road'):
    print(f"Road ID: {road.attrib['id']}, Name: {road.attrib['name']}, Length: {road.attrib['length']}")
    for speed in road.iter('speed'):
        speed_limit = float(speed.attrib['max']) * 3.6  # 转换为 km/h
        print(f"Speed limit: {speed_limit} km/h")
 Result:
[image]

GoodarzMehr, 2024-07-09:
For the roads that have a speed limit, I think the values are already in km/h and not in m/s, so you get the wrong values when multiplying by 3.6.

RodrigoVillarias, 2024-07-12:
Hello, as GoodarzMehr says, the speeds in Carla are applied in km/h and not in m/s, by changing this data the speeds should match well

[image]
[image]
[image]

RodrigoVillarias, 2024-07-19:
Hello, this solved your question? the issue will be closed, in any case if something happens feel free to reopen the issue

closed 2024-07-19 by RodrigoVillarias