carla-simulator/carla #7930: Suggestions for Improving Visual Appearance of Vehicle Lights

GoodarzMehr, 2024-07-09:
Hi,

I have a few suggestions for enhancing the visual appearance of vehicle lights in some small ways based on my recent experience with adding new cars to CARLA.

1) Some cars (especially American-made ones) have red rear blinkers instead of amber ones, but this is not currently supported by CARLA. My solution to this was to split the indicator lights section of the UV map in half (similar to the fog and position lights), as seen below, and update the light material (and the actor blueprint) in UE4 by splitting the blinkers in two for front and back, similar to what is done for the fog/position lights.

[image]

The end result is something similar to the images below. I've also attached the UE material in case anyone is interested.

[image]
[image]
[image]

[M_VehicleLights.zip](https://github.com/user-attachments/files/16138558/M_VehicleLights.zip)

2) A lot of cars use the same bulb for several light types, for instance low/high beam and brake/position, but my understanding is that in CARLA's vehicle models a single mesh is separated across different light types in the UV map. What I've found to be a bit more realistic is to duplicate the light mesh (with the duplicate translated across the longitudinal axis of the car by a very samll amount) and assign each mesh to a separate light type in the UV map, as shown in the image below where rear position/fog/break lights are all duplicates of the same mesh.

[image]

It ends up looking something like this (first image only position, second one position+brake).

[image]
[image]

gsmario, 2024-07-12:
Hello!
Thank you very much for your suggestion; we will take it into account for future vehicles.

I have a question regarding the overlapping geometry planes: is the opacity affecting how the ones behind are seen? Did you have to adjust the light intensity to make them look better?

That's all. Thank you for your time!

GoodarzMehr, 2024-07-12:
You are very welcome!

To answer your question, yes, opacity does impact how the lights are seen, but in my experience this has really only been the case for rear lights, since the multi-layer front lights are usually the low/high beam ones that are bright enough that changing the opacity doesn't make a lot of difference (unless you use a very high or low value). For the rear lights, my rule of thumb has been to set the opacity to about 1/(max number of layers) and play around a bit with the intensities, though you can obviously have materials with different opacity for each layer.

I'll use the 2016 Mustang I added as an example of a somewhat extreme case, since the rear lights have four layers (position/fog/brake/blinker from back to front) I've set the opacity to 0.24. These are the light intensities I use:
[image]

Both blinkers + position + fog + low beam:
[image]

Both blinkers + position + fog + low beam + high beam:
[image]

High beam only:
[image]

No lights:
[image]

Brake only:
[image]

Left blinker only:
[image]

Left blinker + position:
[image]

Left blinker + position + fog:
[image]

Left blinker + position + fog + brake:
[image]

Left blinker + position + brake:
[image]

BTW I've been working on adding lights to CARLA's older vehicles, if you'd like I can send my content library over in case you'd want to use any of it.

gsmario, 2024-07-19:
Wow thanks! really nice work!
Feel free to create a pull request with all these changes, we can have a look and see if we can merge it!
Thanks again :)

closed 2024-07-19 by GoodarzMehr

GoodarzMehr, 2024-09-07:
Hi @gsmario 

I've been looking at submitting my changes as a pull request to the content repo, but Bitbucket seems to have a 5 GB cap on repo size. Given the size of the content library I can't push my local changes to my fork to then create a pull request. Is there any other way of doing this? I can upload it to GDrive along with a changelog if that works for you guys.