carla-simulator/carla #7938: Missing Instance segmentation material

Bvallon-sl, 2024-07-10:
CARLA version: UE5.3 Dev
Platform/OS: Win10
Problem you have experienced:
I wanted to try the new Instance segmentation implementation added by #7935 but the InstanceSegmentation material is missing.

I'm facing this error 
[image]
and indeed the material is missing from the project.

Many thanks for considering my request.

Best regards,

Blyron, 2024-07-19:
Please, update content repository. 
For that go to Carla_Root/Unreal/CarlaUnreal/Content/Carla/ 
and put a git pull there, it may ask for a user and token of bitbucket

Bvallon-sl, 2024-07-19:
Ok, I missed that part. Thanks for the help!

closed 2024-07-19 by Bvallon-sl