carla-simulator/carla #7939: An error occurs when executing the get_blueprint_library() method

seominseokt, 2024-07-11:
CARLA version: 0.9.15
Platform/OS: Windows 11

When using the `get_blueprint_library` method, it returns the following error:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[3], line 1
----> 1 blueprint_library = world.get_blueprint_library()

ValueError: role_name: colors must have 3 channels (R,G,B
```

[image]

I haven't found any other issues that match this problem. Where could this issue be occurring?

jiushixiensi, 2024-07-12:
I also encountered the same problem and wanted to ask if the problem was solved.

seominseokt, 2024-07-12:
@jiushixiensi I formatted my computer for a clean install, but the same problem persists.

chuanye-Wang, 2024-07-15:
Some issue. Besides, I run manual_control.py and encounter the same error

seominseokt, 2024-07-16:
@jiushixiensi @chuanye-Wang Did you guys solve it? Let me know when you do.

chuanye-Wang, 2024-07-17:
commit: 4da0c7415499981f29874b899edd00c1368df06f
This commit wworks just fine.

seominseokt, 2024-07-18:
@chuanye-Wang 

Thank you!! It works correctly after rolling back as follows:

```
git reset --hard "4da0c74"
```

However, this is a temporary solution, so I'll keep the issue open for now.

yyfl8, 2024-07-22:
> commit: [4da0c74](https://github.com/carla-simulator/carla/commit/4da0c7415499981f29874b899edd00c1368df06f) This commit wworks just fine.

您好！我也是最近刚刚下载了这个版本，遇到了同样的问题。想请教一下您具体是怎么解决的这个报错。

chuanye-Wang, 2024-07-22:
Just reverse to commit as I mentioned above, But it's just a temporary solution

Turuu1124, 2024-07-26:
i tried running git reset --hard "4da0c74" but i am still  getting the same issue

Nithin-Supani, 2024-08-06:
@chuanye-Wang @seominseokt 

I tried your temporary solution to roll back "git reset --hard "4da0c74", then rebuild PythonAPI and make launch.
But this solved the error "ValueError: role_name: colors must have 3 channels (R,G,B)"

i am getting new error when i try to spawn a camera and attach it to the vehicle:

53 camera_init_trans = carla.Transform(carla.Location(z=CAMERA_POS_Z,x=CAMERA_POS_X))
54 #this creates the camera in the sim
---> 55 camera = world.spawn_actor(camera_bp,camera_init_trans,attach_to=vehicle)
57 def camera_callback(image,data_dict):
58 data_dict['image'] = np.reshape(np.copy(image.raw_data),(image.height,image.width,4))

**RuntimeError: rpc::rpc_error during call in function spawn_actor_with_parent**

This is the same case if i run any generate_traffic.py:

python generate_traffic.py
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = 0.9.15
WARNING: Simulator API version = 0.9.15-221-g4da0c7415-dirty

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
File "/home/milab/carla/PythonAPI/examples/generate_traffic.py", line 374, in
main()
File "/home/milab/carla/PythonAPI/examples/generate_traffic.py", line 247, in main
for response in client.apply_batch_sync(batch, synchronous_master):
**RuntimeError: rpc::rpc_error during call in function apply_batch**

chuanye-Wang, 2024-08-06:
BTW, I have another guess:  rpc::rpc_error is a romote call error, it means the mismatch of the version between Carla-dev and PythonAPI, so if you dont run ./Update.sh during the buiding carla process, maybe it wont have this issue then.

Nithin-Supani, 2024-08-06:
I ran ./Update.sh before building the PythonAPI, it said content is up-to-date.

even now if i run ./Update.sh i get content is up-to-date.

but i used to get this warning before rolling back also:
"WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = 0.9.15
WARNING: Simulator API version = 0.9.15-221-g4da0c7415-dirty"

isn't client and simulator API version same 0.9.15?

chuanye-Wang, 2024-08-06:
It is the same version in general, but it's different in detail.

I mean you donot run./updat

stavtant, 2024-08-07:
Windows 10 Pro
GeForce RTX 4060
carla 0.9.15
UE4

I face the same issue (ValueError: role_name: colors must have 3 channels (R,G,B)) when i try to run scripts after importing a vehicle which I made via Blender. The strange thing is that only the dynamic weather is working. I tried switching python versions but it didn't work. When I delete the vehicle which I made, from CarlaUE4 editor, the error I'm facing is:

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version = 0.9.15
WARNING: Simulator API version = 0.9.15-221-g4da0c7415-dirty
INFO: Found the required file in cache! Carla/Maps/Nav/Town10HD_Opt.bin
INFO: Found the required file in cache! Carla/Maps/TM/Town10HD_Opt.bin

destroying 0 vehicles
destroying 0 walkers

done.
Traceback (most recent call last):
File "C:\carla\PythonAPI\examples\generate_traffic.py", line 379, in
main()
File "C:\carla\PythonAPI\examples\generate_traffic.py", line 247, in main
for response in client.apply_batch_sync(batch, synchronous_master):
RuntimeError: rpc::rpc_error during call in function apply_batch

Any ideas?

Nithin-Supani, 2024-08-07:
> It is the same version in general, but it's different in detail.
> 
> I mean you donot run ./update.sh during the building process. So that your carla and PythonAPI would both being the old version.
> 
> Or you can try to use release version to validate your code, and use carla-dev to edit map.
> 
> what's your perpose to use carla-dev

Yes, I was planning to do the same thing. For now I just wanted to build a custom car form the Carla-dev.

chuanye-Wang, 2024-08-07:
> Yes, I was planning to do the same thing. For now I just wanted to build a custom car form the Carla-dev. 

You can

chuanye-Wang, 2024-08-07:
You can edit your custom car in carla-dev, export it as an asset, then import your custom car assets into a realse version of carla (e.x. carla 0.9.15).In the release version, it wont have the RGB issue or rpc issue.

I can tell you, that's the way I check my custom map,and do further development.

stavtant, 2024-08-12:
**A mini update, i find out that on PythonAPI folder there is no dist folder which should contain the .egg or .whl file. I tried again the make PythonAPI command and nothing changed although on prompt the output was: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist". Anyone know why?

Turuu1124, 2024-08-14:
try deleting your PythonAPI folder and then run make PythonAPI again, it fixed it for me the 2nd time i did this

stavtant, 2024-08-20:
I did this many times, not just once and it didn't fix the problem. :(

rebekapolei111, 2024-08-26:
does someone solve this problem?

stavtant, 2024-08-26:
**SOLUTION!!**  Follow these steps. 
In a terminal: 
git fetch --all --tags
git checkout 0.9.15

In x64 native tools prompt:
make PythonAPI
make launch

enn25, 2024-08-26:
> **SOLUTION!!** Follow these steps. In a terminal: git fetch --all --tags git checkout 0.9.15
> 
> In x64 native tools prompt: make PythonAPI make launch

I followed the same steps as you mentioned but while running make PythonAPI i am getting this error. Can someone explain why it is showing or know how to fix?Note : I am using Ubuntu 22.04,16 GB RAM, i5 13th H series, RTX 4050..

Util/BuildTools/Linux.mk:163: warning: overriding recipe for target 'downloadplugins'
Util/BuildTools/Linux.mk:142: warning: ignoring old recipe for target 'downloadplugins'
Setup.sh: boost-1.80.0-c10 already installed.
Setup.sh: rpclib-v2.2.1_c5-c10 already installed.
Setup.sh: gtest-1.8.1-c10 already installed.
Setup.sh: recast-c10 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Eigen already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 0.9.15-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Client"
BuildLibCarla.sh: Success!
BuildOSM2ODR.sh: Building OSM2ODR.
fatal: reference is not a tree: 1835e1e9538d0778971acc8b19b111834aae7261
make: *** [Util/BuildTools/Linux.mk:157: osm2odr] Error 128

stavtant, 2024-08-26:
I'm using Windows 10 Pro Platform not Ubuntu so I can't help :(

> I followed the same steps as you mentioned but while running make PythonAPI i am getting this error. Can someone explain why it is showing or know how to fix?Note : I am using Ubuntu 22.04,16 GB [...]

enn25, 2024-08-26:
> I'm using Windows 10 Pro Platform not Ubuntu so I can't help :(

Ohk Let me try again from beginning hope it fix after that. I just have a doubt whether you directly used these command that you said after the value error or rpc error?

saltyfishxbr, 2024-08-29:
I've solved this.
Maybe you build your carla from source. But your carla pythonAPI is not the one you made from source. If you build your carla from source, you need to use the built carla pythonAPI in carla/PythonAPI/carla/dist/carla-xxx.whl
Do not use pip install carla directly from network. It doesn't match your carla version

daohu527, 2024-10-05:
As @saltyfishxbr said. There will be an error message indicating that the **Python API version is inconsistent** (other versions have been installed before). Use the following command to reinstall to solve the problem.
```
pip3 install PythonAPI/carla/dist/carla-xxx.whl --force-reinstall
```

Sephiroth1118, 2024-10-09:
> As @saltyfishxbr said. There will be an error message indicating that the **Python API version is inconsistent** (other versions have been installed before). Use the following command to reinstall solved the problem.
> 
> ```
> make PythonAPI
> 
> pip3 install PythonAPI/carla/dist/carla-xxx.whl --force-reinstall
> ```

Hi! My carla-dev is built from source and has the same problem. This solution worked well. But recently I'm working on a project that use python 3.8.10, the wheel file in PythonAPI/carla/dist/ only has the cp310, which is the version for python 3.10. I can not install the wheel. Do you know where I can find the cp38 version for carla-dev? Thank you so much!

a2mar, 2024-10-12:
> Hi! My carla-dev is built from source and has the same problem. This solution worked well. But recently I'm working on a project that use python 3.8.10, the wheel file in PythonAPI/carla/dist/ only [...]

For that, you can download the binary version of CARLA (the overnight build on github). You can find it under the proposed path in the reinstallation command that was provided in the penultimate comment.

Worked for me.

rohanNkhaire, 2024-10-16:
@chuanye-Wang Hey, I generated a map using the digital twin tool. I wanted to export it as an asset. Could you tell me how do I do that?

chuanye-Wang, 2024-10-17:
> @chuanye-Wang Hey, I generated a map using the digital twin tool. I wanted to export it as an asset. Could you tell me how do I do that?

@rohanNkhaire Just follow the steps right here: https://github.com/carla-simulator/carla/issues/7939#issuecomment-2418129986

Sephiroth1118, 2024-10-17:
> > > As @saltyfishxbr said. There will be an error message indicating that the **Python API version is inconsistent** (other versions have been installed before). Use the following command to reinstall solved the problem.
> > > ```
> > > make PythonAPI
> > > 
> > > pip3 install PythonAPI/carla/dist/carla-xxx.whl --force-reinstall
> > > ```
> > 
> > 
> > Hi! My carla-dev is built from source and has the same problem. This solution worked well. But recently I'm working on a project that use python 3.8.10, the wheel file in PythonAPI/carla/dist/ only has the cp310, which is the version for python 3.10. I can not install the wheel. Do you know where I can find the cp38 version for carla-dev? Thank you so much!
> 
> For that, you can download the binary version of CARLA (the overnight build on github). You can find the `whl` under the proposed path in the reinstallation command that was provided in the penultimate comment.
> 
> Worked for me.

Thanks a lot for your reply! It worked for me!

rohanNkhaire, 2024-10-17:
@chuanye-Wang Hey, thanks for the reference. I'll try that out. 

I tried the steps mentioned by @saltyfishxbr.
```bash
# Uninstalled carla package
pip3 uninstall carla

# Make PythonAPI and then install the .whl file
make PythonAPI
pip3 install PythonAPI/carla/dist/carla-xxx.whl --force-reinstall
```

This solved the role_name error and I was able to run the python examples.

rohanNkhaire, 2024-10-24:
@chuanye-Wang Hey, I tried to make the package of the digital twin map using the reference you mentioned. I am getting signal 11 caught error. Have you faced this issue?

Blyron, 2024-11-06:
Hey folks,
Commented a solution here: https://github.com/carla-simulator/carla/issues/8035#issuecomment-2320778135

closed 2024-11-06 by Blyron