carla-simulator/carla #7950: CARLA 0.9.15: Missing Collision Information for Mitsubishi FusoRosa and European HGV

litroe940408, 2024-07-12:
Sure, here is the text with optimized English expressions:

CARLA Version: 0.9.15
Platform/OS: Windows 10

Issue:

I'm encountering a collision detection anomaly in CARLA version 0.9.15 on Windows 10. When executing ``manual_control.py`` with the ``--sync`` flag, specific vehicles (namely, the Mitsubishi FusoRosa and European HGV models) exhibit the following behavior:

Absent Collision Information: Upon collision with other objects, the collision sensor equipped on the ego vehicle fails to generate any debug information. As a result, the collision HUD (heads-up display) remains devoid of any collision-related data.

Expected Behavior:

As illustrated in the provided image, the Mitsubishi FusoRosa should produce collision information (reflected in both the collision sensor output and the collision HUD) when it collides with another object.

Reproduction Steps:

Execute ``python manual_control.py --sync`` within any CARLA map.
Switch the ego vehicle from a small car to a large vehicle like the Mitsubishi FusoRosa or European HGV.
Initiate a collision between the ego vehicle and another object in the environment.
Observe the absence of collision messages or collision count in the debug output
Additional Information:
[image]

RodrigoVillarias, 2024-07-26:
Hello, we have checked collisions and so on and it seems that everything is ok, except for those two vehicles. We have managed to reproduce the error and we will take a closer look at why it fails. Thanks for waiting

cvrunmin, 2025-03-04:
I managed to fix the missing collision of FusoRosa by enabling "Simulation Generates Hit Events" of "Vehicle_Base" in "Phys_FusoRosa" asset. This property is also disabled in European HGV, but enabled in other vehicles which properly receiving collision event. idk why they're disabled in the first place...