carla-simulator/carla #7951: CARLA 0.9.15: Agent Collision at Junction with Mitsubishi FusoRosa

litroe940408, 2024-07-12:
I'm encountering an issue with agent behavior in CARLA version 0.9.15 on Windows 10. While running ``manual_control.py`` with the ``--sync -a`` flags in the Town02 map, the following behavior is observed:

Collision at Junction: When the agent is controlling a specific vehicle model (e.g., Mitsubishi FusoRosa) and approaching a particular junction, it tends to make a turning error. While attempting a left turn, the agent often turns prematurely, resulting in a collision with a street lamp located at the junction (as shown in the provided image).

[image]

Expected Behavior: The agent should successfully navigate the left turn at the junction without causing a collision with the street lamp.

Reproduction Steps:

Execute python ``manual_control.py --sync -a`` in the Town02 map.
Spawn a vehicle (preferably the Mitsubishi FusoRosa) near the specified location in the image.
Allow the agent to control the vehicle and observe its behavior as it approaches the junction.
Witness the potential collision between the vehicle and the street lamp during the left turn maneuver.
Additional Information:

It's worth noting that this collision also occurs with NPC vehicles controlled by the agent when using ``generate_traffic.py --filterv vehicle.mitsubishi.fusorosa``.

[image]

RodrigoVillarias, 2024-07-12:
Hello, this is a known problem of Carla from the old towns, very tight turns were made, and later when long vehicles were modelled it was seen that they had problems to turn in such small curves, that's why a label was created, to know which vehicles were suitable to drive without problems in those maps. Greetings.

RodrigoVillarias, 2024-07-19:
Hello, this solved your question? the issue will be closed, in any case if something happens feel free to reopen the issue

closed 2024-07-19 by RodrigoVillarias