carla-simulator/carla #7953: Carla crashes when running python scripts : python3 manual_control.py

shraddha-git-tech, 2024-07-14:


Jinviz, 2024-07-23:
Try changing the default value of --generation to 3 at line 1330 of the manual_control.py !

[image]

donghaiwang, 2024-08-16:
When the Python client is the dev branch and the server is the 0.9.15 release, I encountered a similar issue. If both the server and client are the latest dev branches, there is no such problem.

gitwld, 2024-10-11:
I faced the same problem and I solved it by:
As you can see, your Client API version (0.9.15-......) is different with Simulator API version (0.9.15). This is may because you have installed another Carla Python Client API before.
So you can just create a new Python virtual env and:
pip install pygame numpy
cd PythonAPI/carla/dist/
pip install carla-0.9.15-cp37-cp37m-manylinux_2_27_x86_64.whl (maybe your filename is different)

lijieamd, 2024-12-24:
### Problem

- I faced the same problem but my environment is win10 and I was running **CppClient example**

- The version seems to match correctly between server and client

### environment

- system: win10

- carla server: downloaded from **ue4-dev** branch of github README

- carla client: build from **ue4-dev** branch of github

### Log
The log after running cppclient.exe
```
Client API version : 0.9.15
Server API version : 0.9.15
Loading world: /Game/Carla/Maps/Town02
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin 
INFO:  Could not find the required file in cache, downloading...  Carla/Maps/Nav/Town02.bin 

Exception: sticky_control: invalid value type
```