carla-simulator/carla #7961: UE4 Crash while draging Traficlights into the scene and play

chuanye-Wang, 2024-07-17:
0.9.15-dev  
Ubuntu 20.04
issue: CarlaUE4 crashed when added trafficlights into the scene and play.
expectation: I wanna add trafficlights to my custom map exported from Roadrunner.

I import a custom map from RoadRunner, I have placed traficlights in the Roadrunner, but none of them showed up in carlaUE4. So I follow these step to place Trafficlights using CarlaUE4
[image]

After I draged the trafficlights into the scene, and I Play, then CarlaUE4 crashed!
[image]

Is there solution for this? Or, anyother way to add trafficlights to my custom map exported from Roadrunner?
tks!

chuanye-Wang, 2024-07-19:
Thanks, it works

gsmario, 2024-07-26:
I'm glad it worked!
Then i'll close the issue.
Thanks again!

closed 2024-07-26 by gsmario

SExpert12, 2024-08-18:
Hi,
I am also facing the same issue. Earlier I followed the steps given by @chuanye-Wang but traffic lights not working and now I it server is getting crashed. 
How to go to this sublevel  as given in this step 
"First, move to the correct sublevel for traffic lights (double click sublevel "layout")"

Thanks