carla-simulator/carla #7963: How to control the specific light of a TrifficLight

chuanye-Wang, 2024-07-19:
There are 3 seperate lights on one trafficlight. I use "set_state()" method to control the light, but it control the 3 lights the same time. Someone knows how to control them seperately?
[image]

DnDare, 2024-07-19:
Hi, I'm sorry to tell that you can't control separately the lights of a same trafficlight. Each whole traffic light is controlled by the same control box.

Altternatively, you could use multiple traffic lights in the same position, by showing/hidding different parts of the geometry. This way you would have different control boxes, that you would be able to control them separately.

It's not the best solution but it might work for you.

chuanye-Wang, 2024-07-20:
So much fun, and it works, thank you

chuanye-Wang, 2024-07-22:
[image]

DnDare, 2024-07-26:
Glad to know it worked!!

closed 2024-07-26 by DnDare

SExpert12, 2024-08-16:
Hi,
I also want to use traffic lights in customized map in CARLA. For the same I use the code this code:

while start_time <= end_time:
            #print("in while")
            world.tick(5)

            n_vehicles = len(world.get_actors().filter('*vehicle*'))
            #print("while",n_vehicles,max_vehicles)
            vehicle_bp = random.choice(blueprints)
            # Spawn vehicle only after delay
            

            if counter == spawn_delay and n_vehicles < max_vehicles:
                #print("if loop",n_vehicles,max_vehicles)
                
                # vehicle = world.try_spawn_actor(vehicle_bp, spawn_point_1)
                vehicle = world.try_spawn_actor(vehicle_bp, random.choice(spsT))
                if vehicle:
                    vehicle.set_autopilot(True)
                    traffic_light=vehicle.get_traffic_light()
                    # print("traffic_light:",traffic_light)
                    traffic_manager.random_right_lanechange_percentage(vehicle,0)
                    traffic_manager.set_path(vehicle, random.choice(routelistT))

                    # traffic_manager.set_path(vehicle,route_1)
                    
                    traffic_manager.update_vehicle_lights(vehicle, True)
                    traffic_manager.random_left_lanechange_percentage(vehicle, 0)
                    traffic_manager.auto_lane_change(vehicle, False)
                    traffic_manager.distance_to_leading_vehicle(vehicle, 5)

                    
                    traffic_manager.ignore_lights_percentage(vehicle,0)
                    # traffic_manager.set_path(vehicle, route_1)
                    
                counter -= 1
            elif counter > 0:
                counter -= 1
            elif counter == 0:
                counter = spawn_delay   
            world.tick(10)
          
            actor_list=world.get_actors()
            print(actor_list)
            for actor in actor_list:
                if isinstance(actor,carla.TrafficLight):
                    actor.set_red_time(5.0)
                    actor.set_green_time(15.0)
                    actor.set_yellow_time(3.0)
                    
                    # actor.set_red_time(2.0)
                    # actor.set_green_time(0.01)
                    # actor.set_yellow_time(2.0)
                    

            start_time +=0.05

But vehicle is not moving as per traffic lights. I have three lanes per approach and I want to release the traffic of left turn and straight going vehicle at the same time.

Can you please help me?