carla-simulator/carla #7965: Share / export custom maps in collaborative projects

Reitmania, 2024-07-19:
Hi there,

we use Carla 0.9.15 as simulation environment in our research project. I have created a custom map (Blender driving scenario creator --> imported to UE4 4.26 --> customization with assets, traffic lights etc). The map works fine, but I would like to share it with the collaborators in our project. Are there some recommended ways to do this? I know, that I can do a package build, but then I would need to send the whole Carla package with the included custom map to the others. And not everyone is able to build Carla from source and has UE4 running on the computer, so I would like to avoid this way as well.

Any ideas? Thanks in advance!

Blyron, 2024-07-26:
Hello!

There is no easy way to do this. The only idea is that you create a way to compile editor mode and send your team the editor binaries in private mode. Epic does not allow to redistribute editor binaries so we do not support that. I just suggest to use the package with the custom map. 

Kind regards

PatrickPromitzer, 2024-07-30:
Hi,
you could move everything map related in the Content folder (except the Carla folder) to a separate folder.

Example:
```
- Carla
    - Unreal
        - CarlaUE4
            - Content
                - Carla
                - my_map
```
After generating a package, you can send the folder of "my_map" in the package to the other machine.
You need to test it, but it should work.

You still need the package version of Carla on the other computer, but you only have to share your map files to the other machine.

closed 2024-08-02 by Blyron