carla-simulator/carla #7970: How to fetch the instance variable of the actor (trafficlight)

chuanye-Wang, 2024-07-23:
0.9.15-dev
ubuntu20.04

I manually create a variable named "rolename" for a trafficlight in UE4 editor as shown below. Then I config the "rolename" variable in UE4 editor Details window. After I run the simulation, I use PythonAPI actor.attributes() method, print the result, but it turns out to be null, why's that?
How can I get the custom variable that I set to the trafficlight instance?
Any solution? Thanks.
[image]

Blyron, 2024-07-26:
Hello! 

Is not as simple as that. If you want to add attribute to a Python class you will have to add the attribute at PythonAPI (Look At PythonAPI/carla/source/carla/Actor.cpp ) Then you will have to make the entire connection through LibCarla to UE Plugin and create on the plugin of c++ the variable with the proper UPROPERTY() in order to edit the variable in UE. Is quite complex, if you want more support we have Services you can acquire. ( https://carla.org/services/#consulting-services ). 

Additional info that may be interesting:
https://docs.google.com/document/d/1saHuh0QSH43xo4dwa88PxVbOM09C4g1QtK9wlGLBtYs/edit

closed 2024-07-26 by Blyron

chuanye-Wang, 2024-07-26:
Thank you for sure

EmmettJZK, 2024-07-26:
我来解释以下，CARLA PythonAPI里面的Attribute或者ActorBlueprint是他们自己写的概念。是为了配置一些复杂传感器，具体可以去看Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h文件，里面都是传感器的Definition和Description的转换。
你设置的变量，是UE蓝图中的变量，可以在C++代码中读取，可以参考：[使用C++借助反射批量获取/修改蓝图参数](https://www.bilibili.com/read/cv26036003/)或者[蓝图访问C++变量](https://www.bilibili.com/read/cv16294754/)。
我给你简单讲一下C++代码和蓝图的关系：
这个路灯中用到了TrafficLightComponent组件，是CARLA的C++类，通过UE的反射系统实现在Editor中显示，调用。
这个组件下的函数、变量有大部分是在C++代码中写的。
代码如图：
[image]
在UE细节（details）中显示：
[image]

当然你是希望在蓝图中创建变量，然后在PythonAPI中获取，当然这里要涉及C++开发和PythonAPI对应接口开发。如何去获取可以参考上面的文章。
这里我简单演示一下怎么在Details中展示出来：
选择【可编辑实例】
[image]
然后就出现在【细节】中了
[image]

如果希望用CALRA做一些项目，欢迎咨询我们深信科创，我的微信号:1824729026

English Version(By GPT4):
Let me explain. The Attribute or ActorBlueprint in the CARLA PythonAPI is a concept they developed themselves. It is used to configure some complex sensors. You can refer to the file Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.h, which contains the conversion of sensor definitions and descriptions.

The variables you set are variables in the UE blueprint, which can be read in the C++ code. You can refer to: [Batch obtain/modify blueprint parameters using C++ through reflection](https://www.bilibili.com/read/cv26036003/) or [Blueprint access to C++ variables](https://www.bilibili.com/read/cv16294754/).

Let me briefly explain the relationship between C++ code and blueprints:
The traffic light component used in this streetlight is a CARLA C++ class that is displayed and invoked in the Editor through UE's reflection system. Most of the functions and variables under this component are written in C++ code.

Here is the code:
[image]

Displayed in UE details:
[image]

Of course, if you want to create variables in the blueprint and then obtain them in PythonAPI, this will involve C++ development and the development of corresponding interfaces in PythonAPI. You can refer to the above articles on how to obtain them.

Here, I will briefly demonstrate how to display it in the Details:
Select "Editable Instance"
[image]

Then it appears in "Details"
[image]

Blyron, 2024-07-26:
PLEASE DO YOUR COMMENTS IN ENGLISH SO ANYONE CAN READ THEM

EmmettJZK, 2024-07-26:
> PLEASE DO YOUR COMMENTS IN ENGLISH SO ANYONE CAN READ THEM

Thanks for the reminder. I have attached an English copy below.