carla-simulator/carla #7979: New car from Blender file does not accelerate

Hunterling, 2024-07-24:
CARLA version:
Client API version     = 0.9.15-10-ge4f43b1b-dirty
Simulator API version  = 0.9.15-21-g7d25be1c-dirty
Platform/OS: Windows 11 Enterprise

Problem you have experienced:
I tried to create a new car from a Blender file in Carla, although it does not accelerate when I try to drive it with the example given in the last step from the "Add a new vehicle tutorial".

I imported the car from a Blender file with the help of an addon mentioned in another issue, thus I believe I have to adjust some values in either the blueprint for the car itself (as when I hold W for acceleration the car changes gears) or in the collision (as I had quite some problems here before).

The issue: https://github.com/carla-simulator/carla/issues/5399
The addon: https://continuebreak.com/creations/ue4-vehicle-rigging-addon-blender/
Video to the addon: https://www.youtube.com/watch?v=ZKP0heuQdVY

If any files or information are lacking, feel free to ask, I'll try to get them ASAP.

Hunterling, 2024-07-24:
@GoodarzMehr Thanks for the hint, though I'm not versed enough with these values to know which ones are proper for my car.
I'll do it with one of my engineer so it will take a few days until I can give a proper answer.

gsmario, 2024-07-26:
Hello! first, thank you @GoodarzMehr for the help, is always welcome!
You can adjust the torque has he says here:
(go to your vehicle blueprint)
[image]

Also, this is the torque we have in some vehicles, you can copy all the next text, go to the option in your blueprint, right click and then paste.

`(EditorCurveData=(Keys=((Time=-9.310944,Value=749.000000),(Time=4000.000000,Value=749.000000),(Time=5000.000000,Value=610.000000),(Time=6000.000000,Value=507.000000),(Time=7000.000000,Value=435.000000),(Time=8000.000000,Value=381.000000),(Time=9000.000000,Value=338.000000)),DefaultValue=340282346638528859811704183484516925440.000000,PreInfinityExtrap=RCCE_Constant,PostInfinityExtrap=RCCE_Constant),ExternalCurve=None)`

If that doesn't work, I recommend you following the documentation instead of using the pluging 
[Here](https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/#add-a-4-wheeled-vehicle)

Tell me if it works that way.

Thanks for your time!

Hunterling, 2024-08-01:
@gsmario Thanks for the suggestion, it helped me finding the error and it works now!

My problem was naming the bones and parts of the car after making the rigging with the plugin (which named the vehiclebase root and the wheel FR,FL,RL and RR), renaming them as needed in the tutorial fixed it:
[image]

On another note:
In step 2 of "Bind and model the vehicle" of the [tutorial](https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/#add-a-4-wheeled-vehicle) it is said to name the bone for the vehicle itself "VehicleBase" **BUT** in step 3 of "Import and configure the vehicle", where the collision is made its named "Vehicle_Base". To my experience only the latter works not the first, which may be part of my confusion.

As this is resolved for me, you can close this ticket if you have no more questions.

closed 2024-08-07 by Hunterling