carla-simulator/carla #7981: Carla version MIss match

Turuu1124, 2024-07-24:
Hello,
OS: Windows 11 64bit 
i am having issues with version miss match as shown below:
(carla-sim) C:\carla\PythonAPI\examples>python generate_traffic.py
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.15
WARNING: Simulator API version  = 0.9.15-226-g5c2d2978a-dirty
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin
ERROR: invalid color autopilot

destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "generate_traffic.py", line 374, in <module>
    main()
  File "generate_traffic.py", line 195, in main
    blueprints = get_actor_blueprints(world, args.filterv, args.generationv)
  File "generate_traffic.py", line 33, in get_actor_blueprints
    bps = world.get_blueprint_library().filter(filter)
ValueError: role_name: colors must have 3 channels (R,G,B)

I tried changing my python version to 3.7 but the carla dist folder has 3.12 so i am confused why and what it means by 0.9.15-226-g5c2d2978a-dirty did i mess up when i was building carla or is this a new issue. I use anaconda prompt to connect.

Jinviz, 2024-07-25:
You can ignore the version mismatch when you run python script. 
Would you like to enter the Python file and change the default value of --generationv in the main function from 'ALL' to 1 or 2?

Turuu1124, 2024-07-25:
when I run carla using the x64 and running it via the make launch it gives me the above error, but when i launch the file that says CARLA_0.9.15 that i unzipped from the github repository it works fine and all the scirpts are running correctly, so i am a little confused why that is they are both carla 0.9.15 but the one that i manually built from the source doesnt work with the Pythno api anaconda thing

Jinviz, 2024-07-26:
Oh.. I'm understand. Wouldn't it be okay to add the option '--recursive' when you do git clone?

Jinviz, 2024-07-26:
> x64를 사용하여 carla를 실행하고 make launch를 실행하면 위의 오류가 발생하지만 github에서는 압축을 푼 CARLA_0.9.15라는 파일을 시작하면 잘 작동하고 모든 확장이 실행될 때 모두 carla 0.9.15인 이유가 헷갈리네요. 하지만 소스에서 매뉴얼로 빌드한 것은 Pythno api anaconda와 작동하지 않습니다.

Maybe there was some confusion in the branch approach. If you don't give the branch option when you do the current git clone, clone the 'dev' branch with default. This is different from the commit point in the 0.9.15 release. If you want to do the 0.9.15 release version of git clone exactly, you have to do `git clone -b 0.9.15 https://github.com/carla-simulator/carla.git `
+ Last commit of 0.9.15 release is  'd7b45c1'

stavtant, 2024-08-07:
Windows 10 Pro
GeForce RTX 4060 
carla 0.9.15
UE4

I face the same issue (ValueError: role_name: colors must have 3 channels (R,G,B)) when i try to run scripts after importing a vehicle which I made via Blender. The strange thing is that only the dynamic weather is working. I tried switching python versions but it didn't work. When I delete the vehicle which I made, from CarlaUE4 editor, the error I'm facing is:

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.15
WARNING: Simulator API version  = 0.9.15-221-g4da0c7415-dirty
INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin
INFO:  Found the required file in cache!  Carla/Maps/TM/Town10HD_Opt.bin

destroying 0 vehicles
destroying 0 walkers

done.
Traceback (most recent call last):
  File "C:\carla\PythonAPI\examples\generate_traffic.py", line 379, in <module>
    main()
  File "C:\carla\PythonAPI\examples\generate_traffic.py", line 247, in main
    for response in client.apply_batch_sync(batch, synchronous_master):
RuntimeError: rpc::rpc_error during call in function apply_batch 

Any ideas?

jorge-kabuto, 2024-08-30:
Hey folks,
Commented a solution here: https://github.com/carla-simulator/carla/issues/8035#issuecomment-2320778135

Ziefen-Chou, 2024-09-25:
Have you solve this problem?

closed 2025-07-16 by Blyron