carla-simulator/carla #7987: Sensor-induced memory issues cause carla to crash

ZhaoZhiqiao, 2024-07-25:
CARLA version: 0.9.15(both the build version of CARLA and the packaged versionof CARLA.)
Platform/OS: Ubuntu 22.04
Problem you have experienced: Carla crashes and exits automatically
What you expected to happen:  System remains operational
Steps to reproduce:  Frequent and repeated creation and deletion of sensors
Other information (documentation you consulted, workarounds you tried):

Thank you very much for your work.
I have noticed that it seems as if you seem to be solving the problems I am having . So I am providing some specific steps and reasons for the problems I am having.
Because I want to acquire a large number of images from different viewpoints, I am frequently creating cameras, acquiring an image, stopping and deleting them using the PythonAPI. I am more than certain that at 1500-3000 per repetition of this operation, the system crashes. I was able to get the error message and some other statuses from the moment before the crash through the recording monitor.

Here's my error message, after it displays, both carla and the terminal just auto-exit without any more realistic information.
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```

My program generates sensors in groups of two and then deletes them. After successfully deleting sensors 2609, 2610, and successfully generating sensor 2611, the carla crashes when trying to generate sensor 2612, causing the program using the PythonAPI to crash.(Below is a screenshot of the crash, including the system monitor at that time)
[image]

veinlove55, 2024-07-30:
I'm facing the same problem. After I set the parameter "-quality-level=Low" in CarlaUE4.sh script, this can extend the running time, at least 4 hours. But another exception occurs, it's another topic. Hope this info helps.

GoodarzMehr, 2024-08-07:
This may sound trivial, but why don't you move the cameras around instead of repeatedly creating and destroying them?