carla-simulator/carla #7989: spawn an actor at a specific location

ccconquer, 2024-07-26:
CARLA version: 0.9.15
Platform/OS: ubuntu20.04
Problem you have experienced: I want to spawn an actor at a specific location. How can I get the coordinates on the map?

Blyron, 2024-07-26:
You can change the transform of the spawn to change the position. If you want to get which is an specific location you will have to build carla from source and open the desired map and check which is the location you like.

GoodarzMehr, 2024-07-26:
If you are using a packaged version, one thing you could try is to place the spectator at your desired location, then get the spectator's transform from the client. If you want you can then use that location to get a waypoint from the map using the `get_waypoint()` method.

PatrickPromitzer, 2024-07-30:
Hi,
like GoodarzMehr said, the easiest option would be using the spectator.

Move the Spectator to the location you want to spawn the actor, and save the position.
```python
client = carla.Client(server_address, 2000)
world = client.get_world()
spectator = world.get_spectator()
spec_trans = spectator.get_transform()
```

After that, you can spawn the actor at the position.
```
actor_transform = carla.Transform(
    carla.Location(
        x=0.3,
        y=-0.2,
        z=1.7),
    carla.Rotation(
        pitch=0.0,
        yaw=0.0,
        roll=0.0))
actor = world.spawn_actor(actor_bp, actor_transform)
```