carla-simulator/carla #7991: Carla UE5 build failed due to clang 16 compatibility issue

jeromepatel, 2024-07-26:
Hello, 
I am trying to build the Carla version with Unreal Engine 5 from the official docs [given here](https://carla.readthedocs.io/en/latest/build_linux_ue5)
I am getting clang ` cannot find /lib64/ld-linux-x86-64.so.2
` error, I have tried on two different linux machines and encountering the same error. 

My machine has 64 GB RAM and 24 GB VRAM so I don't think it's an issue. I tried reinstalling system clang from the git (to version 20.0) and it still gives error. I figured the error lies from the fact that clang 16.0.6 from UE5 might be having error with the Ubuntu version 24.04 during compilation (or boost) of carla since the error comes when running `./Setup.sh` and I can see 
```

**********  SUCCESS ****************
Setup successful.

``` 
from UE5 setup script. It's the configuration line in carla setup script which gives this error:
```
retry --until=success --times=10 -- cmake -G Ninja -S . -B Build \
    --toolchain=$PWD/CMake/LinuxToolchain.cmake \
    -DLAUNCH_ARGS="-prefernvidia" \
    -DCMAKE_BUILD_TYPE=Release \
    -DENABLE_ROS2=ON \
    -DBUILD_CARLA_UNREAL=ON \
    -DCARLA_UNREAL_ENGINE_PATH=$CARLA_UNREAL_ENGINE_PATH
```

The error is as follows:
```
CMake Error at /opt/cmake-3.28.3-linux-x86_64/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/home/jeromepatel/Documents/CVL/CarlaUE5/Build/CMakeFiles/CMakeScratch/TryCompile-tedqrJ'
    
    Run Build Command(s): /usr/bin/ninja -v cmTC_afeca
    [1/2] /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang   -fms-extensions -fno-math-errno -fdiagnostics-absolute-paths -MD -MT CMakeFiles/cmTC_afeca.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_afeca.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_afeca.dir/testCCompiler.c.o -c /home/jeromepatel/Documents/CVL/CarlaUE5/Build/CMakeFiles/CMakeScratch/TryCompile-tedqrJ/testCCompiler.c
    [2/2] : && /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang  -stdlib=libc++ -L/home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Source/ThirdParty/Unix/LibCxx/lib/Unix/x86_64-unknown-linux-gnu CMakeFiles/cmTC_afeca.dir/testCCompiler.c.o -o cmTC_afeca   && :
    FAILED: cmTC_afeca 
    : && /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang  -stdlib=libc++ -L/home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Source/ThirdParty/Unix/LibCxx/lib/Unix/x86_64-unknown-linux-gnu CMakeFiles/cmTC_afeca.dir/testCCompiler.c.o -o cmTC_afeca   && :
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/Scrt1.o: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: warning: /lib/x86_64-linux-gnu/libc.so.6: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0008002
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: /lib/x86_64-linux-gnu/libc.so.6: unknown type [0x13] section `.relr.dyn'
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: skipping incompatible /lib/x86_64-linux-gnu/libc.so.6 when searching for /lib/x86_64-linux-gnu/libc.so.6
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: cannot find /lib/x86_64-linux-gnu/libc.so.6
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: /lib64/ld-linux-x86-64.so.2: unknown type [0x13] section `.relr.dyn'
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: skipping incompatible /lib64/ld-linux-x86-64.so.2 when searching for /lib64/ld-linux-x86-64.so.2
    /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/x86_64-unknown-linux-gnu-ld: cannot find /lib64/ld-linux-x86-64.so.2
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:34 (project)

-- The C compiler identification is Clang 16.0.6
-- The CXX compiler identification is Clang 16.0.6
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang
-- Check for working C compiler: /home/jeromepatel/Documents/CVL/UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang - broken
-- Configuring incomplete, errors occurred!
retry: cmake returned 1, backing off for 10 seconds and trying again...
```

To check whether the issue is clang from UE5, I tried running the following simple command 
```UnrealEngine5_carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v22_clang-16.0.6-centos7/x86_64-unknown-linux-gnu/bin/clang test.c -o test_program
```
and it gives the exact same error so now I know where it's coming from (clang is broken inside UE5). I tried to look it up where it's setting up this exact version of LLVM but couldn't find anywhere where it's setting this up from. 
Is there a way to resolve this issue ? Any help would be appreciated. I am guessing if I can work with my system clang (since it works) then I would be able to build the CarlaUE5 properly. 

Thanks!

jeromepatel, 2024-07-29:
Closing this issue since after spending many hours I still couldn't figure out how to make Carla UE5.3 work in Ubuntu 24.04 (probably in future if someone visits this issue again). The solution is to basically change OS to either Ubuntu 22.04 or Windows 10/11. I did later and with few minor errors I was able to make UE5.3 work in windows 11. 

In case someone encounters this issue, it is regarding UE5.3 internal Clang Toolchain compiler only supporting Linux version upto 22.04 ([more info here](https://dev.epicgames.com/documentation/en-us/unreal-engine/linux-development-requirements-for-unreal-engine?application_version=5.3)), This is exact compiler which is being used inside Setup.sh script during compiling. (One crude solution is to remove that and use system's default Clang toolchain but also requires configuring UE5 compiler too), in case the clang support has been extended in future then you can use Ubuntu 24.04 with Carla UE5 without any hassle. Commenting this so that no-one else have to waste many hours on a dead-end.

closed 2024-07-29 by jeromepatel

J160KU, 2024-07-30:
Hello Jerome,

I would use Ubuntu 22.04, since this is the version which is recommended on the offical CARLA docs. Could not read anything mentioned Ubuntu 24.04

jeromepatel, 2024-07-30:
Hi, 
Yes, I resorted to using Windows 11 since it was more convenient for me to use. You are right that Ubuntu 24.04 is not mentioned anywhere, I was hoping to make it work but didn't succeed.

Gragonfly, 2025-01-18:
I use ue4.26 and Ubuntu 24.04 met the same problem, is the same as UE4 or the problem is only for UE5

IoanFerezanRepli5, 2025-01-31:
I have the same issue. Fresh install of W11 
`-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe - broken
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.29/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'C:/Carla/CarlaUE5/Build/CMakeFiles/CMakeScratch/TryCompile-tmlafj'

    Run Build Command(s): C:/Users/fere_/AppData/Local/Microsoft/WindowsApps/ninja.exe -v cmTC_5c587
    [1/2] C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1436~1.325\bin\Hostx64\x64\cl.exe  /nologo   /DWIN32 /D_WINDOWS  /Ob0 /Od /RTC1 -MDd -Zi /showIncludes /FoCMakeFiles\cmTC_5c587.dir\testCCompiler.c.obj /FdCMakeFiles\cmTC_5c587.dir\ /FS -c C:\Carla\CarlaUE5\Build\CMakeFiles\CMakeScratch\TryCompile-tmlafj\testCCompiler.c
    [2/2] C:\WINDOWS\system32\cmd.exe /C "cd . && "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_5c587.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1436~1.325\bin\Hostx64\x64\link.exe /nologo CMakeFiles\cmTC_5c587.dir\testCCompiler.c.obj  /out:cmTC_5c587.exe /implib:cmTC_5c587.lib /pdb:cmTC_5c587.pdb /version:0.0 /machine:x64  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    FAILED: cmTC_5c587.exe
    C:\WINDOWS\system32\cmd.exe /C "cd . && "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_5c587.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests  -- C:\PROGRA~1\MICROS~1\2022\COMMUN~1\VC\Tools\MSVC\1436~1.325\bin\Hostx64\x64\link.exe /nologo CMakeFiles\cmTC_5c587.dir\testCCompiler.c.obj  /out:cmTC_5c587.exe /implib:cmTC_5c587.lib /pdb:cmTC_5c587.pdb /version:0.0 /machine:x64  /debug /INCREMENTAL /subsystem:console  kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib && cd ."
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_5c587.dir/manifest.res CMakeFiles\cmTC_5c587.dir/manifest.rc" failed (exit code 0) with the following output:
    no such file or directory
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:40 (project)

-- Configuring incomplete, errors occurred!

C:\Carla\CarlaUE5>`

innovationcell3, 2025-02-13:
I also have this issue on W11 install. Did anyone figure out a fix?

nallen94, 2025-04-09:
still experiencing the same issue on Ubuntu 24.04. Is there any fix yet ?