carla-simulator/carla #8000: make pythonAPI时遇到问题：OSM2ODR文件不在

King8139, 2024-07-31:


King8139, 2024-07-31:
D:\carla>make PythonAPI

D:\carla>setlocal enabledelayedexpansion

D:\carla>rem Run it through a cmd with the x64 Visual C++ Toolset enabled.

D:\carla>set LOCAL_PATH=D:\carla\Util\BuildTools\

D:\carla>set FILE_N=-[BuildUE4Plugins]:

D:\carla>rem Print batch params (debug purpose)

D:\carla>echo -[BuildUE4Plugins]: [Batch params]: --build --all
-[BuildUE4Plugins]: [Batch params]: --build --all

D:\carla>rem ============================================================================

D:\carla>rem -- Parse arguments ---------------------------------------------------------

D:\carla>rem ============================================================================

D:\carla>set DOC_STRING=Build LibCarla.

D:\carla>set USAGE_STRING=Usage: -[BuildUE4Plugins]: [-h|--help] [--rebuild] [--build] [--clean] [--no-pull]

D:\carla>set BUILD_STREETMAP=false

D:\carla>set GIT_PULL=true

D:\carla>set CURRENT_STREETMAP_COMMIT=260273d6b7c3f28988cda31fd33441de7e272958

D:\carla>set STREETMAP_BRANCH=master

D:\carla>set STREETMAP_REPO=https://github.com/carla-simulator/StreetMap.git

D:\carla>if not "--build" == "" (
if "--build" == "--rebuild" (
set REMOVE_INTERMEDIATE=true
 set BUILD_STREETMAP=true
)
 if "--build" == "--build" (set BUILD_STREETMAP=true )
 if "--build" == "--no-pull" (set GIT_PULL=false )
 if "--build" == "--clean" (set REMOVE_INTERMEDIATE=true )
 if "--build" == "-h" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
)
 if "--build" == "--help" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
)
 shift
 goto :arg-parse
)

D:\carla>if not "--all" == "" (
if "--all" == "--rebuild" (
set REMOVE_INTERMEDIATE=true
 set BUILD_STREETMAP=true
)
 if "--all" == "--build" (set BUILD_STREETMAP=true )
 if "--all" == "--no-pull" (set GIT_PULL=false )
 if "--all" == "--clean" (set REMOVE_INTERMEDIATE=true )
 if "--all" == "-h" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
)
[... 210 lines cut ...]
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
D:\VS1\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -ID:\Python37\include -ID:\Python37\include -ID:\VS1\VC\Tools\MSVC\14.29.30133\ATLMFC\include -ID:\VS1\VC\Tools\MSVC\14.29.30133\include "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" -ID:\VS1\VC\Tools\MSVC\14.29.30133\ATLMFC\include -ID:\VS1\VC\Tools\MSVC\14.29.30133\include "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include\carla/road/element/RoadInfoMarkRecord.h(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的 字符。请将该文件保存为 Unicode 格式以防止数据丢失
D:\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: 无法打开包括文件: “OSM2ODR.h”: No such file or directory
error: command 'D:\\VS1\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!
这是我编译的过程，它显示 无法打开包括文件: “OSM2ODR.h”，怎么解决

veinlove55, 2024-08-01:
你单独跑一下Util/BuildTools/BuildOSM2ODR.bat看看能不能把这个依赖库装成功

King8139, 2024-08-01:
> 你单独跑一下Util/BuildTools/BuildOSM2ODR.bat看看能不能把这个依赖库装成功
我的结果是这样的:
PS D:\carla\Util\BuildTools> .\BuildOSM2ODR.bat
-[BuildOSM2ODR]: [Batch params]:
Nothing selected to be done.

TingzheYu, 2024-08-01:
> > 你单独跑一下Util/BuildTools/BuildOSM2ODR.bat看看能不能把这个依赖库装成功
> > 我的结果是这样的:
> > PS D:\carla\Util\BuildTools> .\BuildOSM2ODR.bat
> > -[BuildOSM2ODR]: [Batch params]:
> > Nothing selected to be done.没有选择执行任何操作。

hello, I also met this problem today and here is my solution for this problem. I hope it would be helpful for you.

TingzheYu, 2024-08-01:
> > 你单独跑一下Util/BuildTools/BuildOSM2ODR.bat看看能不能把这个依赖库装成功
> > 我的结果是这样的:
> > PS D:\carla\Util\BuildTools> .\BuildOSM2ODR.bat
> > -[BuildOSM2ODR]: [Batch params]:
> > Nothing selected to be done.没有选择执行任何操作。

hello, I also met this problem today and here is my solution for this problem. I hope it would be helpful for you. The reason of this problem is that the "BuildOSM2ODR.bat" uses "tar" to unzip the downloaded file "OSM2ODR.zip" in the "D:\carla\Util\BuildTools\BuildOSM2ODR.bat". And the downloaded file "OSM2ODR.zip" would be deleted after "BuildOSM2ODR.bat" running. So there are two solutions.

> 1. Modify "BuildOSM2ODR.bat". Specifically, in line 98, change "tar -xf OSM2ODR.zip" to "7z x OSM2ODR.zip -aoa". Of course, before that, "7zip" should be installed and “7zip” should be added to your system's environment variables to make sure it can be found. Usually the path is "C:\Program Files\7-Zip". Finally you can run "makePythonAPI" as mentioned in Carla document.

> 2. Unzip "OSM2ODR.zip" manually. You need to install "OSM2ODR.zip" from github manually and unzip it in "D:\carla\Build". And then run "makePythonAPI".

> If there is still any question, feel free to get touch with me (382091009@qq.com)