carla-simulator/carla #8006: make PythonAPI crashes with 'int64_t' error

waldiwunder, 2024-07-31:
Hello!

Platform/OS: Ubuntu 20.04

CARLA version: recent version from repo

Problem you have experienced:
I want to build carla. I followed the description here: https://carla.readthedocs.io/en/latest/build_linux/. After the installation of the Unreal Engine, cloning the carla repo and downloading the assets, I tried to do "make PythonAPI".

I get following error message: 

`FAILED: src/CMakeFiles/proj.dir/proj_json_streaming_writer.cpp.o 
/home/user/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++  -DMUTEX_pthread -DPROJ_LIB=\"/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Build/proj-install/share/proj\" -I../src -I../include -Isrc -I/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Build/sqlite-install/include -std=c++14 -fPIC -O3 -DNDEBUG -fvisibility=hidden   -Wall -Wextra -Wswitch -Wshadow -Wunused-parameter -Wmissing-declarations -Wformat -Wformat-security -Wfloat-conversion -std=c++11 -MD -MT src/CMakeFiles/proj.dir/proj_json_streaming_writer.cpp.o -MF src/CMakeFiles/proj.dir/proj_json_streaming_writer.cpp.o.d -o src/CMakeFiles/proj.dir/proj_json_streaming_writer.cpp.o -c ../src/proj_json_streaming_writer.cpp
In file included from ../src/proj_json_streaming_writer.cpp:34:
../src/proj_json_streaming_writer.hpp:42:9: error: no type named 'int64_t' in namespace 'std'; did you mean simply 'int64_t'?
typedef std::int64_t GIntBig;
        ^~~~~~~~~~~~
        int64_t
/usr/include/x86_64-linux-gnu/bits/stdint-intn.h:27:19: note: 'int64_t' declared here
typedef __int64_t int64_t;
                  ^
In file included from ../src/proj_json_streaming_writer.cpp:34:
../src/proj_json_streaming_writer.hpp:43:9: error: no type named 'uint64_t' in namespace 'std'; did you mean simply 'uint64_t'?
typedef std::uint64_t GUInt64;
        ^~~~~~~~~~~~~
        uint64_t
/usr/include/x86_64-linux-gnu/bits/stdint-uintn.h:27:20: note: 'uint64_t' declared here
typedef __uint64_t uint64_t;
                   ^
In file included from ../src/proj_json_streaming_writer.cpp:34:
../src/proj_json_streaming_writer.hpp:93:10: error: class member cannot be redeclared
    void Add(GIntBig nVal);
         ^
../src/proj_json_streaming_writer.hpp:91:10: note: previous definition is here
    void Add(int nVal) { Add(static_cast<GIntBig>(nVal)); }
         ^
../src/proj_json_streaming_writer.hpp:94:10: error: class member cannot be redeclared
    void Add(GUInt64 nVal);
         ^
../src/proj_json_streaming_writer.hpp:91:10: note: previous definition is here
    void Add(int nVal) { Add(static_cast<GIntBig>(nVal)); }
         ^
../src/proj_json_streaming_writer.cpp:45:9: error: no type named 'uint64_t' in namespace 'std'; did you mean simply 'uint64_t'?
typedef std::uint64_t GUIntBig;
        ^~~~~~~~~~~~~
        uint64_t
/usr/include/x86_64-linux-gnu/bits/stdint-uintn.h:27:20: note: 'uint64_t' declared here
typedef __uint64_t uint64_t;
                   ^
../src/proj_json_streaming_writer.cpp:238:30: error: redefinition of 'Add'
void CPLJSonStreamingWriter::Add(GIntBig nVal)
                             ^
../src/proj_json_streaming_writer.hpp:91:10: note: previous definition is here
    void Add(int nVal) { Add(static_cast<GIntBig>(nVal)); }
         ^
6 errors generated.
[57/205] Generating proj.db
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1
`

What you expected to happen: I would expect after following exactly the description in https://carla.readthedocs.io/en/latest/build_linux/, that the build would finish sucessfully.

Other information (documentation you consulted, workarounds you tried): On my system a working Carla installation was there before. Unfortunately it was broken and I decided to go for a new installation. So my system is not a clean system. I found the issue about using a dockercontainer for the build (#6776). I am not familiar with vscode and docker containers, therefore I did not understand what do from this solution. I hope that maybe someone could give me some advice what else, apart from using a container, could help or which tests I could perform to understand better the true source of my problem.

Thank you very much!

-->

PatrickPromitzer, 2024-08-01:
Hi,
you can try the following
open path/to/Carla/Until/BuildTools/Setup.sh
Go to the PROJ install line (around line 637)

Go to the line
```

  log "Extracting PROJ"
  tar -xzf ${PROJ_TAR}
  mv ${PROJ_VERSION} ${PROJ_SRC_DIR}

  mkdir ${PROJ_SRC_DIR}/build
  mkdir ${PROJ_INSTALL_DIR}
```
and add this lines after
```
  sed -i 's;#include <string>;#include <string>\n#include <stdint.h>\n#include <cstdint>;g' ${PROJ_SRC_DIR}/src/proj_json_streaming_writer.hpp
  sed -i 's;#include <string>;#include <string>\n#include <stdint.h>\n#include <cstdint>;;g' ${PROJ_SRC_DIR}/src/proj_json_streaming_writer.cpp
```

I added it to my Setup.sh for some time, because the PROJ installation was making me problems.

I changed the version of PROJ, too
```
PROJ_VERSION=proj-9.3.0
```

I hope this will fix it.

waldiwunder, 2024-08-01:
Thank you very much! This helped me a lot, make PythonAPI finished successfully.

I tried now to do "make launch", but unfortunately run into another error. 

`path to carsim uproject  CarlaUE4.uproject
Traceback (most recent call last):
  File "/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/../../Util/BuildTools/enable_carsim_to_uproject.py", line 65, in <module>
    main()
  File "/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/../../Util/BuildTools/enable_carsim_to_uproject.py", line 33, in main
    uproject_file = open(args.file, 'r')
FileNotFoundError: [Errno 2] No such file or directory: 'CarlaUE4.uproject'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 1
`
Maybe this is due to the fact that I could not execute ./Update.sh to get the assets due to a certificate error. I checked before the file of Update.sh and the URL was according to #7192 the correct one (CONTENT_LINK=https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz). I checked the internet and did, as someone proposed in #7192, copy the assets directly (git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla). This seemed to work fine, but probably it is now the source of the error.

closed 2024-08-01 by waldiwunder