carla-simulator/carla #8011: Error make launch. No such file or directory: 'CarlaUE4.uproject'

waldiwunder, 2024-08-01:
Hi everyone!

CARLA version: Still try to get it running 
Platform/OS: Ubuntu 20.04
Problem you have experienced:

After a reinstallation of Carla, I tried to start Carla with "make launch", but unfortunately run into another error. 

`path to carsim uproject  CarlaUE4.uproject
Traceback (most recent call last):
  File "/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/../../Util/BuildTools/enable_carsim_to_uproject.py", line 65, in <module>
    main()
  File "/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/../../Util/BuildTools/enable_carsim_to_uproject.py", line 33, in main
    uproject_file = open(args.file, 'r')
FileNotFoundError: [Errno 2] No such file or directory: 'CarlaUE4.uproject'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 1
`
Maybe this is due to the fact that I could not execute ./Update.sh to get the assets due to a certificate error. I checked before the file of Update.sh and the URL was according to #7192 the correct one (CONTENT_LINK=https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz). I checked the internet and did, as someone proposed in #7192, copy the assets directly (git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla). This seemed to work fine, but probably it is now the source of the error.

I hope someone can give me some advice here.

What you expected to happen: I would expect after a successful build of PythonAPI that Carla would open and I could run a simulation.

Other information (documentation you consulted, workarounds you tried): My system is not clean, there have been a local Carla installation before, which broke. Therefore I try to reinstall now thing, following the description in https://carla.readthedocs.io/en/latest/build_linux/.

I had before an error during the make PythonAPI, but could solve it with the help of the github community, see #8006

Thanks for your help.

PatrickPromitzer, 2024-08-01:
To make sure the content data is fine, you can download the file from the browser.

You need to replace the ${CONTENT_ID} with the Id of the right version
https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz

You can look up the content id in 
https://github.com/carla-simulator/carla/blob/dev/Util/ContentVersions.txt

After that, you need to create a folder with the name "Content" and a "Carla" folder in it.
- path/to/carla
    - Unreal
     - CarlaUE4
       - Content
         - Carla
The content of the downloaded file need to be copied in the in the "Unreal/CarlaUE4/Content/Carla/" folder.

If you still have a problem after that, you can try delete everything in the "Build" folder, and install the dependencies again.
"make PythonAPI" or "make launch"

waldiwunder, 2024-08-01:
Hi! I get still an error and cannot download the newer files:

$ wget --no-check-certificate "http://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz"
URL transformed to HTTPS due to an HSTS policy
--2024-08-01 16:15:18--  https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz
Resolving carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)... 149.137.137.254
Connecting to carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)|149.137.137.254|:443... connected.
WARNING: cannot verify carla-assets.s3.us-east-005.backblazeb2.com's certificate, issued by ‘emailAddress=support@fortinet.com,CN=FG3H0E5819904539,OU=Certificate Authority,O=Fortinet,L=Sunnyvale,ST=California,C=US’:
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 403 Forbidden
2024-08-01 16:15:19 ERROR 403: Forbidden.

PatrickPromitzer, 2024-08-01:
try https://
```wget  "https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz"```

waldiwunder, 2024-08-02:
With `wget  "https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz"`:
--2024-08-02 08:39:51--  https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz
Resolving carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)... 149.137.137.254
Connecting to carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)|149.137.137.254|:443... connected.
ERROR: cannot verify carla-assets.s3.us-east-005.backblazeb2.com's certificate, issued by ‘emailAddress=support@fortinet.com,CN=FG3H0E5819904539,OU=Certificate Authority,O=Fortinet,L=Sunnyvale,ST=California,C=US’:
  Unable to locally verify the issuer's authority.
To connect to carla-assets.s3.us-east-005.backblazeb2.com insecurely, use `--no-check-certificate'.

`wget --no-check-certificate "https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz"`
--2024-08-02 08:40:47--  https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz
Resolving carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)... 149.137.137.254
Connecting to carla-assets.s3.us-east-005.backblazeb2.com (carla-assets.s3.us-east-005.backblazeb2.com)|149.137.137.254|:443... connected.
WARNING: cannot verify carla-assets.s3.us-east-005.backblazeb2.com's certificate, issued by ‘emailAddress=support@fortinet.com,CN=FG3H0E5819904539,OU=Certificate Authority,O=Fortinet,L=Sunnyvale,ST=California,C=US’:
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 403 Forbidden
2024-08-02 08:40:47 ERROR 403: Forbidden.

Unfortunately I have still the same error.

PatrickPromitzer, 2024-08-02:
It is strange because on my Computer it works.
It is a problem, but not that important for now.

[image]

Can you go to the file
path/to/carla/Until/BuildTools/enable_carsim_to_uproject.py
and edit this line
```python
    #line 30
    # Read uproject (json) file
    print(os.getcwd()) # new
    print(args.file) # new
    uproject_file = open(args.file, 'r')
    uproject_json = json.load(uproject_file)
    uproject_file.close()

```
I am not sure if a print() will show in the command line, and you may have to write it in a file to read it.
```
with open("/path/to/a/temp/folder/carla_print.txt", "w") as f:
    f.write("{0}\n{1}".format(os.getcwd(), str(args.file)))
```

With that path, look if the "CarlaUE4.uproject" exist.

waldiwunder, 2024-08-02:
I found a fix for the CarlaUE4.uproject error by copying the files directly from the github page. (I did git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla and copy the carla/Unreal folder from this github page and merged them). When I run make launch:

I get for the your suggested print lines:
`/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4
CarlaUE4.uproject
`
and 
`BuildCarlaUE4.sh: Build CarlaUE4 project.`

Unfortunately there is now a new error at a later stage:

```python
4 warnings generated.
[23/29] Compile CarlaUE4.cpp
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
6 warnings generated.
8 warnings generated.
3 warnings generated.
10 warnings generated.
15 warnings generated.
26 warnings generated.
14 warnings generated.
32 warnings generated.
[24/29] Link (lld) libUE4Editor-Carla.so
[25/29] Link (lld) libUE4Editor-StreetMapRuntime.so
[26/29] Link (lld) libUE4Editor-CarlaUE4.so
[27/29] Link (lld) libUE4Editor-StreetMapImporting.so
[28/29] Link (lld) libUE4Editor-CarlaTools.so
ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::AddObjKey(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced 137 more times

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::Add(double, int)
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/o-cwunder1/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
[... cut ...]
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 2

``` 

I tried today also to use the asstes from an installation of a different pc, but they seem to be outdated, a sensor was missing and also received an error.

I think the download problem is due to the network I am using (and all other pcs where I tried so far). This is weird, because some months ago I did not have this problem, but I do not know if something changed since then. I want to try in the weekend the download in a different network, maybe it works and solves my problem.

PatrickPromitzer, 2024-08-05:
Hi,
you could test if the file has a problem or not.
You can compare the hash of your file with the hash it should have
```
sha256sum 20231108_c5101a5.tar.gz
```
Hash:
(file from 14.12.2023)
158d15ac7ced5881b645244055af6117fcc521bca1af7b58a119cb28137a8906 20231108_c5101a5.tar.gz
(file from 27.06.2024; current)
9287d628547d4e925d25de72b97b61dc0f9173725286bf018449edb3eeecba8b 20231108_c5101a5.tar.gz

waldiwunder, 2024-08-06:
Hello! I was finally able to download 20231108_c5101a5.tar.gz in a different network. I made the suggested test, it is the correct file:

`sha256sum 20231108_c5101a5.tar.gz 
9287d628547d4e925d25de72b97b61dc0f9173725286bf018449edb3eeecba8b  20231108_c5101a5.tar.gz`

I remove the carla root folder and cloned again the carla repo with `git clone https://github.com/carla-simulator/carla`.

Then I created a folder `Unreal\CarlaUE4\Content\Carla` and unpacked the tar.gz file.

I executed make PythonAPI and built sucessfully carla.

Then I executed make launch. I get still the same error as above:

```python
python
4 warnings generated.
[23/29] Compile CarlaUE4.cpp
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
4 warnings generated.
8 warnings generated.
3 warnings generated.
6 warnings generated.
10 warnings generated.
15 warnings generated.
26 warnings generated.
14 warnings generated.
32 warnings generated.
[24/29] Link (lld) libUE4Editor-Carla.so
[25/29] Link (lld) libUE4Editor-StreetMapRuntime.so
[26/29] Link (lld) libUE4Editor-CarlaUE4.so
[27/29] Link (lld) libUE4Editor-StreetMapImporting.so
[28/29] Link (lld) libUE4Editor-CarlaTools.so
ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::AddObjKey(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced 137 more times

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::Add(double, int)
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::Meridian::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
[... cut ...]
>>>               metadata.cpp.o:(osgeo::proj::metadata::Identifier::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/user/UnrealEngine_4.26/Engine/Binaries/Linux/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 2
```

PatrickPromitzer, 2024-08-08:
Hi,
I can't tell if the error came from the version change of PROJ but it is possible.
The only other thing I have is this line in my .bashrc file
```
export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/bin/python3.8"
```

If this doesn't help, I need to go through my changes or test it myself.

sometimes it helps using this command again.
```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
```

btw. I use the Ubuntu 18.04 Software requirements instead of the Ubuntu 20.04 one.

waldiwunder, 2024-08-09:
Hi! Thanks for your patience and advice.

I added the line suggested by you to my .bashrc.
Sofar I compiled my code in a miniconda env. I tested now without the environemnt with Python3.8 from /usr/bin/python3.8.

I installed the Ubuntu 18.04 Software requirements. Still the program seems to use clang-10, maybe because the Unreal Engine was compiled with it. I did not touch the Unreal Engine installation sofar, if I execute ./UE4Editor, it opens sucessfully.

I cloned again the carla repo and unpacked the assets to have it fresh and clean and I built the PythonAPI (with the edits from #8006). I tested if it works without editing Setup.sh, but it doesn't.

When I execute make launch, I get the same error as in my last comment. I am sorry, but I am still not able to launch Carla.

I wonder if I should still try to use a vscode dockercontainer like in #6776, because I see not much hope to resolve the error.

Thanks for the support. I am thankful for further advice and help.

FengD, 2024-09-10:
I have exectly the same problem as mentioned above. I am in Ubuntu 22.04. With Conda Env, python 3.8.
The UE could be opened successfully. the lastest dev and tag 0.9.15.2 has the same problem above.

I tried to clone a PROJ project and build the libproj.a by myself, and there is no difference. 

```
[48/57] Link (lld) libUE4Editor-CarlaExporter.so
15 warnings generated.
31 warnings generated.
14 warnings generated.
[49/57] Link (lld) libUE4Editor-HoudiniEngineRuntime.so
[50/57] Link (lld) libUE4Editor-HoudiniEngine.so
[51/57] Link (lld) libUE4Editor-HoudiniEngineEditor.so
14 warnings generated.
[52/57] Link (lld) libUE4Editor-Carla.so
[53/57] Link (lld) libUE4Editor-StreetMapRuntime.so
[54/57] Link (lld) libUE4Editor-CarlaUE4.so
[55/57] Link (lld) libUE4Editor-StreetMapImporting.so
[56/57] Link (lld) libUE4Editor-CarlaTools.so
ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::AddObjKey(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationMethod::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced 114 more times

ld.lld: error: undefined symbol: osgeo::proj::CPLJSonStreamingWriter::Add(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationMethod::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by singleoperation.cpp
>>>               singleoperation.cpp.o:(osgeo::proj::operation::OperationParameterValue::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
>>>               coordinatesystem.cpp.o:(osgeo::proj::cs::CoordinateSystemAxis::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
>>> referenced by coordinatesystem.cpp
[... cut ...]
>>> referenced by metadata.cpp
>>>               metadata.cpp.o:(osgeo::proj::metadata::Identifier::_exportToJSON(osgeo::proj::io::JSONFormatter*) const) in archive /home/ding/Documents/sim/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/ding/Documents/sim/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 2

```