carla-simulator/carla #8014: Make launch时我遇到 fatal error C1083:  无法打开包括文件: “Editor/FoliageEdit/Public/FoliageEdMode.h”: No such file or directory make: *** [CarlaUE4Editor] 错误 6

King8139, 2024-08-02:


King8139, 2024-08-02:
这是我运行过程：
:\carla>make launch

D:\carla>setlocal enabledelayedexpansion

D:\carla>rem Run it through a cmd with the x64 Visual C++ Toolset enabled.

D:\carla>set LOCAL_PATH=D:\carla\Util\BuildTools\

D:\carla>set FILE_N=-[BuildUE4Plugins]:

D:\carla>rem Print batch params (debug purpose)

D:\carla>echo -[BuildUE4Plugins]: [Batch params]: --build --all
-[BuildUE4Plugins]: [Batch params]: --build --all

D:\carla>rem ============================================================================

D:\carla>rem -- Parse arguments ---------------------------------------------------------

D:\carla>rem ============================================================================

D:\carla>set DOC_STRING=Build LibCarla.

D:\carla>set USAGE_STRING=Usage: -[BuildUE4Plugins]: [-h|--help] [--rebuild] [--build] [--clean] [--no-pull]

D:\carla>set BUILD_STREETMAP=false

D:\carla>set GIT_PULL=true

D:\carla>set CURRENT_STREETMAP_COMMIT=260273d6b7c3f28988cda31fd33441de7e272958

D:\carla>set STREETMAP_BRANCH=master

D:\carla>set STREETMAP_REPO=https://github.com/carla-simulator/StreetMap.git

D:\carla>if not "--build" == "" (
if "--build" == "--rebuild" (
set REMOVE_INTERMEDIATE=true
 set BUILD_STREETMAP=true
)
 if "--build" == "--build" (set BUILD_STREETMAP=true )
 if "--build" == "--no-pull" (set GIT_PULL=false )
 if "--build" == "--clean" (set REMOVE_INTERMEDIATE=true )
 if "--build" == "-h" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
)
 if "--build" == "--help" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
)
 shift
 goto :arg-parse
)

D:\carla>if not "--all" == "" (
if "--all" == "--rebuild" (
set REMOVE_INTERMEDIATE=true
 set BUILD_STREETMAP=true
)
 if "--all" == "--build" (set BUILD_STREETMAP=true )
 if "--all" == "--no-pull" (set GIT_PULL=false )
 if "--all" == "--clean" (set REMOVE_INTERMEDIATE=true )
 if "--all" == "-h" (
echo Build LibCarla.
 echo Usage: -[BuildUE4Plugins]: [-h  | --help] [--rebuild] [--build] [--clean] [--no-pull]
 GOTO :eof
[... 302 lines cut ...]
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(73): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(189): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um\winnt.h(620): warning C4005: “TEXT”: 宏重定义
  D:\UE_4.26\Engine\Source\Runtime\Core\Public\HAL/Platform.h(1082): note: 参见“TEXT”的前一个定义
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: 无法打开包括文件: “Renderer/Public/GBufferView.h”: No such file or directory
  [9/17] Module.Carla.2_of_4.cpp
  C:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um\winnt.h(620): warning C4005: “TEXT”: 宏重定义
  D:\UE_4.26\Engine\Source\Runtime\Core\Public\HAL/Platform.h(1082): note: 参见“TEXT”的前一个定义
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(120): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(148): warning C4996: 'UVehicleWheel': PhysX is deprecated. Use the UChaosVehicleWheel from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(73): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(189): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/SoilTypeManager.cpp(26): warning C4458: “LargeMapManager”的声明隐藏了类成员
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\MapGen/SoilTypeManager.h(54): note: 参见“ASoilTypeManager::LargeMapManager”的声明
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: 无法打开包括文件: “Renderer/Public/GBufferView.h”: No such file or directory
  [10/17] Module.CarlaTools.cpp
  C:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um\winnt.h(620): warning C4005: “TEXT”: 宏重定义
[... cut ...]
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(189): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/MapGeneratorWidget.cpp(16): fatal error C1083:  无法打开包括文件: “Editor/FoliageEdit/Public/FoliageEdMode.h”: No such file or directory
make: *** [CarlaUE4Editor] 错误 6
有没有大神解决

MarcelPiNacy-CVC, 2024-08-30:
Hi,

You don't have the correct version of CARLA's fork of UE.

donghaiwang, 2024-09-02:
> Hi,
> 
> You don't have the correct version of CARLA's fork of UE.

Using the [latest CarlaEngine](https://github.com/CarlaUnreal/UnrealEngine) , the same error is reported.

MarcelPiNacy-CVC, 2024-09-02:
Are you sure you're on the carla branch and have the environment variable UE4_ROOT set?

donghaiwang, 2024-09-02:
> Are you sure you're on the carla branch and have the environment variable UE4_ROOT set?

Thank you. I have muti-version UE. Switching UE version through right click .uproject file to solve it.

closed 2024-09-03 by MarcelPiNacy-CVC