carla-simulator/carla #8024: Carla encountered an error loading xodr file (missing spiral segment)

Mayuemei, 2024-08-02:
CARLA version:0.9.15

Platform/OS:windows

Problem you have experienced:

I generated an xodr file using [https://odrviewer.io/](url) After loading, it can display normally, as shown in the figure below. However, when Carla loads, it crashes. It is speculated that this is due to an error in the road topology in the xodr file. I printed the road points and found that they were missing in the position of the spiral segment, as shown in the figure.May I ask if Carla's requirements for loading xodr are not completely consistent with the opendrive format, or if there are errors in my xodr file?

What you expected to happen:

The map successfully loaded through [https://odrviewer.io/](url) can also be loaded correctly on Carla
[image]
[image]
[image]
`<?xml version="1.0" encoding="utf-8"?>
<OpenDRIVE>
    <header date="2024-08-02 17:10:27.384713" east="0.0" name="myroad" north="0.0" revMajor="1" revMinor="5" south="0.0" west="0.0"/>
    <road id="1" junction="-1" length="1486.035381584003" rule="RHT">
        <link/>
        <planView>
            <geometry hdg="0" length="5" s="0" x="0" y="0">
                <line/>
            </geometry>
            <geometry hdg="0" length="180.0" s="5" x="5.0" y="0.0">
                <spiral curvEnd="0" curvStart="0"/>
            </geometry>
            <geometry hdg="0.0" length="380" s="185.0" x="185.0" y="0.0">
                <line/>
            </geometry>
            <geometry hdg="0.0" length="180.0" s="565.0" x="565.0" y="0.0">
                <spiral curvEnd="0" curvStart="0"/>
            </geometry>
            <geometry hdg="0.0" length="365" s="745.0" x="745.0" y="0.0">
                <line/>
            </geometry>
            <geometry hdg="0.0" length="180.0" s="1110.0" x="1110.0" y="0.0">
                <spiral curvEnd="0.003205128205128205" curvStart="0"/>
            </geometry>
            <geometry hdg="0.28846153846153844" length="196.0353815840031" s="1290.0" x="1288.507977796179" y="17.205094690461763">
                <arc curvature="0.003205128205128205"/>
            </geometry>
        </planView>
        <elevationProfile/>
        <lateralProfile/>
        <lanes>
            <laneSection s="0">
                <left>
                    <lane id="1" level="false" type="driving">
                        <link/>
                        <width a="5" b="0" c="0" d="0" sOffset="0"/>
                        <roadMark color="standard" height="0.02" sOffset="0" type="solid" weight="standard" width="0.2"/>
                    </lane>
                </left>
                <center>
                    <lane id="0" level="false" type="none">
                        <roadMark color="standard" height="0.02" sOffset="0" type="solid" weight="standard" width="0.2"/>
                    </lane>
                </center>
                <right>
                    <lane id="-1" level="false" type="driving">
                        <link/>
                        <width a="5" b="0" c="0" d="0" sOffset="0"/>
                        <roadMark color="standard" height="0.02" sOffset="0" type="solid" weight="standard" width="0.2"/>
                    </lane>
                </right>
            </laneSection>
        </lanes>
    </road>
</OpenDRIVE>
`

PatrickPromitzer, 2024-08-05:
Hi,
can you try changing
```
<spiral curvEnd="0" curvStart="0"/>
```
to 
```
<line/>
```

M-Colley, 2026-10-26:
Draft PR #10017 proposes a `ue58-dev` fix for constant-curvature spiral geometry (`curvStart == curvEnd`), with native geometry regressions. It addresses that degenerate spiral calculation; the complete reported OpenDRIVE file and live Unreal map-loading behavior remain unverified.