carla-simulator/carla #8026: Road Surface Appears Gray and Hazy After Importing from MATLAB RoadRunner

chuanye-Wang, 2024-08-04:
carla-dev
ubuntu 20.04

Description of the Issue:
After importing the road created using MATLAB RoadRunner into CARLA, the road surface appears to be gray and hazy. I'm not sure if this is a rendering issue.

Steps to Reproduce:

Create a road model using MATLAB RoadRunner.
Export the created road model in a format compatible with CARLA.
Load the exported road model into CARLA.
Observe the display of the road surface.
Expected Behavior:
The road surface should appear clear and as expected from the creation.

Actual Behavior:
The road surface appears gray and hazy, lacking clarity, and it's uncertain if this is a rendering issue.
[image]

PatrickPromitzer, 2024-08-05:
Hi,
with a flat surface it worked for me, but after changing the Z value of some polygons I got a similar effect.

Shadows are darker at some places and around the Z value changes it has a weird texture.
It could happen with a flat surface, too, but I can't confirm it it does.

It is only a guess, but I believe it is not only the RoadRunner texture with this behavior. 
[image]
[image]
[image]
The pictures are from a custom map we created for our project.

chuanye-Wang, 2024-08-06:
Thanks. Did you figure out how to solve this problem, and make it look normal?

PatrickPromitzer, 2024-08-06:
Hi,
I didn't fix it, but my guess is that it is an generated texture, and to have a good quality you have to create/use your own texture.
Another problem could be, that UnrealEngine has a problem with shadows or the quality of the picture is too low.

Maybe someone with more experience with RoadRunner can answer that.

chuanye-Wang, 2024-08-06:
Sounds good, I'll try. Thanks a lot

jiaery, 2024-10-19:
[image]

hi!  When I export the created road model in a CARLA compatible format, it shows the following road, am I missing that step? How can I export the same road model as you, and I look forward to your reply

PatrickPromitzer, 2024-10-21:
Hi,
did you Import the .xodr + .fbx with this steps?
https://carla.readthedocs.io/en/0.9.15/tuto_M_add_map_source/