carla-simulator/carla #8027: Traffic Light

hjg857, 2024-08-04:
CARLA version:0.9.15
Platform/OS:Windows11
Problem you have experienced:I'd like to know what the default duration of the traffic lights at Carla Town02 is.

ahlamiki, 2024-08-09:
Hi @hjg857 The default duration of the traffic lights at Carla Town02 is 10 s. You can check it through the BP_TrafficLight as you can see in the attached picture.

[image]

DnDare, 2024-08-09:
Trafficlights have the same timing in all towns.
You can do as @ahlamiki says,

There are other older issues related to this topic that you can also review:

https://github.com/carla-simulator/carla/issues/6932
https://github.com/carla-simulator/carla/issues/6121
https://github.com/carla-simulator/carla/issues/7625