carla-simulator/carla #8028: windows11在make launch启动Carla0.9.15时加载到75%时提示fatal error

King8139, 2024-08-04:


King8139, 2024-08-04:
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x0000000000000038

0x00007ffa86ecccb5 UE4Editor-Carla.dll!ULoadAssetMaterialsCommandlet::ULoadAssetMaterialsCommandlet() [D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Commandlet\LoadAssetMaterialsCommandlet.cpp:38]
0x00007ffae032bd53 UE4Editor-CoreUObject.dll!UClass::CreateDefaultObject() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\Class.cpp:3707]
0x00007ffae05ee0de UE4Editor-CoreUObject.dll!UObjectLoadAllCompiledInDefaultProperties() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\UObjectBase.cpp:908]
0x00007ffae05d14ff UE4Editor-CoreUObject.dll!ProcessNewlyLoadedUObjects() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\CoreUObject\Private\UObject\UObjectBase.cpp:997]
0x00007ffae0501745 UE4Editor-CoreUObject.dll!TBaseStaticDelegateInstance<void __cdecl(FName,bool),FDefaultDelegateUserPolicy>::ExecuteIfSafe() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:731]
0x00007ffaddd19a0a UE4Editor-Core.dll!TMulticastDelegate<void __cdecl(FName,bool),FDefaultDelegateUserPolicy>::Broadcast() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateSignatureImpl.inl:955]
0x00007ffaddd41737 UE4Editor-Core.dll!FModuleManager::LoadModuleWithFailureReason() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Core\Private\Modules\ModuleManager.cpp:519]
0x00007ffb1d55e6e3 UE4Editor-Projects.dll!FModuleDescriptor::LoadModulesForPhase() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Projects\Private\ModuleDescriptor.cpp:561]
0x00007ffb1d57320a UE4Editor-Projects.dll!FPluginManager::TryLoadModulesForPlugin() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Projects\Private\PluginManager.cpp:1295]
0x00007ffb1d55e4e0 UE4Editor-Projects.dll!FPluginManager::LoadModulesForEnabledPlugins() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Projects\Private\PluginManager.cpp:1369]
0x00007ff72a6652c7 UE4Editor.exe!FEngineLoop::LoadStartupModules() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:3808]
0x00007ff72a66874d UE4Editor.exe!FEngineLoop::PreInitPostStartupScreen() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:3198]
0x00007ff72a660e6d UE4Editor.exe!GuardedMain() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Launch\Private\Launch.cpp:127]
0x00007ff72a6611ca UE4Editor.exe!GuardedMainWrapper() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:137]
0x00007ff72a674e5d UE4Editor.exe!WinMain() [D:\UE_4.26\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:268]
0x00007ff72a677ba6 UE4Editor.exe!__scrt_common_main_seh() [D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
0x00007ffb83bb257d KERNEL32.DLL!UnknownFunction []
0x00007ffb84d2af28 ntdll.dll!UnknownFunction []
这个问题怎么解决，加急！！！！

MarcelPiNacy-CVC, 2024-08-30:
Hi,

This could be an issue with Content, make sure that you have correctly downloaded it as it appears that UE can't find an asset.

closed 2024-09-13 by MarcelPiNacy-CVC

JJ-passion, 2024-12-25:
Fatal error!

Unhandled Exception: 0xe06d7363

0x00007ff8ddabfb4c KERNELBASE.dll!UnknownFunction []
0x00007ff8be885267 VCRUNTIME140.dll!UnknownFunction []
0x00007ff6b9008e62 CarlaUE4-Win64-Shipping.exe!clmdep_asio::detail::throw_exception<std::system_error>() []
0x00007ff6b900c720 CarlaUE4-Win64-Shipping.exe!clmdep_asio::detail::do_throw_error() []
0x00007ff6b9009257 CarlaUE4-Win64-Shipping.exe!clmdep_asio::basic_socket_acceptor<clmdep_asio::ip::tcp,clmdep_asio::socket_acceptor_service<clmdep_asio::ip::tcp> >::basic_socket_acceptor<clmdep_asio::ip::tcp,clmdep_asio::socket_acceptor_service<clmdep_asio::ip::tcp> >() []
0x00007ff6b9009726 CarlaUE4-Win64-Shipping.exe!rpc::server::server() []
0x00007ff6b738e294 CarlaUE4-Win64-Shipping.exe!FCarlaServer::FPimpl::FPimpl() [c:\jenkins\workspace\carla_0.9.11\unreal\carlaue4\plugins\carla\source\carla\server\carlaserver.cpp:78]
0x00007ff6b73fe6a1 CarlaUE4-Win64-Shipping.exe!FCarlaEngine::NotifyInitGame() [c:\jenkins\workspace\carla_0.9.11\unreal\carlaue4\plugins\carla\source\carla\game\carlaengine.cpp:64]
0x00007ff6b73eb525 CarlaUE4-Win64-Shipping.exe!ACarlaGameModeBase::InitGame() [c:\jenkins\workspace\carla_0.9.11\unreal\carlaue4\plugins\carla\source\carla\game\carlagamemodebase.cpp:104]
0x00007ff6b8c9e63b CarlaUE4-Win64-Shipping.exe!UWorld::InitializeActorsForPlay() [c:\program files\epic games\ue_4.24\engine\source\runtime\engine\private\world.cpp:4029]
0x00007ff6b8c56fca CarlaUE4-Win64-Shipping.exe!UEngine::LoadMap() [c:\program files\epic games\ue_4.24\engine\source\runtime\engine\private\unrealengine.cpp:12730]
0x00007ff6b8c492b3 CarlaUE4-Win64-Shipping.exe!UEngine::Browse() [c:\program files\epic games\ue_4.24\engine\source\runtime\engine\private\unrealengine.cpp:11975]
0x00007ff6b887b928 CarlaUE4-Win64-Shipping.exe!UGameInstance::StartGameInstance() [c:\program files\epic games\ue_4.24\engine\source\runtime\engine\private\gameinstance.cpp:534]
0x00007ff6b731dac0 CarlaUE4-Win64-Shipping.exe!FEngineLoop::Init() [c:\program files\epic games\ue_4.24\engine\source\runtime\launch\private\launchengineloop.cpp:3695]
0x00007ff6b7324757 CarlaUE4-Win64-Shipping.exe!GuardedMain() [c:\program files\epic games\ue_4.24\engine\source\runtime\launch\private\launch.cpp:159]
0x00007ff6b732487a CarlaUE4-Win64-Shipping.exe!GuardedMainWrapper() [c:\program files\epic games\ue_4.24\engine\source\runtime\launch\private\windows\launchwindows.cpp:134]
0x00007ff6b732e20b CarlaUE4-Win64-Shipping.exe!WinMain() [c:\program files\epic games\ue_4.24\engine\source\runtime\launch\private\windows\launchwindows.cpp:263]
0x00007ff6b954ecfe CarlaUE4-Win64-Shipping.exe!__scrt_common_main_seh() [d:\agent\_work\2\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
0x00007ff8df98259d KERNEL32.DLL!UnknownFunction []
0x00007ff8e07eaf38 ntdll.dll!UnknownFunction []

I met the same problem, how to resolve it?