carla-simulator/carla #8029: CarlaUE4.sh cannot launch

yuchen-he, 2024-08-05:
After run the command `./CarlaUE4.sh -RenderOffScreen -nosound`, Carla simulator exited without any errors.
[image]

Basic information of my environment as below:
- CARLA version: 0.9.12 and 0.9.15
- Platform/OS: Ubuntu18.04 and Ubuntu 20.04 in a docker container
- Problem you have experienced: CarlaUE4.sh cannot launch
- What you expected to happen: The simulator can start and hold, then I can see some GPU usage
- Steps to reproduce: Build a docker container, then install Carla by CARLA_0.9.15.tar.gz, then run CarlaUE4.sh

PatrickPromitzer, 2024-08-05:
Hi,
can you share the Dockerfile?

If you run Carla in a docker, you need more than just the base Ubuntu image.

You can use the official docker image 
https://hub.docker.com/r/carlasim/carla/tags

Make sure you installed the nvidia-docker2 package
https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html

Carla docu
https://carla.readthedocs.io/en/0.9.15/build_docker/

yuchen-he, 2024-08-05:
> Hi, can you share the Dockerfile?
>
> If you run Carla in a docker, you need more than just the base Ubuntu image.
>
> You can use the official docker image https://hub.docker.com/r/carlasim/carla/tags [...]

Thank you for your advice. 
I run Carla in a docker with base Ubuntu 18.04 and CUDA 11.1. I guess this environment is not sufficient for Carla.
However, I have some problem about downloading an official docker image since I am in a network with proxy.
Do you have any other suggestions or could you tell me about how to build an environment in docker for Carla?

PatrickPromitzer, 2024-08-05:
I don't know if it works, but you can try this Dockerfile.
```
FROM ubuntu:20.04

USER root

ENV DEBIAN_FRONTEND=noninteractive
RUN apt-get update ; \
  apt-get install -y wget software-properties-common && \
  add-apt-repository ppa:ubuntu-toolchain-r/test && \
  wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key|apt-key add - && \
  apt-get update ; \
  apt-get install -y build-essential \
    libvulkan1 \
    python3.8 && \
  apt-get update ; \
    apt-get install -y python3.8-dev \
    python3-pip \
    python3.8-distutils \
    libpng-dev \
    libtiff5-dev \
    libjpeg-dev \
    tzdata \
    sed \
    curl \
    unzip \
    autoconf \
    libtool \
    rsync \
    libxml2-dev \
    git \
    aria2 && \
    apt-get update ;  \
    apt-get install -y libsdl2-2.0-0 xserver-xorg libvulkan1 libomp5 vulkan-utils && \
  python3.8 -m pip install -Iv setuptools==47.3.1 && \
  python3.8 -m pip install distro && \
  python3.8 -m pip install py-cpuinfo && \
  python3.8 -m pip install pygame && \
  python3.8 -m pip install python-tr && \
  python3.8 -m pip install numpy==1.18.4 && \
  python3.8 -m pip install wheel auditwheel  && \
  python3.8 -m pip install future  && \
  python3.8 -m pip install open3d

RUN useradd -m carla
USER carla
WORKDIR /home/carla
```

yuchen-he, 2024-08-06:
> I don't know if it works, but you can try this Dockerfile.
>
> ```
> FROM ubuntu:20.04
>
> USER root
>
> ENV DEBIAN_FRONTEND=noninteractive
> RUN apt-get update ; \
>   apt-get install -y wget [...]

Thank you so much for the suggestions.
I have tried this solution, but Carla still exits as before. Maybe I should use the official docker images for running Carla in a docker container.

I took a workaround by starting Carla server outside a docker container and tried a connect to it and run my algorithm inside the container. Anyway, it works currently.

CoderXuans, 2024-08-07:
@yuchen-he 请问在Docker容器外启动Carla服务器，并尝试在容器内连接并运行我的算法，你是如何实现的呢？
我是租了两张显卡，也属于Docker容器吧？

CoderXuans, 2024-08-07:
@yuchen-he How can I start a CARLA server outside a Docker container and run my algorithm inside the container? I have rented two GPUs, so I am also using Docker containers, right?