carla-simulator/carla #8030: Crash when package branch ue5-dev on Ubuntu22.04

anschen1994, 2024-08-05:
# Crash when package with branch dev-ue5 on Ubuntu22.04

Thanks the community for the contribution of Carla with UE5.
We follow the instruction: https://github.com/carla-simulator/carla/blob/ue5-dev/UE5_BUILD_README.md step by step and all operations are on Ubuntu22.04.
We can build successfully until the package stage. Specificly, 
1. Configure; 
2. Build Carla;
3. Build and install the Python API
4. Launch the Editor 
The above four procedures have no problems.

The last step --- package can not build successfully.
The ERROR Log are shown in the attached file.

Can anyone help us fix this problem?

anschen1994, 2024-08-05:
The Log is archived in the following content. 
[Log.zip](https://github.com/user-attachments/files/16491087/Log.zip)

The core trace info is shown in the figure:
[image]

qhaas, 2024-08-23:
I'm unable to `cmake --build Build --target package` as well in Ubuntu 22.04.  [Following the documentation](https://carla.readthedocs.io/en/latest/build_linux_ue5/)  results in `LogInit: Display: Failure - 26616 error(s), 217757 warning(s)`

Source versions:
* [carla's Unreal Engine fork, ue5-dev-carla branch, commit f1c3b5d77ccf403e25b59f6735157ccf5e3961e7](https://github.com/CarlaUnreal/UnrealEngine/commit/f1c3b5d77ccf403e25b59f6735157ccf5e3961e7)
* [carla ue5-dev branch, commit f1c3b5d77ccf403e25b59f6735157ccf5e3961e7](https://github.com/carla-simulator/carla/commit/aa9d95eb21d03abe143471d9a7e854a794e5533d)
* [carla content ue5-dev branch, commit 3929df3027f2d1a97b28e3f401942d69c47666cd](https://bitbucket.org/carla-simulator/carla-content/commits/3929df3027f2d1a97b28e3f401942d69c47666cd)

Unreal Engine itself, `cmake --build Build`, `cmake --build Build --target carla-python-api-install`, and `cmake --build Build --target launch` work, but not the final `cmake --build Build --target package` to create the packaged release.

I assume the docs mean [`CarlaSetup.sh`](https://github.com/carla-simulator/carla/blob/ue5-dev/CarlaSetup.sh) instead of `Setup.sh`.  I had to install additional dependencies beyond what is deployed by that script to get this far:
* libwebsockets-dev
* libnss3-dev
* libatk-bridge2.0-dev
* libpango1.0-dev \
* libsdl2-dev

Curiously, started getting errors about too my FDs open after deploying `libwebsockets-dev`, so used [ulimits](https://access.redhat.com/solutions/61334) to cap `nproc` / `nofile` to 1024 based on a hint I found online with another project.

I also set `CARLA_UNREAL_ENGINE_PATH` and `CARLA_PATH` to the root of unreal engine and carla respectively.

I suspect that the asset content commit is out-of-sync with the carla src commit based on many warnings like this:
```
LogInit: Display: Warning/Error Summary (Unique only)
LogInit: Display: -----------------------------------
LogInit: Display: LogEditorAssetSubsystem: Error: GetPathNameForLoadedAsset failed: The Asset is not valid.
LogInit: Display: LogEditorAssetSubsystem: Error: LoadAsset failed: Can't convert the path '/NonL.L' because it does not map to a root.
LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
LogInit: Display: LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
LogInit: Display: LogUObjectGlobals: Warning: [AssetLog] /mnt/carla-src/Unreal/CarlaUnreal/Plugins/CarlaTools/Content/OnroadMapGenerator/BP_OpenDriveToMap.uasset: Failed to load '/Game/Carla/Static/Static/SM_LightBox2': Can't find file.
...
LogInit: Display: LogScript: Warning: Script Msg: Attempted to access index 0 from array 'CallFunc_GetAllActorsOfClassWithTag_OutActors_1' of length 0 in '/Game/Carla/Blueprints/BP_GeneralSceneSettings.BP_GeneralSceneSettings_C:ExecuteUbergraph_BP_GeneralSceneSettings'!
LogInit: Display: LogScript: Warning: Script Msg called by: BP_GeneralSceneSettings_C /Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt:PersistentLevel.BP_GeneralSceneSettings_C_1
```

Blyron, 2024-11-06:
Without modifying any cmakelists or cmake scripts. You should be able to launch 
cmake --build Build --target package.
Make sure you update latest content too

closed 2024-11-23 by Blyron