carla-simulator/carla #8037: Cmake issue what should i do?

Vishvamunjani19, 2024-08-07:
C:\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  24
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
The system cannot find the path specified.
        0 file(s) copied.
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called "rpclib-src".
    -[install_rpclib]: Generating build...
CMake Error at CMakeLists.txt:2 (project):
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.
    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Windows "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

Usotsu7ki, 2024-10-11:
Have you solved bro？

Vishvamunjani19, 2024-10-15:
Hello,

Sorry for misunderstanding I am a girl. Now I have problem in installation
of Carla and not generate the generate_traffic.py  file.

You have idea what should I do?

Thank you for concern

Best regards,
Vishva