carla-simulator/carla #8039: "Link [x64] UnrealEditor-WebBrowser.lib" build fail

Augustusmyc, 2024-08-08:
CARLA version: dev + UE5,   build from source
Platform/OS: Windows 11
Problem you have experienced:  After 2-3 hours by the last command "cmake --build Build --target launch", the build failed
What you expected to happen: successful build
Steps to reproduce:   cmake --build Build --target launch
Other information (documentation you consulted, workarounds you tried):

Every step same with https://carla.readthedocs.io/en/latest/build_windows_ue5/
The error information:
[image]

visual studio 2022 install
[image]

MarcelPiNacy-CVC, 2024-08-30:
Hi,

Sorry for the delay, could you share the entire build log?

Augustusmyc, 2024-09-13:
Thanks for your response. The above errors disappear after official Epic Games Launcher was installed.

However, only default map(Town10HD_Opt) can be normally rendered in my ue5 editor, like this:
[image]

After change to other maps, like Town3:

[image]
[image]

No light mode(alt+3):

[image]

Here is my vs components:
[image]

I doult that some error happens during the install process. See the logs below:

***********##########
First step
***********##########

call cmake -G Ninja -S . -B Build -DCMAKE_BUILD_TYPE=Release -DBUILD_CARLA_UNREAL=ON -DCARLA_UNREAL_ENGINE_PATH=%CARLA_UNREAL_ENGINE_PATH% || exit /b:

......

[1425 bytes data]

  [1434 bytes data]

  [1425 bytes data]

  [1434 bytes data]

  [2859 bytes data]

  [1434 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [1434 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [1186 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [1434 bytes data]

  [2859 bytes data]

  [2859 bytes data]

  [1434 bytes data]

  [2859 bytes data]
[... 9292 lines cut ...]
D:\MycProjects\CarlaProjects\CarlaUE5\LibCarla\source\carla\AtomicSharedPtr.h(27): note: 在编译 类 模板 成员函数“void carla::AtomicSharedPtr<carla::streaming::detail::Session>::store(std::shared_ptr<_Ty>) noexcept”时
        with
        [
            _Ty=carla::streaming::detail::tcp::ServerSession
        ]
D:\MycProjects\CarlaProjects\CarlaUE5\LibCarla\source\carla\streaming\detail\MultiStreamState.h(87): note: 请参阅 "carla::streaming::detail::MultiStreamState::ConnectSession" 中对 "carla::AtomicSharedPtr<carla::streaming::detail::Session>::store" 的第一个引用
D:\MycProjects\CarlaProjects\CarlaUE5\LibCarla\source\carla\streaming\detail\MultiStreamState.h(137): note: 查看对正在编译的 类 模板 实例化“carla::AtomicSharedPtr<carla::streaming::detail::Session>”的引用
D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Source\CarlaTools\Private\MapGeneratorWidget.cpp(1284) : warning C4701:  使用了可能未初始化的局部变量“NumOfTilesForSpreadedActorsGrid”
[30/35] Link [x64] UnrealEditor-CarlaTools.lib
  正在创建库 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.lib 和对象 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.exp
[31/35] Link [x64] UnrealEditor-CarlaUnreal.dll
  正在创建库 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaUnreal\UnrealEditor-CarlaUnreal.sup.lib 和对象 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaUnreal\UnrealEditor-CarlaUnreal.sup.exp
[32/35] Link [x64] UnrealEditor-StreetMapRuntime.dll
  正在创建库 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\StreetMap\Intermediate\Build\Win64\x64\UnrealEditor\Development\StreetMapRuntime\UnrealEditor-StreetMapRuntime.sup.lib 和对象 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\StreetMap\Intermediate\Build\Win64\x64\UnrealEditor\Development\StreetMapRuntime\UnrealEditor-StreetMapRuntime.sup.exp
[33/35] Link [x64] UnrealEditor-CarlaTools.dll
  正在创建库 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.sup.lib 和对象 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\CarlaTools\Intermediate\Build\Win64\x64\UnrealEditor\Development\CarlaTools\UnrealEditor-CarlaTools.sup.exp
[34/35] Link [x64] UnrealEditor-Carla.dll
  正在创建库 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Intermediate\Build\Win64\x64\UnrealEditor\Development\Carla\UnrealEditor-Carla.sup.lib 和对象 D:\MycProjects\CarlaProjects\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Intermediate\Build\Win64\x64\UnrealEditor\Development\Carla\UnrealEditor-Carla.sup.exp
[35/35] WriteMetadata CarlaUnrealEditor.target
Total time in Parallel executor: 798.20 seconds
Total execution time: 959.79 seconds

[3/4] Launching Carla Unreal...

After "[3/4] Launching Carla Unreal..." the Epic Games Launcher starts, and After close the Launcher, the install process stops.

Augustusmyc, 2024-09-13:
like to add, accrording to https://github.com/carla-simulator/carla/pull/8033,

I add "check_cxx_compiler_flag(/utf-8 HAS_MSVC_UTF8)" in "D:\MycProjects\CarlaProjects\CarlaUE5\CMake\Common.cmake", like this: 

...
set (CMAKE_C_STANDARD 11)
set (CMAKE_C_STANDARD_REQUIRED ON)

set (CMAKE_POSITION_INDEPENDENT_CODE ON)

check_cxx_compiler_flag(/utf-8 HAS_MSVC_UTF8)
if (HAS_MSVC_UTF8)
  add_compile_options (/utf-8)
endif ()

if (CMAKE_BUILD_TYPE STREQUAL "Debug")
  carla_warning (
    "CARLA is set to be built in Debug mode. This may cause issues when building CarlaUnrealEditor."
  )
endif ()
...

MarcelPiNacy-CVC, 2024-09-13:
Right, we currently do support towns other than 10 currently for UE5.

MarcelPiNacy-CVC, 2024-09-13:
The log you shared looks fine, we do need to clean up warnings, but it seems to be working well.
Btw please attach logs as files. Beyond being easier to read, you may ping random github users if the log happens to have `@` characters, which isn't unusual.

closed 2024-09-24 by MarcelPiNacy-CVC