carla-simulator/carla #8041: Add texture to buildings

chuanye-Wang, 2024-08-08:
CARLA-dev  0.9.15
Ubuntu 20.04

issue: Buildings in map are simply grey Cuboid.

expect: I expect it to have the same effect as digital-twin-tool.

Details:
I want to use the building infomation in .osm file(downloaded from Openstreetmap), So I import the osm file into blender, and export the blender project as .fbx file. Then I copy the .fbx file into carla Import dir, and run "make import". Here's what I got:

The buildings are simply grey Cuboid.
[image]

But if I use carla digital-twin-tool, The buildings are exquisite like this:
[image]

How can I get the same result but dont use digital-twin-tool?

PatrickPromitzer, 2024-08-08:
Hi,
do the building have that gray look in blender, too?
The building are gray because the Material is missing.
I can't say if the problem is the import or they never had an Material in the first place.

It is hard to reproduce it because I would need to know the blender package you used for creating the Objects.

You can import the .fbx file with drag and drop into the file explorer of Unreal Engine.
In that way, it will tell you import options.

chuanye-Wang, 2024-08-08:
yeah, it has the same grey look in blender as well:
[image]

what I did is import the osm file into blender with a GIS plugin (in red circle). Because of the digital-twin-tool utilize the osm file as well, so I guess it is the digital-twin-tool that add meterial to the building, but I didnt know how it works yet.

Anyway, I just want to make my building more beautiful, any solutions? really need help, thanks a lot

PatrickPromitzer, 2024-08-08:
You have to model the building yourself if you want a less block like look.
If you want a single color, you could make a simple Material or use one from Unreal Engine.
If you want complex and colorful buildings, you have to crate them yourself. 

You can try finding a different plugin, but it looks like without modeling the building yourself, this plugin gives you what you see.

chuanye-Wang, 2024-08-08:
You're right. I wander is it any way to use some part of the digital-twin-tool. So that it can automatically generate models for different shape of buildings

PatrickPromitzer, 2024-08-08:
You can try exporting the buildings into a .fbx file.
On the right side should be a list of items in the scene. If you click on a building in the preview screen, it should be selected.
I am not sure if it work with the digital-twin tool, but I remember seeing a "export map" option.

RodrigoVillarias, 2024-08-09:
Hello, as already commented the problem seems to be that the buildings you import have no complexity and only a basic material, on the subject of digital twins to generate individual models I'll see if I can talk to who is responsible for making the tool, but you can try to find the option to export the fbx to see if it is available, if not the other solution is to model your some buildings with their textures and add them to the scene

chuanye-Wang, 2024-08-09:
Thanks a lot, looking forwered to that.

closed 2024-08-16 by MarcelPiNacy-CVC