carla-simulator/carla #8042: Confusion about recommended Ubuntu version to build CARLA with UE5

J160KU, 2024-08-08:
Hello everybody,

On GitHub Ubuntu 22 is recommended:
[image]

Following the link further down to the official CARLA docs not Ubuntu 20 but 22 is recommended:
[image]

Could you please define this clearly? Otherwise I would expect that both versions Ubuntu 20 and 22 are recommended.

DnDare, 2024-08-09:
HI, 
If you wany to try UE5.3  you've better use Ubuntu 22.04

gsmario, 2024-08-09:
Yes, as @DnDare said, you should use ubuntu 22.04 to install carla on ue5.
Thank you for leting us know about this, we will update the documentation!

MattRoweEAIF, 2024-08-09:
README.md is now updated, thank you for pointing this out. Closing this issue

closed 2024-08-09 by MattRoweEAIF