carla-simulator/carla #8043: Can not run Carla 0.9.15(precompiled/released) by ./CarlaUE4.sh

Noah-Zhiyi-Lin, 2024-08-09:
Hi everyone! I am a student, and I am trying to use Carla for my research on Autonomous Driving. However, I meet a tough problem, which makes me mad now. 

The following is my problem:
I have tried to launch Carla by ./CarlaUE4.sh which is mentioned in the official documentation, but it can not run normally and report
```
Disabling core dumps.
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Serial number of failed request:  296
  Current serial number in output stream:  306
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)

```
in my terminal.

Here is the information of my environment:
OS: Ubuntu 20.04 LTS
CPU: Intel i9-11900
GPU: Nvidia RTX-3070 and  the Nvidia driver version is  550.107.02
RAM: 32GB

I have tried my best to solve this problem and referred some issues.
In #3490, some answers said that such problem can be solved by `./CarlaUE4.sh -opengl`, but Carla uses valkan after 0.9.12, so this method can not work. The other method is using `./CarlaUE4.sh -perfernvidia`. I tried this and received the same error mentioned above.
In #2232, some other answers suggested to try to use `export VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/nvidia_icd.json"`. I found the `nvidia_icd.json` in `/etc/vulkan/icd.d`, so I tried to use `export VK_ICD_FILENAMES="/etc/vulkan/icd.d/nvidia_icd.json"`. However, the program reports that `Can not find a compatible Vulkan Driver(ICD).`

I am very upset now because I actually have no idea to solve this problem. I would be extremely appreciated if you can give me some advice, regardless of whether it ultimately works or not.

PatrickPromitzer, 2024-08-13:
Hi,
are you local on the computer or do you use ssh?

You can try 
```
./CarlaUE4.sh -RenderOffScreen
```
which will deactivate the spectator and GUI.
If the "-RenderOffScreen" works, your computer can't open the window Carla wants to to show.

Noah-Zhiyi-Lin, 2024-08-13:
Hi, thank you @PatrickPromitzer. I want to use Carla on my local computer, and I have tried your method, `./CarlaUE4.sh -RenderOffScreen`. The terminal did not report any error, and no window was opened. So, I think you are right. But do you have any idea that why my computer can not open the window which Carla wants to show?

Noah-Zhiyi-Lin, 2024-08-13:
@PatrickPromitzer, I thought the problem was caused by the Nvidia driver at first, so I reinstalled the OS and Nvidia driver today. However, this did not work and the terminal reported the same error which was mentioned in the issue.

Noah-Zhiyi-Lin, 2024-08-13:
> Hi, are you local on the computer or do you use ssh?
> 
> You can try
> 
> ```
> ./CarlaUE4.sh -RenderOffScreen
> ```
> 
> which will deactivate the spectator and GUI. If the "-RenderOffScreen" works, your computer can't open the window Carla wants to to show.

Hi, thank you @PatrickPromitzer. I want to use Carla on my local computer, and I have tried your method, ./CarlaUE4.sh -RenderOffScreen. The terminal did not report any error, and no window was opened. So, I think you are right. But do you have any idea that why my computer can not open the window which Carla wants to show?

I thought the problem was caused by the Nvidia driver at first, so I reinstalled the OS and Nvidia driver today. However, this did not work and the terminal reported the same error which was mentioned in the issue.

Noah-Zhiyi-Lin, 2024-08-16:
> Hi, are you using the same account you are logged in with? If you use switch account on your machine in the terminal ("su username"), it can't show a window, too.
> 
> You can try if other gui software work in your terminal, and look if it is only a carla problem or a Ubuntu terminal problem.
> 
> ```
> xeyes
> ```

Hi, thank you for your reply! I indeed used the same account I was logged in with.

I have tried `xeyes` and it worked (I saw a pair of cute eyes). So, do you think it is only a Carla problem?

PatrickPromitzer, 2024-08-19:
Hi,
could be, or could be something difference.
It is hard to tell with the different parts which could create an problem.

I know that Carla works, but it could be a special case which could give Carla an error.
For me it looks like some kind of GUI or graphic card problem.
Could be permission, driver, software,or something else.

You could try to run Carla with ```./CarlaUE4.sh -RenderOffScreen``` and start an example script from 
```
path/to/Carla/PythonAPI/examples/
```
Maybe it crashes with an example, too, and showing the GUI was the limit which pushed Carla the crash in the first place.

Noah-Zhiyi-Lin, 2024-08-28:
> Hi, could be, or could be something difference. It is hard to tell with the different parts which could create an problem.
>
> I know that Carla works, but it could be a special case which could give [...]

Hi! I have tried your method `./CarlaUE4.sh -RenderOffScreen` to run the Carla server without rendering and I found something interesting. I ran the `automatic_control.py` by
```
python automatic_control.py
```
and the simulation window could be shown normally.

It seems that the Carla server can not be rendered but the Carla client can be rendered.

PatrickPromitzer, 2024-08-28:
Hi.

The script you tried is using pygame, which creates the pictures with a camera sensor. At this point we know, the simulation itself works, but the preview window doesn't.

You look try if Carla is using GPU resources.
- ./CarlaUE4.sh -RenderOffScreen
- python automatic_control.py
- nvidia-smi

While the simulation is running, repeat the "nvidia-smi" command.
Carla should use GPU resources.
You can try the same without "-RenderOffScreen"

suihan21, 2024-10-13:
I had the same problem and wondered if you had solved it?
(base) rossuihan@rossuihan:~/CARLA_0.9.10.1$ ./CarlaUE4.sh --world-port=2000 -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 3762] 
Failed to link program [Program V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865]. Current total programs: 220, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=144272 LargeMemoryPoolOffset=275408 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

PatrickPromitzer, 2024-10-14:
> I had the same problem and wondered if you had solved it? (base) rossuihan@rossuihan:~/CARLA_0.9.10.1$ ./CarlaUE4.sh --world-port=2000 -opengl 4.24.3-0+++UE4+Release-4.24 518 0 Disabling core dumps. [...]

Hi,
this is a common output for a Carla error, and doesn't tell the kind of error.

Check if the port is free
```
sudo lsof -i:2000
```
and try starting Carla without GUI
```
./CarlaUE4.sh -RenderOffScreen
```

CaMi1le, 2024-10-23:
hi I am using Ubuntu22.04 with RTX3090 running carla 0.9.12, I try your methods above but seems to have some same issue, the terminal still get crashed in about 50 secs.

`(base) zhaotr@zhaotr-Tower:~/Carla/CARLA_avp_final$ ./CarlaUE4.sh -RenderOffScreen
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
WARNING: cannot parse georeference: ''. Using default values. 
LowLevelFatalError [File:Unknown] [Line: 1214] 
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)`

jcuic5, 2024-11-14:
> hi I am using Ubuntu22.04 with RTX3090 running carla 0.9.12, I try your methods above but seems to have some same issue, the terminal still get crashed in about 50 secs.
>
> `(base) [...]

Hey, I meet similar problem, have you solved it?

chuanye-Wang, 2024-11-16:
same issue, does anyone know how to solve it?

Noah-Zhiyi-Lin, 2024-11-16:
To all:

I turned to another machine on mid of August because I can not solve this problem, and I have finished my using of CARLA now. I still have no idea to solve this problem. I think may be some of you can contact the team of CARLA. It seems that this is not a uncommon problem. I hope this problem can be solved as soon as possible.

This issue will be open until the problem is solved.

CaMi1le, 2024-11-17:
> same issue, does anyone know how to solve it?

> Hey, I meet similar problem, have you solved it?

I solved it by downgrade the system to 20.04, then reinstall the NV-driver with openGL. My friend works fine on his 22.04, so I am not sure if it is the openGL issue in my previous failure

suihan21, 2024-11-17:
You can try this./CarlaUE4.sh -prefernvidia发自我的手机-------- 原始邮件 --------发件人： CaMi1le ***@***.***>日期： 2024年11月17日周日 晚上10:11收件人： carla-simulator/carla ***@***.***>抄送： suihan ***@***.***>, Comment ***@***.***>主    题： Re: [carla-simulator/carla] Can not run Carla 0.9.15(precompiled/released) by ./CarlaUE4.sh (Issue #8043)

same issue, does anyone know how to solve it?

hi I am using Ubuntu22.04 with RTX3090 running carla 0.9.12, I try your methods above but seems to have some same issue, the terminal still get crashed in about 50 secs.
(base) ***@***.***:~/Carla/CARLA_avp_final$ ./CarlaUE4.sh -RenderOffScreen 4.26.2-0+++UE4+Release-4.26 522 0 Disabling core dumps. WARNING: cannot parse georeference: ''. Using default values. LowLevelFatalError [File:Unknown] [Line: 1214] GameThread timed out waiting for RenderThread after 60.00 secs Signal 11 caught. Malloc Size=65538 LargeMemoryPoolOffset=65554 CommonUnixCrashHandler: Signal=11 Malloc Size=131160 LargeMemoryPoolOffset=196744 Malloc Size=131160 LargeMemoryPoolOffset=327928 Engine crash handling finished; re-raising signal 11 for the default handler. Good bye. Segmentation fault (core dumped)

Hey, I meet similar problem, have you solved it?

I solved it by downgrade the system to 20.04, then reinstall the NV-driver with openGL. My friend works fine on his 22.04, so I am not sure if it is the openGL issue in my previous failure

zdy1013, 2025-01-08:
> 我遇到了同样的问题，想知道你有没有解决它？ （基础） rossuihan@rossuihan：~/CARLA_0.9.10.1$ ./CarlaUE4.sh --world-port=2000 -opengl 4.24.3-0+++UE4+Release-4.24 518 0 禁用核心转储。 LowLevelFatalError [文件：未知] [行：3762] 链接程序 [程序 V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865] 失败。当前程序总数：220，预编译：0 捕获信号 11。 Malloc 大小=65538 LargeMemoryPoolOffset=65554 CommonUnixCrashHandler：信号=11 Malloc 大小=65535 LargeMemoryPoolOffset=131119 Malloc 大小=144272 LargeMemoryPoolOffset=275408 引擎崩溃处理完成;re-raise signal 11 的默认处理程序。再见。分段错误（核心转储）

Hello, I have the same problem, how did you solve it?

suihan21, 2025-01-12:
You can try this command./CarlaUE4.sh -prefernvidia发自我的手机-------- 原始邮件 --------发件人： ZPOP ***@***.***>日期： 2025年1月8日周三 中午11:15收件人： carla-simulator/carla ***@***.***>抄送： suihan ***@***.***>, Comment ***@***.***>主    题： Re: [carla-simulator/carla] Can not run Carla 0.9.15(precompiled/released) by ./CarlaUE4.sh (Issue #8043)

我遇到了同样的问题，想知道你有没有解决它？ （基础） ***@***.***：~/CARLA_0.9.10.1$ ./CarlaUE4.sh --world-port=2000 -opengl 4.24.3-0+++UE4+Release-4.24 518 0 禁用核心转储。 LowLevelFatalError [文件：未知] [行：3762] 链接程序 [程序 V_3AA5F7CCAD2B351344BE53DDA22E8BAEBA78720A_P_ABADFE631118A2CF0719C49D3F936DABF0CE8865] 失败。当前程序总数：220，预编译：0 捕获信号 11。 Malloc 大小=65538 LargeMemoryPoolOffset=65554 CommonUnixCrashHandler：信号=11 Malloc 大小=65535 LargeMemoryPoolOffset=131119 Malloc 大小=144272 LargeMemoryPoolOffset=275408 引擎崩溃处理完成;re-raise signal 11 的默认处理程序。再见。分段错误（核心转储）

Hello, I have the same problem, how did you solve it?