carla-simulator/carla #8044: Abnormal Dark Areas on Imported Car Paint

GoodarzMehr, 2024-08-09:
Hi,

For an imported vehicle, I'm applying the car paint material as instructed in the tutorial, and the static mesh view looks something like this:
[image]

However, when running the simulation, dark areas appear on the car's body, as seen below:
[image]

Here are the properties of the car paint material instance that I'm using:
[image]

Is there a step to the process that I've perhaps missed?

gsmario, 2024-08-09:
Hello! I'm assuming that you have glasses or some other meshes beside the skeletal mesh of the car.
You should disable the mesh distance field of those meshes, because from the images, I think thats the problem.
Probably the glass mesh, because the dark areas appears around them.
Also, you can check this going into unlit mode in editor. If there is no dark areas, then is the mesh distance field.
Try it and tell me if it works!

GoodarzMehr, 2024-08-10:
Thanks for your help! That was the case, and unchecking _Affect Distance Field Lighting_ for those meshes in the blueprint resolved the issue.

One question though, I checked the blueprint of other cars and this field is checked for those, so why doesn't this seem to be an issue for those cars?